nickg/nvc #415: SIGABRT and two fatals

jklockars, 2021-04-22:
Three at once this time. ;-)

There is a fatal error regarding a signal with 'range, where the same using 'high downto 'low works fine.
The same thing for a variable causes a SIGABRT.

As a bonus, another fatal for an initial others assignment to an array of record.

Not including logs here.
Please, let me know if you cannot replicate the errors.

[bug5.txt](https://github.com/nickg/nvc/files/6354570/bug5.txt)
[bug6.txt](https://github.com/nickg/nvc/files/6354571/bug6.txt)

nickg, 2021-05-04:
I think these should be fixed now. Would you mind testing again?

jklockars, 2021-05-04:
Seems to be fine now. :-)
Initially I got a linker error for the initialiser assignment in my real code, but that was apparently due to the array (testing with a fixed array instead of allocating memory on the fly) being larger than 4 GByte, and I can understand that might be an issue.

I did run into one other small thing just now when testing my real code with the recent updates. Note that this was for obviously bad code - I got my types mixed up - but a nicer way to tell the developer that he is being stupid might still be a good idea ;-)

The "mem" array below was _not_ actually an access type, and should not have been assigned via new, but...

** Error: cannot determine type of allocator expression from context
	File memory.vhd, Line 259
            mem(free)          := new chunk_item;
                                  ^^^^^^^^^^^^^^
Assertion failed: (type_is_access(access_type)), function sem_check_new, file ../src/sem.c, line 4186.

*** Caught signal 6 (SIGABRT) ***

[0x1021c0bd4] 3   nvc                                 0x00000001021c0bd4 bt_sighandler + 328
...

nickg, 2021-05-05:
> Initially I got a linker error for the initialiser assignment in my real code, but that was apparently due to the array (testing with a fixed array instead of allocating memory on the fly) being larger than 4 GByte, and I can understand that might be an issue.

I guess this is a problem with some address being too large for a 32-bit relocation. I can try to reproduce it but not sure if it will be simple to fix or not.

> I did run into one other small thing just now when testing my real code with the recent updates. Note that this was for obviously bad code - I got my types mixed up - but a nicer way to tell the [...]

Thanks. I've fixed this one too.

closed 2021-05-05 by nickg