nickg/nvc #417: file locking problems using nvc in NFS-mounted directories

dbf, 2021-06-22:
nvc does currently not play well when used in NFSv4-mounted directories ("Fatal: flock: Bad file descriptor" when trying to analyse a file into work library). Trying the same in a local file system works fine. The problem appears to be the use of flock(). I suggest a patch like the one attached (warning: My C programming skills are ragged at best, so there may be unwanted side effects).

[nvc-fcntl-diff.txt](https://github.com/nickg/nvc/files/6693022/nvc-fcntl-diff.txt)

nickg, 2021-06-22:
On 2021-06-22 17:42, Daniel B. wrote:
> nvc does currently not play well when used in NFSv4-mounted
> directories ("Fatal: flock: Bad file descriptor" when trying to
> analyse a file into work library). Trying the same in a local file
> system works fine. The problem appears to be the use of flock(). I
> suggest a patch like the one attached (warning: My C programming
> skills are ragged at best, so there may be unwanted side effects).
> 
> nvc-fcntl-diff.txt [1]
> 

Thanks for the problem report and patch. Are you using Linux or some 
other operating system? I tried just now on a NFSv4 mounted filesystem 
on Linux and it worked ok, although I can imagine Mac/BSD would behave 
differently.

I also saw this in the Linux flock(2) man page:

> Since Linux 2.6.12, NFS clients support flock() locks by emulating them
> as fcntl(2)  byte-range locks  on the entire file.

So it looks like the two locking mechanisms should be equivalent.

The memset calls in the patch seem wrong: they zero out the F_RDLCK, 
etc. parameters set on the previous line. What happens if you take those 
out?

dbf, 2021-06-22:
> Thanks for the problem report and patch. Are you using Linux or some other operating system? I tried just now on a NFSv4 mounted filesystem on Linux and it worked ok, although I can imagine Mac/BSD [...]

NFS-client runs AlmaLinux 8.4 (effectively RHEL 8.4), NFS-server runs CentOS 7.9.2009. fstab on client reads

`server:/home   /home   nfs4   defaults,lookupcache=none   0   0`

which ends up as

`server:/home on /home type nfs4 (rw,relatime,vers=4.2,rsize=1048576,wsize=1048576,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=XXX,lookupcache=none,local_lock=none,addr=YYY)`

Don't know, why `local_lock=none` is in there; according to `man 5 nfs` this option is for NFS2/3 only.

> So it looks like the two locking mechanisms should be equivalent. The memset calls in the patch seem wrong: they zero out the F_RDLCK, etc. parameters set on the previous line. What happens if you take those out?

The error reoccurs and i can't even build:

  $ make
  CC       src/nvc.o
  [...]
    CXXLD    bin/nvc
    NVC      lib/std/STD.STANDARD
    NVC      lib/ieee/IEEE.STD_LOGIC_1164
    NVC      lib/ieee/IEEE.NUMERIC_BIT
  ** Fatal: fcntl_WRLCK: Bad file descriptor
  make[1]: *** [Makefile:3057: lib/ieee/IEEE.NUMERIC_BIT] Error 1
  make: *** [Makefile:1360: all] Error 2

So my patch is garbage, sorry. If fcntl() doesn't make a difference, there's probably either NFS misconfigured on my side or a bug in the operating system. I've attached some `strace` output that may be useful. Otherwise i'm out of ideas. Close?

[nvc-strace.txt](https://github.com/nickg/nvc/files/6694017/nvc-strace.txt)

nickg, 2021-06-23:
> `server:/home on /home type nfs4 (rw,relatime,vers=4.2,rsize=1048576,wsize=1048576,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=XXX,lookupcache=none,local_lock=none,addr=YYY)` [...]

I suspect there's some mount option that would fix it but unfortunately it's beyond my NFS knowledge. The `local_lock=none` option seems to mean a NFS lock server is involved in all file locking operations.

> The error reoccurs and i can't even build:
>
> $ make
> CC src/nvc.o
> [...]
> CXXLD bin/nvc
> NVC lib/std/STD.STANDARD
> NVC lib/ieee/IEEE.STD_LOGIC_1164
> NVC lib/ieee/IEEE.NUMERIC_BIT
> ** Fatal: fcntl_WRLCK: Bad [...]

OK, the error is the same as with `flock()` which makes sense as the manual says `flock` falls back to `fcntl` on NFS.

> I've attached some `strace` output that may be useful. Otherwise i'm out of ideas. Close?
> 

Can you try the `flock` shell utility and see if that behaves differently? E.g.

```
flock -x lock-file true
```

On my system that just calls `flock()` directly. If it works on your NFS mount then the strace output would be useful to see what it does.

dbf, 2021-06-23:
Thank you for your patience, Nick.

I tried to use shell flock, and saw the following:

Used on a local filesystem:

```
erwin@sopwith:/tmp$ strace flock -x lock-file true
execve("/usr/bin/flock", ["flock", "-x", "lock-file", "true"], 0x7fff503f4ff8 /* 41 vars */) = 0
brk(NULL)                               = 0x563c855a1000
[...]
openat(AT_FDCWD, "lock-file", O_RDONLY|O_CREAT|O_NOCTTY, 0666) = 3
flock(3, LOCK_EX)                       = 0
rt_sigaction(SIGCHLD, {sa_handler=SIG_DFL, sa_mask=[CHLD], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fdd1f38f400}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fdd1fb40a10) = 2388215
wait4(2388215, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 2388215
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=2388215, si_uid=1338400007, si_status=0, si_utime=0, si_stime=0} ---
close(1)                                = 0
close(2)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++
```

Looks ok to me. Same thing in NFS-mounted home directory:

```
erwin@sopwith:~$ strace flock -x lock-file true
execve("/usr/bin/flock", ["flock", "-x", "lock-file", "true"], 0x7fff964509c8 /* 41 vars */) = 0
brk(NULL)                               = 0x55c9c3696000
[...]
openat(AT_FDCWD, "lock-file", O_RDONLY|O_CREAT|O_NOCTTY, 0666) = 3
flock(3, LOCK_EX)                       = -1 EBADF (Bad file descriptor)
access("lock-file", R_OK|W_OK)          = 0
close(3)                                = 0
openat(AT_FDCWD, "lock-file", O_RDWR|O_CREAT|O_NOCTTY, 0666) = 3
flock(3, LOCK_EX)                       = 0
rt_sigaction(SIGCHLD, {sa_handler=SIG_DFL, sa_mask=[CHLD], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7f475994c400}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f475a0fda10) = 2388825
wait4(2388825, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 2388825
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=2388825, si_uid=1338400007, si_status=0, si_utime=0, si_stime=0} ---
close(1)                                = 0
close(2)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++
```

It appears, that on NFS, shell flock tries a second time to acquire the lock and then succeeds (after having failed the first time). On the 2nd try, `openat()` is done with `O_RDWR` instead of `O_RDONLY`, though. Could it be, probably for some reason related to NFSv4 and/or the now ancient 3.10 Linux kernel on my CentOS NFS server, mandatory to open a file rw in order to get an exclusive lock?

nickg, 2021-06-23:
> It appears, that on NFS, shell flock tries a second time to acquire the lock and then succeeds (after having failed the first time). On the 2nd try, `openat()` is done with `O_RDWR` instead of [...]

OK, so does something simple like this work?

```diff
diff --git a/src/lib.c b/src/lib.c
index 0790234c0e68..3aaf0003d2e5 100644
--- a/src/lib.c
+++ b/src/lib.c
@@ -188,7 +188,7 @@ static lib_t lib_init(const char *name, const char *rpath, int lock_fd)

    if (l->lock_fd == -1 && rpath != NULL) {
       LOCAL_TEXT_BUF lock_path = lib_file_path(l, "_NVC_LIB");
-      if ((l->lock_fd = open(tb_get(lock_path), O_RDONLY)) < 0)
+      if ((l->lock_fd = open(tb_get(lock_path), O_RDWR)) < 0)
          fatal_errno("lib_init: %s", tb_get(lock_path));

       file_read_lock(l->lock_fd);
```

dbf, 2021-06-24:
Apparently yes, but it has side effects:

```
erwin@sopwith:/tmp$ strace nvc -a hw.vhdl 
execve("/home/opt/nickg-nvc/git-latest/bin/nvc", ["nvc", "-a", "hw.vhdl"], 0x7ffd13927940 /* 68 vars */) = 0
[...]
lstat("/home/opt/nickg-nvc/git-latest/share/nvc/std", {st_mode=S_IFDIR|0775, st_size=212, ...}) = 0
openat(AT_FDCWD, "/home/opt/nickg-nvc/git-latest/share/nvc/std/_NVC_LIB", O_RDWR) = -1 EACCES (Permission denied)
write(2, "\33[31m** Fatal: \33[0mlib_init: /ho"..., 112** Fatal: lib_init: /home/opt/nickg-nvc/git-latest/share/nvc/std/_NVC_LIB:
          Permission denied
) = 112
write(2, "\33[0m** Note: \33[0mGC: freed 4 obj"..., 50** Note: GC: freed 4 objects; 0 allocated
) = 50
exit_group(1)                           = ?
+++ exited with 1 +++
```

nvc is built, installed and owned by a non-privileged user existing for this (and only this) very purpose. Actual users of nvc do not have write permissions in the installation directory. If write permissions are given, it works (apparently, i haven't done extensive testing, yet). If no other files in the installation directory are required to be opened rw (apparently this is the case, but i may be overlooking something), i'd be happy with a post-install workaround like

```
find /home/opt/nickg-nvc/git-latest/share/nvc -name "_NVC_LIB" -exec chmod o+w {} \;
```

I trust my users in this regard.

nickg, 2021-06-24:
> nvc is built, installed and owned by a non-privileged user existing for this (and only this) very purpose. Actual users of nvc do not have write permissions in the installation directory. If write permissions are given, it works (apparently, i haven't done extensive testing, yet). 

Right, so the exclusive lock is only ever needed for writing into that library, which can't happen if it's in a read-only location. I've just pushed a commit 24f2196 to the master branch that first tries to open O_RDWR and then falls back to O_RDONLY if that fails with EACCES. I think this ought to fix your problem with no other regressions. Would you mind testing?

dbf, 2021-06-25:
We will gladly do that, but it will take a couple of days, since some users are currently absent.

dbf, 2021-06-29:
The fix is working for us. Thank you very much for your help on this issue.

closed 2021-06-29 by dbf