nickg/nvc #418: build-xilinx-vivado fails with latest 1.6-devel

dbf, 2021-06-29:
_./tools/build-xilinx-vivado.rb_ fails with latest 1.6-devel on my system:

```
$ ./tools/build-xilinx-vivado.rb
Using Vivado installation in /opt/xilinx/vivado2018.1/Vivado/2018.1
[...]
nvc --work=/localhome/animal/.nvc/lib/unisim -a --relax=prefer-explicit,pure-files /opt/xilinx/vivado2018.1/Vivado/2018.1/data/vhdl/src/unisims/primitive/DSP48E2.vhd
** Error: ambiguous use of enumeration literal '1' (STD_ULOGIC, BIT)
        File /opt/xilinx/vivado2018.1/Vivado/2018.1/data/vhdl/src/unisims/primitive/DSP48E2.vhd, Line 2038
      U_out <= std_logic_vector('1'          & (unsigned(mult(43 downto 0)) and ps_
                                ^^^
** Note: visible declaration of '1' as BIT
        File ../lib/std/standard.vhd, Line 9
** Note: visible declaration of '1' as STD_ULOGIC
        File ../lib/ieee/std_logic_1164.vhdl, Line 62
```

_./tools/build-xilinx-vivado.rb_ works fine with nvc 1.5.1 and Vivado 2018.1, so this may be caused by a change in 1.6-devel.

nickg, 2021-06-30:
Yes it's probably a regression on the development branch. I don't have Vivado installed at the moment - is it possible to paste the complete statement that causes the error starting at line 2038 in that file?

nickg, 2021-06-30:
I found at least one thing wrong that may cause this which is fixed in the commit above. Would you mind retesting with the latest master?

dbf, 2021-06-30:
Thank you very much, the problem appears to be solved by your latest commit.
For completeness, here's the (hopefully) complete statement in question:

```
[...]
--  DSP_C_DATA
-- *********************************************************
-- *** Input register C with 1 level deep of register
-- *********************************************************

--  CLK_creg  <=  CLK_in  when  CREG_BIN = CREG_1  else  '0';

  process (CLK_creg) begin
    if  (glblGSR = '1') then C_reg <= (others => '0');
    elsif (rising_edge(CLK_creg)) then
      if    (RSTC_in = '1') then C_reg <= (others => '0');
      elsif (CEC_in = '1')  then C_reg <= C_in;
      end if;
    end if;
  end process;

  C_DATA_out  <=  C_reg  when  CREG_BIN = CREG_1  else  C_in;

--  DSP_MULTIPLIER
  a_mult_mux <= signed(A2A1_in) when AMULTSEL_BIN = AMULTSEL_A else
                signed(AD_DATA_in);
  b_mult_mux <= signed(B2B1_in) when BMULTSEL_BIN = BMULTSEL_B else
                signed(AD_DATA_in(17 downto 0)) ;
-- replace {18{a_mult_mux[26]}} shorthand
  a_mult_mux_26_18 <= "000000000000000000" when a_mult_mux(26) = '0' else
                      "111111111111111111";
-- replace {27{b_mult_mux[17]}} shorthand
  b_mult_mux_17_27 <= "000000000000000000000000000" when b_mult_mux(17) = '0' else
                      "111111111111111111111111111";

  AMULT26_out <= a_mult_mux(26);
  BMULT17_out <= b_mult_mux(17);
  U_out <= std_logic_vector('1'          & (unsigned(mult(43 downto 0)) and ps_u_mask));
  V_out <= std_logic_vector(not mult(44) & (unsigned(mult(43 downto 0)) and ps_v_mask));

  mult <= (others => '0') when USE_MULT_BIN = USE_MULT_NONE else
          a_mult_mux * b_mult_mux;
--          (a_mult_mux_26_18 & a_mult_mux) * (b_mult_mux_17_27 & b_mult_mux);
[...]
```

closed 2021-06-30 by dbf