nickg/nvc #421: nvc 1.5.2 test failure

chenrui333, 2021-07-30:
👋  trying to upgrade the formula to the latest release build, but runs into some error as below:

```
** Fatal: access: /usr/local/Cellar/nvc/1.5.2/share/nvc/std/_NVC_LIB: Operation not permitted
```

relates to Homebrew/homebrew-core#82232

nickg, 2021-07-31:
How can I reproduce this? I tried:

```
brew install --build-from-source ./nvc.rb
```

With the `nvc.rb` from the linked PR and it worked fine.

nickg, 2021-07-31:
Never mind, it fails with `brew test ./nvc.rb`. I'm not very familiar with macOS but it seems when running under `sandbox-exec`, `access(2)` can fail with `EPERM` (contrary to what the man page says). You should be able to apply 64c25212 as a patch to work around this.

closed 2021-07-31 by nickg

chenrui333, 2021-08-15:
@nickg the patch works for me, thanks!!

commit 64c25212 (2021-07-31): Handle access(2) returning EPERM in macOS sandbox. Fixes #421
commit fc13d62f (2021-07-31): Handle access(2) returning EPERM in macOS sandbox. Fixes #421