nickg/nvc #422: Support for `text` types?

Cushychicken, 2021-08-17:
I've been working on a little DDS block as a way to get sharper on VHDL and DSP, and I've started wanting to add file output of my sin ROM so that I can use Python to evaluate the noise spectrum of my sin rom. 

However, it seems that I can't do file output the expected way in VHDL, which is using a `file` object with type `text`, as it has no visible declaration:

```vhdl
nreilly@Nashs-MacBook-Pro dds-engine % nvc -a test/test_sin_rom.vhd src/sin_rom_11bits_1024words.vhd --relax=prefer-explicit -e test_sin_rom -r -w
** Error: no visible declaration for TEXT
	File test/test_sin_rom.vhd, Line 46
      file file_Result_Vectors : TEXT;
```

Does this mean that nvc doesn't support `ieee.std_logic_textio.all;`?

Cushychicken, 2021-08-17:
Closing this - dumb error on my part. I didn't realize you need two libraries to properly do text io: `ieee.std_logic_textio.all`, and `std.textio.all`

closed 2021-08-18 by Cushychicken

nickg, 2021-08-18:
On 2021-08-18 06:29, Nash Reilly wrote:
> Closing this - dumb error on my part. I didn't realize you need two
> libraries to properly do text io: ieee.std_logic_textio.all, and
> std.textio.all
> 

Yes `use foo.all` only makes names declared directly within that package 
visible, it doesn't transitively import names from other packages. So 
you still need `use std.textio.all` even though the package definition 
of std_logic_textio also does that.