nickg/nvc #423: ** Fatal: sorry, attribute LAST_ACTIVE not implemented

avelure, 2021-10-04:
I ran into this issue when analyzing some vendor libraries
Here is an MWE

```vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture rtl of test is
  signal sig : std_logic := '0';
begin

  process
  begin
    sig <= '0' after 1 ps;
    wait for 3 ps;
    assert sig'last_active = 3 ps - 1 ps report "Signal should have been active" severity error;
    wait;
  end process;
end architecture rtl;
```

While looking around the source I also noticed the attributes `'driving_value` and `'driving` are missing as well, though I don't have a need for their functionality at the moment. I could add separate issues and MWEs for them if preferred,

nickg, 2021-10-05:
Thanks for reporting this. I've added `'last_active` as well as `'driving` and `'driving_value`.

closed 2021-10-05 by nickg

avelure, 2021-10-05:
Thanks for the quick work!
I'll continue to try to get my project runnable with nvc.

commit a57e0f4c (2021-10-05): Add support for 'LAST_ACTIVE attribute. Fixes #423