nickg/nvc #424: Linking problem with Mingw64

avelure, 2021-10-05:
I have a problem with undefined references to IEEE libraries when running with MSYS2 Mingw64 on win10.

The following code
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

PACKAGE test IS
  FUNCTION to_signInteger ( din : std_logic_vector ) return integer;
END test;

PACKAGE BODY test IS

  FUNCTION to_signInteger ( din : std_logic_vector ) return integer is
  begin
    return to_integer(signed(din));
  end to_signInteger;

END test;
```
causes
```
C:\proj\public\minimal\nvc$ nvc -a test_to_integer.vhd
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\proj\public\minimal\nvc\work\_WORK.TEST.obj: in function `WORK.TEST.TO_SIGNINTEGER_lp_V_rp_I':
test_to_integer.vhd:13: undefined reference to `IEEE.NUMERIC_STD.TO_INTEGER_lp_23IEEE.NUMERIC_STD.SIGNED_rp_I'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\proj\public\minimal\nvc\work\_WORK.TEST.obj:WORK.TEST:(.rdata$.refptr.IEEE.NUMERIC_STD[.refptr.IEEE.NUMERIC_STD]+0x0): undefined reference to `IEEE.NUMERIC_STD'
collect2.exe: error: ld returned 1 exit status
C:\msys64\mingw64\bin\gcc.exe -shared -o C:\proj\public\minimal\nvc\work\_WORK.TEST.dll C:\proj\public\minimal\nvc\work\_WORK.TEST.obj -Wl,--out-implib=C:\proj\public\minimal\nvc\work\_WORK.TEST.lib -LC:\proj\public\minimal\nvc\work -LC:\msys64\mingw64\share\nvc -lnvcimp
** Fatal: C:\msys64\mingw64\bin\gcc.exe failed with status 1
```

It seems to be missing the library reference as I can point it manually
```C:\msys64\mingw64\bin\gcc.exe -shared -o C:\proj\public\minimal\nvc\work\_WORK.TEST.dll C:\proj\public\minimal\nvc\work\_WORK.TEST.obj -Wl,--out-implib=C:\proj\public\minimal\nvc\work\_WORK.TEST.lib -LC:\proj\public\minimal\nvc\work -LC:\msys64\mingw64\share\nvc -lnvcimp -LC:\msys64\mingw64\share\nvc\ieee -l_IEEE.NUMERIC_STD```
and it works.

It causes the same issue if I run it from the Mingw shell. The same test works fine on Ubuntu.

The compile of NVC went fine except for various `-Wformat` warnings . I could run the `make check` fine as well.

I see there is some code in `cgen.c:4500` that iterates over the dependency libraries to add them with `-L` but it adds a `.lib` suffix to the path and then checks if the file exists, which fails. As far as I can see the the standard libraries are compiled with a file ending of `.dll` on windows and `.so` on linux. If I change to `.dll` at 4506 it works.

nickg, 2021-10-05:
The `.lib` files are supposed to be "import libraries" generated by the `-Wl,--out-implib` argument to gcc. I think at one point they were needed to correctly resolve symbols on Cygwin and I've just carried the code around ever since (I don't have any easy way to test on Windows except through GitHub Actions). It could be that Mingw doesn't need import libraries at all, and you can just link against the DLL directly, in which case we could simplify the Windows code a lot by removing that. It might break Cygwin but I don't think many people are using that any more.

https://stackoverflow.com/questions/27824102/how-do-import-libraries-work-and-why-doesnt-mingw-need-them

avelure, 2021-10-05:
According to this https://stackoverflow.com/questions/3573475/how-does-the-import-library-work-details/3573527#3573527 the GNU compiler for Windows does not need the lib, it can handle DLL directly. Which explains why it worked when I changed lib to dll.

My confusion is why the `make check` works, is it because it does not run the test with nvc from the commandline?

I guess if someone uses Cygwin they will open an issue, besides I don't see Cygwin listed as supported in the readme.

nickg, 2021-10-06:
I think I've found the problem: it's a simple typo in the makefile which meant the import libraries for the standard libs weren't installed. It should be fixed by the commit above. I'll look into removing them completely though.

> 
> My confusion is why the `make check` works, is it because it does not run the test with nvc from the commandline?
> 

It does test running the full program from the command line, but it does so from within the build tree (i.e. doesn't do `make install` first), so doesn't catch this issue. The GitHub Actions tests could probably be improved to do a simple sanity test after installing.

avelure, 2021-10-06:
Thanks, that solves it for me :)

nickg, 2021-10-06:
And the commit above completely removes import library generation for VHDL packages.

closed 2021-10-06 by nickg