nickg/nvc #425: "** Fatal: type kind T_RECORD does not have item I_ELEM" occurs when elaborating VHDL with complex records

ikwzm, 2021-10-05:
I have elaborated VHDL with complex records such as:

```VHDL
-----------------------------------------------------------------------------------
--
-----------------------------------------------------------------------------------
library ieee;
use     ieee.std_logic_1164.all;
package SMPL_TYPES is
    constant  SMPL_ADDR_MAX_WIDTH  :  integer := 64;
    constant  SMPL_DATA_MAX_WIDTH  :  integer := 1024;
    constant  SMPL_STRB_MAX_WIDTH  :  integer := SMPL_DATA_MAX_WIDTH/8;
    constant  SMPL_SIZE_MAX_WIDTH  :  integer := 8;

    constant  SMPL_INFO_WIDTH      :  integer := 4;
    subtype   SMPL_INFO_TYPE       is std_logic_vector(SMPL_INFO_WIDTH-1 downto 0);

    constant  SMPL_RESP_WIDTH      :  integer := 2;
    subtype   SMPL_RESP_TYPE       is std_logic_vector(SMPL_RESP_WIDTH-1 downto 0);
    
    type      SMPL_A_SIGNALS_TYPE is record
        ADDR     : std_logic_vector(SMPL_ADDR_MAX_WIDTH-1 downto 0);
        WRITE    : std_logic;
        SIZE     : std_logic_vector(SMPL_SIZE_MAX_WIDTH-1 downto 0);
        INFO     : SMPL_INFO_TYPE;
        VALID    : std_logic;
        READY    : std_logic;
    end record;
    constant  SMPL_A_SIGNALS_DONTCARE : SMPL_A_SIGNALS_TYPE := (
        ADDR    => (others => '-'),
        WRITE   => '-',
        SIZE    => (others => '-'),
        INFO    => (others => '-'),
        VALID   => '-',
        READY   => '-'
    );
    constant  SMPL_A_SIGNALS_NULL     : SMPL_A_SIGNALS_TYPE := (
        ADDR    => (others => '0'),
        WRITE   => '0',
        SIZE    => (others => '0'),
        INFO    => (others => '0'),
        VALID   => '0',
        READY   => '0'
    );
    type      SMPL_W_SIGNALS_TYPE is record
        DATA     : std_logic_vector(SMPL_DATA_MAX_WIDTH-1 downto 0);
        STRB     : std_logic_vector(SMPL_STRB_MAX_WIDTH-1 downto 0);
        VALID    : std_logic;
        READY    : std_logic;
    end record;
    constant  SMPL_W_SIGNALS_DONTCARE : SMPL_W_SIGNALS_TYPE := (
        DATA    => (others => '-'),
        STRB    => (others => '-'),
        VALID   => '-',
        READY   => '-'
    );
    constant  SMPL_W_SIGNALS_NULL     : SMPL_W_SIGNALS_TYPE := (
        DATA    => (others => '0'),
        STRB    => (others => '0'),
        VALID   => '0',
        READY   => '0'
    );
    type      SMPL_R_SIGNALS_TYPE is record
        DATA     : std_logic_vector(SMPL_DATA_MAX_WIDTH-1 downto 0);
        RESP     : SMPL_RESP_TYPE;
        VALID    : std_logic;
        READY    : std_logic;
    end record;
    constant  SMPL_R_SIGNALS_DONTCARE : SMPL_R_SIGNALS_TYPE := (
        DATA    => (others => '-'),
        RESP    => (others => '-'),
        VALID   => '-',
        READY   => '-'
    );
    constant  SMPL_R_SIGNALS_NULL     : SMPL_R_SIGNALS_TYPE := (
        DATA    => (others => '0'),
        RESP    => (others => '0'),
        VALID   => '0',
        READY   => '0'
    );
    type      SMPL_B_SIGNALS_TYPE is record
        RESP     : SMPL_RESP_TYPE;
        VALID    : std_logic;
        READY    : std_logic;
    end record;
    constant  SMPL_B_SIGNALS_DONTCARE : SMPL_B_SIGNALS_TYPE := (
        RESP    => (others => '-'),
        VALID   => '-',
        READY   => '-'
    );
    constant  SMPL_B_SIGNALS_NULL     : SMPL_B_SIGNALS_TYPE := (
        RESP    => (others => '0'),
        VALID   => '0',
        READY   => '0'
    );
    type      SMPL_SIGNALS_TYPE is record
        AR       : SMPL_A_SIGNALS_TYPE;
        AW       : SMPL_A_SIGNALS_TYPE;
        R        : SMPL_R_SIGNALS_TYPE;
        W        : SMPL_W_SIGNALS_TYPE;
        B        : SMPL_B_SIGNALS_TYPE;
    end record;
    constant  SMPL_SIGNALS_DONTCARE : SMPL_SIGNALS_TYPE := (
        AR      => SMPL_A_SIGNALS_DONTCARE,
        AW      => SMPL_A_SIGNALS_DONTCARE,
        R       => SMPL_R_SIGNALS_DONTCARE,
        W       => SMPL_W_SIGNALS_DONTCARE,
        B       => SMPL_B_SIGNALS_DONTCARE
    );
end SMPL_TYPES;
-----------------------------------------------------------------------------------
--
-----------------------------------------------------------------------------------
library ieee;
use     ieee.std_logic_1164.all;
use     WORK.SMPL_TYPES.all;
entity  CHANNEL_PLAYER is
    generic (
        CHANNEL         : integer;
        MASTER          : boolean   := FALSE
    );
end CHANNEL_PLAYER;
architecture MODEL of CHANNEL_PLAYER is
    function  GEN_INIT_signals return SMPL_SIGNALS_TYPE is
        variable  value : SMPL_SIGNALS_TYPE;
    begin
        value := SMPL_SIGNALS_DONTCARE;
        if (MASTER) then
            case CHANNEL is
                when 1 =>
                    value.AR       := SMPL_A_SIGNALS_NULL;
                    value.AR.READY := '-';
                when 2 =>
                    value.AW       := SMPL_A_SIGNALS_NULL;
                    value.AW.READY := '-';
                when 3  =>
                    value.W        := SMPL_W_SIGNALS_NULL;
                    value.W.READY  := '-';
                when 4  => 
                    value.R        := SMPL_R_SIGNALS_DONTCARE;
                    value.R.READY  := '0';
                when 5  =>
                    value.B        := SMPL_B_SIGNALS_DONTCARE;
                    value.B.READY  := '0';
                when others =>
                    null;
            end case;
        end if;
        return value;
    end function;
    constant  INIT_SIGNALS  : SMPL_SIGNALS_TYPE := GEN_INIT_SIGNALS;
begin 
    process
        variable  out_signals   : SMPL_SIGNALS_TYPE;
    begin
        out_signals := INIT_SIGNALS;
        wait;
    end process;
end MODEL;
-----------------------------------------------------------------------------------
--
-----------------------------------------------------------------------------------
library ieee;
use     ieee.std_logic_1164.all;
entity  MASTER_PLAYER is
end MASTER_PLAYER;
architecture MODEL of MASTER_PLAYER is
begin
    C0: entity WORK.CHANNEL_PLAYER generic map (0, FALSE);
    C1: entity WORK.CHANNEL_PLAYER generic map (1, TRUE );
    C2: entity WORK.CHANNEL_PLAYER generic map (2, TRUE );
    C3: entity WORK.CHANNEL_PLAYER generic map (3, TRUE );
    C4: entity WORK.CHANNEL_PLAYER generic map (4, TRUE );
    C5: entity WORK.CHANNEL_PLAYER generic map (5, TRUE );
end MODEL;
-----------------------------------------------------------------------------------
--
-----------------------------------------------------------------------------------
library ieee;
use     ieee.std_logic_1164.all;
entity  TEST_NG is
end     TEST_NG;
architecture MODEL of TEST_NG is
begin
    M: entity WORK.MASTER_PLAYER;
    S: entity WORK.MASTER_PLAYER;
end MODEL;
```

Then, a Fatal Error occurred as follows.

```shell
shell$ nvc -a test_ng.vhd
/usr/bin/ld -shared -o /home/ikwzm/work/nvc/issues/new/work/_WORK.SMPL_TYPES.so /home/ikwzm/work/nvc/issues/new/work/_WORK.SMPL_TYPES.o
shell$ nvc -e test_ng
** Fatal: type kind T_RECORD does not have item I_ELEM"
[0x7ff42125e051] ../src/object.c:56 object_lookup_failed
[0x7ff421236a1a] ../src/type.c:329 type_elem
[0x7ff42124e977] ../src/eval.c:2173 eval_value_to_tree
[0x7ff42124e945] ../src/eval.c:2165 eval_value_to_tree
[0x7ff42125268f] ../src/eval.c:2279 eval
[0x7ff421249769] ../src/simp.c:156 simp_tree
[0x7ff42125eb94] ../src/object.c:479 object_rewrite
[0x7ff42125ecbc] ../src/object.c:406 object_rewrite
[0x7ff42125ed5a] ../src/object.c:412 object_rewrite
[0x7ff421236175] ../src/tree.c:1186 tree_rewrite
[0x7ff42124a404] ../src/simp.c:1301 simplify
[0x7ff421246d03] ../src/elab.c:1151 elab_stmts
[0x7ff42124714b] ../src/elab.c:1426 elab_arch
[0x7ff421246d28] ../src/elab.c:1158 elab_stmts
[0x7ff42124714b] ../src/elab.c:1426 elab_arch
[0x7ff4212476fd] ../src/elab.c:1576 elab_entity_arch
[0x7ff421247a71] ../src/elab.c:1670 elab
[0x7ff4212181f4] ../src/nvc.c:324 process_command
[0x7ff421216c16] ../src/nvc.c:1062 main
```

nickg, 2021-10-06:
Thanks for reporting this bug. It should be fixed on the master branch now and I'll also apply the patch to the 1.5 branch.

ikwzm, 2021-10-06:
Thank you for your quick and polite response.