nickg/nvc #436: "** Fatal: expression cannot be folded to an integer constant" occurs when a record type field is specified for a range.

ikwzm, 2022-01-13:
```VHDL
-- test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
entity  TEST_NG is
end entity;
architecture MODEL of TEST_NG is
    type      VEC_RANGE_TYPE is record
              DATA_LO        :  integer;
              DATA_HI        :  integer;
    end record;
    function  SET_VEC_RANGE return VEC_RANGE_TYPE is
        variable  v          :  VEC_RANGE_TYPE;
    begin
        v.DATA_HI := 31;
        v.DATA_LO :=  0;
        return v;
    end function;
    constant  VEC_RANGE      :  VEC_RANGE_TYPE  := SET_VEC_RANGE;
    signal    data           :  std_logic_vector(VEC_RANGE.DATA_HI downto VEC_RANGE.DATA_LO);
begin
end MODEL;
```

```console
shell$ nvc --version
nvc 1.6-devel (Using LLVM 7.0.0)
Copyright (C) 2011-2022  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```

```console
shell$ nvc -a test_ng.vhd
shell$ nvc -e test_ng
** Fatal: expression cannot be folded to an integer constant
	File test_ng.vhd, Line 19
```

ikwzm, 2022-01-13:
By the way, with the following VHDL

```VHDL:test_ng_2.vhd
-- test_ng_2.vhd
library ieee;
use     ieee.std_logic_1164.all;
entity  SAMPLE is
    generic ( 
        DATA_BITS : integer := 0
    );
    port (
        DATA : std_logic_vector(DATA_BITS-1 downto 0)
    );
end entity;
architecture RTL of SAMPLE is
begin
end RTL;

library ieee;
use     ieee.std_logic_1164.all;
entity  TEST_NG_2 is
end entity;
architecture MODEL of TEST_NG_2 is
    type      VEC_RANGE_TYPE is record
              DATA_LO        :  integer;
              DATA_HI        :  integer;
    end record;
    function  SET_VEC_RANGE return VEC_RANGE_TYPE is
        variable  v          :  VEC_RANGE_TYPE;
    begin
        v.DATA_HI := 31;
        v.DATA_LO :=  0;
        return v;
    end function;
    constant  VEC_RANGE      :  VEC_RANGE_TYPE  := SET_VEC_RANGE;
    signal    data           :  std_logic_vector(VEC_RANGE.DATA_HI downto VEC_RANGE.DATA_LO);
begin
    U: entity WORK.SAMPLE generic map(DATA_BITS => data'length) port map(DATA => data);
end MODEL;
```
It looks like this.

```console
shell$ nvc -a test_ng_2.vhd
shell$ nvc -e test_ng_2
** Fatal: cannot rewrite generic DATA_BITS to tree kind T_ATTR_REF
[0x7fa8cc40e29c] ../src/util.c:687 fatal_trace
[0x7fa8cc4400b3] simp_ref.isra.7
[0x7fa8cc4400b3] ../src/simp.c:1677 simp_tree.lto_priv.493
[0x7fa8cc461b25] ../src/object.c:635 object_rewrite
[0x7fa8cc461bee] ../src/object.c:604 object_rewrite
[0x7fa8cc461c7a] ../src/object.c:610 object_rewrite
[0x7fa8cc461bee] ../src/object.c:604 object_rewrite
[0x7fa8cc461c7a] ../src/object.c:610 object_rewrite
[0x7fa8cc461bee] ../src/object.c:604 object_rewrite
[0x7fa8cc461bee] ../src/object.c:604 object_rewrite
[0x7fa8cc461c7a] ../src/object.c:610 object_rewrite
[0x7fa8cc41c6f0] ../src/tree.c:1165 tree_rewrite
[0x7fa8cc43d926] ../src/simp.c:1826 simplify_global
[0x7fa8cc444350] ../src/elab.c:991 elab_stmts
[0x7fa8cc44527e] ../src/elab.c:1385 elab_top_level.lto_priv.519
[0x7fa8cc4192e6] ../src/elab.c:1431 elab
[0x7fa8cc4192e6] ../src/elab.c:317 elaborate
[0x7fa8cc41561a] ../src/nvc.c:910 process_command
[0x7fa8cc40b99a] ../src/nvc.c:1006 main
```

nickg, 2022-01-13:
These two should be fixed now, thanks!

closed 2022-01-13 by nickg