nickg/nvc #438: "** Fatal: Expected &(ptr->pointer[-1]) to have kind VALUE_RECORD but is [{-2147483648,-2147483648,-2147483648}]" occurs when a record type vector element is specified as an argument of a subprogram in a function

ikwzm, 2022-01-14:
```VHDL:test_ng.vhd
-- test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
entity  TEST_NG is
    generic(
        INFO_SIZE  : integer := 1;
        INFO_BITS  : integer := 8
   );
end entity;
architecture MODEL of TEST_NG is
    type      INFO_RANGE_TYPE is record
              VAL_POS           : integer;
              DATA_LO           : integer;
              DATA_HI           : integer;
    end record;
    type      INFO_RANGE_VECTOR is array (integer range <>) of INFO_RANGE_TYPE;
    type      VEC_RANGE_TYPE is record
              VAL_LO            : integer;
              VAL_HI            : integer;
              DATA_LO           : integer;
              DATA_HI           : integer;
              INFO_LIST         : INFO_RANGE_VECTOR(INFO_SIZE-1 downto 0);
    end record;
    function  SET_VEC_RANGE return VEC_RANGE_TYPE is
        variable  v_pos         : integer;
        variable  d_pos         : integer;
        variable  v             : VEC_RANGE_TYPE;
        procedure SET_INFO_RANGE(INFO_RANGE: inout INFO_RANGE_TYPE; BITS: in integer) is
        begin
            INFO_RANGE.VAL_POS  := v_pos;
            INFO_RANGE.DATA_LO  := d_pos;
            INFO_RANGE.DATA_HI  := d_pos + BITS-1;
            v_pos := v_pos + 1;
            d_pos := d_pos + BITS;
        end procedure;
    begin
        v_pos := 0;
        d_pos := 0;
        v.VAL_LO  := v_pos;
        v.DATA_LO := d_pos;

        for i in 0 to INFO_SIZE-1 loop
            SET_INFO_RANGE(v.INFO_LIST(i), INFO_BITS);
        end loop;

        v.VAL_HI  := v_pos - 1;
        v.DATA_HI := d_pos - 1;

        return v;
    end function;
    constant  VEC_RANGE      :  VEC_RANGE_TYPE  := SET_VEC_RANGE;
    signal    data           :  std_logic_vector(VEC_RANGE.DATA_HI downto VEC_RANGE.DATA_LO);
begin

end MODEL;
```

```console
shell$ nvc --version
nvc 1.6-devel (Using LLVM 7.0.0)
Copyright (C) 2011-2022  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```

```console
shell$ nvc -a test_ng.vhd
shell$ nvc -e test_ng

Name       WORK.TEST_NG.SET_VEC_RANGE()33WORK.TEST_NG-MODEL.VEC_RANGE_TYPE.SET_INFO_RANGE(34WORK.TEST_NG-MODEL.INFO_RANGE_TYPEI)$thunk
Kind       thunk
Context    WORK.TEST_NG.SET_VEC_RANGE()27WORK.TEST_NG.VEC_RANGE_TYPE$thunk
Blocks     1
Registers  36
Types      15
  WORK.TEST_NG-MODEL.INFO_RANGE_TYPE    {-2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1}
Variables  0
Parameters 3
  r0    context                         // P<dummy>
  r1    INFO_RANGE                      // @<WORK.TEST_NG-MODEL.INFO_RANGE_TYPE{}>
  r2    BITS                            // -2^31..2^31-1
Begin
   0: // Dead temp stack mark definition of r3 
      r4 := var upref 1, V_POS          // @<-2^31..2^31-1> => -2^31..2^31-1
      r5 := load indirect r4            // -2^31..2^31-1
      r6 := const -2147483648           // -2^31..2^31-1 => -2^31
      r7 := const 2147483647            // -2^31..2^31-1 => 2^31-1
      r8 := const 0                     // 0..1 => 0
      // Dead debug locus definition of r9 
      // Dead debug locus definition of r10 
      r11 := record ref r1 field 0      // @<-2^31..2^31-1> => -2^31..2^31-1	 <----
      r11 := store indirect r5
      // Dead temp stack mark definition of r12 
      r13 := var upref 1, D_POS         // @<-2^31..2^31-1> => -2^31..2^31-1
      r14 := load indirect r13          // -2^31..2^31-1
      // Dead debug locus definition of r15 
      // Dead debug locus definition of r16 
      r17 := record ref r1 field 1      // @<-2^31..2^31-1> => -2^31..2^31-1
      r17 := store indirect r14
      // Dead temp stack mark definition of r18 
      r19 := load indirect r13          // -2^31..2^31-1
      r20 := add r19 + r2               // -2^31..2^31-1 => -4294967296..4294967294
      r21 := const 1                    // -2^31..2^31-1 => 1
      r22 := sub r20 - r21              // -2^31..2^31-1 => -4294967297..4294967293
      r23 := debug locus WORK.TEST_NG-MODEL+658 // D<>
      r24 := debug locus WORK.TEST_NG-MODEL+616 // D<>
      range check r22 left r6 right r7 dir r8 locus r23 hint r24
      r25 := record ref r1 field 2      // @<-2^31..2^31-1> => -2^31..2^31-1
      r25 := store indirect r22
      // Dead temp stack mark definition of r26 
      r27 := load indirect r4           // -2^31..2^31-1
      r28 := add r27 + r21              // -2^31..2^31-1 => -2147483647..2147483648
      r29 := debug locus WORK.TEST_NG-MODEL+702 // D<>
      r30 := debug locus WORK.TEST_NG-MODEL+684 // D<>
      range check r28 left r6 right r7 dir r8 locus r29 hint r30
      r4 := store indirect r28
      // Dead temp stack mark definition of r31 
      r32 := load indirect r13          // -2^31..2^31-1
      r33 := add r32 + r2               // -2^31..2^31-1 => -4294967296..4294967294
      r34 := debug locus WORK.TEST_NG-MODEL+746 // D<>
      r35 := debug locus WORK.TEST_NG-MODEL+728 // D<>
      range check r33 left r6 right r7 dir r8 locus r34 hint r35
      r13 := store indirect r33
      return 

** Fatal: Expected &(ptr->pointer[-1]) to have kind VALUE_RECORD but is [{-2147483648,-2147483648,-2147483648}]
          (at ../src/exec.c:1631)
[0x7f4c2500e32c] ../src/util.c:690 fatal_trace
[0x7f4c25097d96] ../src/exec.c:248 eval_assert_fail.constprop.15
[0x7f4c25043d18] ../src/exec.c:1631 eval_op_record_ref
[0x7f4c25043d18] ../src/exec.c:2041 eval_vcode.lto_priv.306
[0x7f4c25044439] ../src/exec.c:994 eval_op_fcall
[0x7f4c250418e9] ../src/exec.c:1886 eval_vcode.lto_priv.306
[0x7f4c25044439] ../src/exec.c:994 eval_op_fcall
[0x7f4c250418e9] ../src/exec.c:1886 eval_vcode.lto_priv.306
[0x7f4c2503339a] ../src/exec.c:2438 simp_fold.isra.4
[0x7f4c25033b6c] ../src/simp.c:272 simp_type_conv
[0x7f4c25033b6c] ../src/simp.c:1703 simp_tree.lto_priv.486
[0x7f4c25061fe5] ../src/object.c:635 object_rewrite
[0x7f4c250620ae] ../src/object.c:604 object_rewrite
[0x7f4c2506213a] ../src/object.c:610 object_rewrite
[0x7f4c250620ae] ../src/object.c:604 object_rewrite
[0x7f4c250620ae] ../src/object.c:604 object_rewrite
[0x7f4c250620ae] ../src/object.c:604 object_rewrite
[0x7f4c2506213a] ../src/object.c:610 object_rewrite
[0x7f4c2501c810] ../src/tree.c:1165 tree_rewrite
[0x7f4c250319d6] ../src/simp.c:1818 simplify_global
[0x7f4c25039208] ../src/elab.c:1381 elab_top_level.lto_priv.512
[0x7f4c25019406] ../src/elab.c:1431 elab
[0x7f4c25019406] ../src/elab.c:317 elaborate
[0x7f4c2501573a] ../src/nvc.c:910 process_command
[0x7f4c2500ba2a] ../src/nvc.c:1006 main
```

nickg, 2022-01-14:
This should be fixed now too. Sorry there are so many bugs!

closed 2022-01-14 by nickg

ikwzm, 2022-01-14:
No problem. Thank you.