nickg/nvc #439: "** Fatal: out of space in arena WORK.PUMP_AXI4_TO_AXI4_TEST_BENCH_32_32.elab" occurs when elabolate

ikwzm, 2022-01-17:
I'm currently trying to simulate a VHDL project with nvc.
The issue I recently submitted to you was to simulate this project.
Thanks to your efforts, we have made considerable progress, but we just can't figure out the cause of the following error: If you don't mind, I would like your advice.

The project is included at https://github.com/ikwzm/PipeWorkTest. 
You can download it as follows:

```console
shell$ git clone --branch develop-nvc-debug --recursive https://github.com/ikwzm/PipeWorkTest
Cloning into 'PipeWorkTest'...
remote: Enumerating objects: 2900, done.        
remote: Counting objects: 100% (246/246), done.        
remote: Compressing objects: 100% (159/159), done.        
remote: Total 2900 (delta 117), reused 181 (delta 63), pack-reused 2654        
Receiving objects: 100% (2900/2900), 3.50 MiB | 17.16 MiB/s, done.
Resolving deltas: 100% (1570/1570), done.
Submodule 'Dummy_Plug' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug'
Submodule 'PipeWork' (https://github.com/ikwzm/PipeWork.git) registered for path 'PipeWork'
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/Dummy_Plug'...
remote: Enumerating objects: 5478, done.        
remote: Counting objects: 100% (118/118), done.        
remote: Compressing objects: 100% (75/75), done.        
remote: Total 5478 (delta 51), reused 90 (delta 32), pack-reused 5360        
Receiving objects: 100% (5478/5478), 6.80 MiB | 6.61 MiB/s, done.
Resolving deltas: 100% (4391/4391), done.
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/PipeWork'...
remote: Enumerating objects: 2718, done.        
remote: Counting objects: 100% (154/154), done.        
remote: Compressing objects: 100% (103/103), done.        
remote: Total 2718 (delta 99), reused 94 (delta 48), pack-reused 2564        
Receiving objects: 100% (2718/2718), 2.14 MiB | 5.90 MiB/s, done.
Resolving deltas: 100% (1993/1993), done.
Submodule path 'Dummy_Plug': checked out '1192b10219033972e418ae270244f72cc4edffbe'
Submodule 'docs' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug/docs'
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/Dummy_Plug/docs'...
remote: Enumerating objects: 5478, done.        
remote: Counting objects: 100% (118/118), done.        
remote: Compressing objects: 100% (75/75), done.        
remote: Total 5478 (delta 51), reused 90 (delta 32), pack-reused 5360        
Receiving objects: 100% (5478/5478), 6.80 MiB | 10.75 MiB/s, done.
Resolving deltas: 100% (4391/4391), done.
Submodule path 'Dummy_Plug/docs': checked out 'a801570f3d1d46f52edd64d02ba9960fef2d5728'
Submodule path 'PipeWork': checked out '854351a8c182c88d35a26eb09f6a530893c48e78'
```

The project is sim/nvc/pump_axi4. 
You can recreate the error by running:

```console
shell$ cd PipeWorkTest/sim/nvc/pump_axi4/
shell$ make
sh analyze_libs.sh
nvc -L ./  --work=work -e pump_axi4_to_axi4_test_bench_32_32
** Fatal: out of space in arena WORK.PUMP_AXI4_TO_AXI4_TEST_BENCH_32_32.elab
[0x7ff2eb40e35c] ../src/util.c:690 fatal_trace
[0x7ff2eb47bf23] ../src/object.c:323 object_new
[0x7ff2eb47c128] ../src/object.c:917 object_copy_mark.lto_priv.399
[0x7ff2eb47c1a0] ../src/object.c:930 object_copy_mark.lto_priv.399
[0x7ff2eb47c1a0] ../src/object.c:930 object_copy_mark.lto_priv.399
[0x7ff2eb47c22a] ../src/object.c:937 object_copy_mark.lto_priv.399
[0x7ff2eb47c1a0] ../src/object.c:930 object_copy_mark.lto_priv.399
[0x7ff2eb47c22a] ../src/object.c:937 object_copy_mark.lto_priv.399
[0x7ff2eb47c1a0] ../src/object.c:930 object_copy_mark.lto_priv.399
[0x7ff2eb47c1a0] ../src/object.c:930 object_copy_mark.lto_priv.399
[0x7ff2eb47c1a0] ../src/object.c:930 object_copy_mark.lto_priv.399
[0x7ff2eb47c22a] ../src/object.c:937 object_copy_mark.lto_priv.399
[0x7ff2eb47c22a] ../src/object.c:937 object_copy_mark.lto_priv.399
[0x7ff2eb47c22a] ../src/object.c:937 object_copy_mark.lto_priv.399
[0x7ff2eb47c22a] ../src/object.c:937 object_copy_mark.lto_priv.399
[0x7ff2eb47c22a] ../src/object.c:937 object_copy_mark.lto_priv.399
[0x7ff2eb47c22a] ../src/object.c:937 object_copy_mark.lto_priv.399
[0x7ff2eb4289ca] ../src/object.c:969 object_copy
[0x7ff2eb4289ca] ../src/object.c:1178 tree_copy
[0x7ff2eb442dbd] ../src/elab.c:290 elab_copy.lto_priv.555
[0x7ff2eb4440a6] ../src/elab.c:984 elab_stmts
[0x7ff2eb443cd8] ../src/elab.c:1007 elab_stmts
[0x7ff2eb443cd8] ../src/elab.c:1007 elab_stmts
[0x7ff2eb444b0e] ../src/elab.c:1385 elab_top_level.lto_priv.553
[0x7ff2eb4194f6] ../src/elab.c:1431 elab
[0x7ff2eb4194f6] ../src/elab.c:317 elaborate
[0x7ff2eb41582a] ../src/nvc.c:910 process_command
[0x7ff2eb40ba5a] ../src/nvc.c:1006 main
Makefile:30: recipe for target 'pump_axi4_to_axi4_test_bench_32_32' failed
make: *** [pump_axi4_to_axi4_test_bench_32_32] Error 1
```

nickg, 2022-01-17:
This happens when the size of the elaborated design hits an internal limit of 16 megabytes. I've improved the error message and also added a `-M` to control the memory limit. If you run `nvc -M 32m ...` when elaborating it works ok.

But I ran into another error when I tried to run the simulation. I guess it's another bug but not sure where.

```
        35 ns| MARCHAL < PUMP_AXI4_TO_AXI4_TEST_32_32 I_DATA_WIDTH=32 O_DATA_WIDTH=32 MAX_XFER_SIZE=64
        55 ns| MARCHAL < PUMP_AXI4_TO_AXI4_TEST_32_32.1.0
##### Failure :     100065 ns (O.AW) EXECUTE_TRANSACTION_SLAVE_WRITE_ADDR WAIT AW Time Out!
   name       : O
   stream_name: pump_axi4_to_axi4_test_bench_32_32.snr(62,1,3)
   curr_state : STATE_BLOCK_MAP_IMPLICIT_END(7,MAPKEY_NULL)
   prev_state : STATE_BLOCK_MAP_IMPLICIT_END(5,MAPKEY_NULL)
   prev_state : STATE_BLOCK_SEQ_END(3,MAPKEY_NULL)
   prev_state : STATE_BLOCK_MAP_IMPLICIT_END(3,MAPKEY_NULL)
   prev_state : STATE_BLOCK_SEQ_END(1,MAPKEY_NULL)
   prev_state : STATE_DOCUMENT(0,MAPKEY_NULL)
   text_line  : ---
               |^  |
** Fatal: 100065ns+1: Assertion Failure: EXECUTE_TRANSACTION_SLAVE_WRITE_ADDR
          WAIT AW Time Out!
        Procedure EXECUTE_ABORT [CORE_TYPE, STRING, STRING]
            File ../../../Dummy_Plug/src/main/vhdl/core/core.vhd, Line 1703
        Procedure WAIT_UNTIL_XFER_AW [CORE_TYPE, STRING, INTEGER]
            File ../../../Dummy_Plug/src/main/vhdl/axi4/axi4_channel_player.vhd,
          Line 806
        Procedure EXECUTE_TRANSACTION_SLAVE_WRITE_ADDR []
            File ../../../Dummy_Plug/src/main/vhdl/axi4/axi4_channel_player.vhd,
          Line 2643
        Process :pump_axi4_to_axi4_test_bench_32_32:tb:o:aw:channel_a:process_a
            File ../../../Dummy_Plug/src/main/vhdl/axi4/axi4_channel_player.vhd,
          Line 2818
```

ikwzm, 2022-01-17:
thank you. I understand about memory control options.

Also, it seems that there is another cause for the problems after that, so I would like to report it again after checking the details here. 
I would like to close this issue.
Thank you very much.

closed 2022-01-17 by ikwzm

nickg, 2022-01-24:
I think I've fixed this now in the commit above, your test can run without errors.

ikwzm, 2022-01-24:
Oh, thank you!!