nickg/nvc #440: 1.6.0 fails to build: Error: unit STD.STANDARD not found in library STD

yurivict, 2022-01-26:
```
===>  Building for nvc-1.6.0
gmake[1]: Entering directory '/disk-samsung/freebsd-ports/cad/nvc/work/nvc-r1.6.0'
gmake[2]: Entering directory '/disk-samsung/freebsd-ports/cad/nvc/work/nvc-r1.6.0'
src/lexer.c:1300:13: warning: misleading indentation; statement is not part of the previous 'if' [-Wmisleading-indentation]
            if ( ! (yy_state_buf) )
            ^
src/lexer.c:1298:9: note: previous statement is here
        if ( ! (yy_state_buf) )
        ^
src/lexer.c:2476:3: warning: misleading indentation; statement is not part of the previous 'if' [-Wmisleading-indentation]
                return yy_is_jam ? 0 : yy_current_state;
                ^
src/lexer.c:2473:2: note: previous statement is here
        if ( ! yy_is_jam )
        ^
2 warnings generated.
** Error: unit STD.STANDARD not found in library STD
gmake[2]: *** [Makefile:3091: lib/nvc.08/NVC.POLYFILL] Error 1
```

clang-12
OS: FreeBSD 13

nickg, 2022-01-27:
I see you're building with GNU make but did you also set `MAKE=gmake` when running configure? There's some extra makefile dependencies that only get enabled if `make` is the GNU variant. There are some alternative rules for BSD make but I just noticed there was one missing dependency, which I've added in the commit above.

Either applying that patch and using BSD make or setting `MAKE=gmake` before configure should fix your issue.

nickg, 2022-02-05:
This is fixed in the updated 1.6.1 release.

yurivict, 2022-02-05:
This failure is still in 1.6.1:
```
** Error: unit STD.STANDARD not found in library STD
```

nickg, 2022-02-06:
OK I managed to reproduce this. The proper fix is the commit above.

@yurivict a simple fix is to change the port makefile like this:

```diff
--- Makefile.orig       2022-02-06 00:23:35.979173000 +0000
+++ Makefile    2022-02-06 00:27:41.269168000 +0000
@@ -1,6 +1,6 @@
 PORTNAME=      nvc
 DISTVERSIONPREFIX=     r
-DISTVERSION=   1.5.3
+DISTVERSION=   1.6.1
 CATEGORIES=    cad

 MAINTAINER=    yuri@FreeBSD.org
@@ -13,7 +13,7 @@
 RUN_DEPENDS=   llvm${LLVM_VERSION}>0:devel/llvm${LLVM_VERSION}
 TEST_DEPENDS=  check>0:devel/check

-USES=          autoreconf compiler:c++14-lang gmake localbase pkgconfig
+USES=          autoreconf compiler:c++14-lang localbase pkgconfig

 USE_GITHUB=    yes
 GH_ACCOUNT=    nickg

```

I tried this on a FreeBSD 13 VM and it works fine. The makefile doesn't depend on GNU Make: the configure script just enables different rules depending depending on whether `$MAKE` is GNU or not.

nickg, 2022-02-06:
Or disable `make -j` as a workaround (`ALLOW_MAKE_JOBS=no`?)

yurivict, 2022-02-06:
This works.

Thank you for the fix!

closed 2022-02-06 by yurivict