nickg/nvc #445: 1 of 2 tests failed on FreeBSD 13

yurivict, 2022-02-06:
```
FAIL: bin/unit_test
[...]
PASS: bin/run_regr
===============================================
1 of 2 tests failed
Please report to Nick Gasson <nick@nickg.me.uk>
===============================================
```

```

$ bin/unit_test
Running suite(s): ident
100%: Checks: 22, Failures: 0, Errors: 0
Running suite(s): misc
100%: Checks: 12, Failures: 0, Errors: 0
Running suite(s): lib
100%: Checks: 3, Failures: 0, Errors: 0
Running suite(s): parse
100%: Checks: 59, Failures: 0, Errors: 0
Running suite(s): sem
100%: Checks: 118, Failures: 0, Errors: 0
Running suite(s): bounds
100%: Checks: 17, Failures: 0, Errors: 0
Running suite(s): simplify
100%: Checks: 38, Failures: 0, Errors: 0
Running suite(s): value
100%: Checks: 3, Failures: 0, Errors: 0
Running suite(s): lower
100%: Checks: 92, Failures: 0, Errors: 0
Running suite(s): elab
100%: Checks: 51, Failures: 0, Errors: 0
Running suite(s): json
100%: Checks: 2, Failures: 0, Errors: 0
Running suite(s): debug
0%: Checks: 1, Failures: 1, Errors: 0
test/test_debug.c:41:F:Core:test_capture:0: Failure 'f0->symbol == NULL' occurred
Running suite(s): eopt
100%: Checks: 27, Failures: 0, Errors: 0
Running suite(s): exec
100%: Checks: 8, Failures: 0, Errors: 0
```

nickg, 2022-02-06:
I also saw this when I tried `make test` in the port tree but I don't get the failure if I run `make check` in a tree I built manually? Do you strip `bin/unit_test` before running the tests? This test is testing the `libdwarf` wrapper for accessing debug symbols so would probably fail if stripped.

yurivict, 2022-02-06:
bin/nvc is stripped, but not the test executable.

nickg, 2022-04-03:
This should be fixed in the 1.6.2 release

closed 2022-04-03 by nickg