nickg/nvc #446: "** Fatal: pointer and stored value do not have same type" occrus when elabolating VHDL.

ikwzm, 2022-02-07:
Sorry to bother you again.

```VHDL:test_ng.vhd
-- test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;
-----------------------------------------------------------------------------------
--
-----------------------------------------------------------------------------------
package IMAGE_TYPES is

    -------------------------------------------------------------------------------
    --
    -------------------------------------------------------------------------------
    type      IMAGE_SHAPE_SIDE_DICIDE_TYPE is (
                  IMAGE_SHAPE_SIDE_DICIDE_CONSTANT ,
                  IMAGE_SHAPE_SIDE_DICIDE_EXTERNAL ,
                  IMAGE_SHAPE_SIDE_DICIDE_AUTO      
    );
    -------------------------------------------------------------------------------
    --
    -------------------------------------------------------------------------------
    type      IMAGE_SHAPE_SIDE_TYPE        is record
                  DICIDE_TYPE              :  IMAGE_SHAPE_SIDE_DICIDE_TYPE;
                  ELEM_IN_DATA             :  boolean;
                  ATRB_IN_DATA             :  boolean;
                  LO                       :  integer;
                  HI                       :  integer;
                  SIZE                     :  integer;
                  MAX_SIZE                 :  integer;
    end record;
    -------------------------------------------------------------------------------
    --
    -------------------------------------------------------------------------------
    function  NEW_IMAGE_SHAPE_SIDE_CONSTANT(SIZE    : integer; ELEM_IN_DATA: boolean := TRUE; ATRB_IN_DATA: boolean := TRUE) return IMAGE_SHAPE_SIDE_TYPE;
    function  NEW_IMAGE_SHAPE_SIDE_CONSTANT(LO,HI   : integer; ELEM_IN_DATA: boolean := TRUE; ATRB_IN_DATA: boolean := TRUE) return IMAGE_SHAPE_SIDE_TYPE;
    -------------------------------------------------------------------------------
    --
    -------------------------------------------------------------------------------
    type      IMAGE_SHAPE_TYPE             is record
                  ELEM_BITS                :  integer;
                  C                        :  IMAGE_SHAPE_SIDE_TYPE;
                  D                        :  IMAGE_SHAPE_SIDE_TYPE;
                  X                        :  IMAGE_SHAPE_SIDE_TYPE;
                  Y                        :  IMAGE_SHAPE_SIDE_TYPE;
    end record;
    -------------------------------------------------------------------------------
    --
    -------------------------------------------------------------------------------
    function  NEW_IMAGE_SHAPE         (ELEM_BITS: integer;
                                       C,D,X,Y  : IMAGE_SHAPE_SIDE_TYPE) return IMAGE_SHAPE_TYPE;
    function  NEW_IMAGE_SHAPE_CONSTANT(ELEM_BITS,    X,Y: integer      ) return IMAGE_SHAPE_TYPE;

    -------------------------------------------------------------------------------
    --
    -------------------------------------------------------------------------------
    type      IMAGE_STREAM_BORDER_TYPE is (
                  IMAGE_STREAM_BORDER_NONE,
                  IMAGE_STREAM_BORDER_CONSTANT,
                  IMAGE_STREAM_BORDER_REPEAT_EDGE
    );
    -------------------------------------------------------------------------------
    --
    -------------------------------------------------------------------------------
    type      IMAGE_STREAM_ATRB_TYPE       is record
                  VALID             :  boolean;
                  START             :  boolean;
                  LAST              :  boolean;
    end record;
[... 1389 lines cut ...]
      r581 := record ref r578 field 2   // @<[*] : 0..8> => 0..8
      r581 := store indirect r528
      r582 := record ref r578 field 3   // @<[*] : 0..8> => 0..8
      r582 := store indirect r577
      r583 := const 0                   // 0..8 => 0
      r584 := const [r583]*14           // @<0..8> => 0..8
      r585 := record ref r2 field 0     // @<[*, *, *, *] : 0..8> => 0..8
      r585 := store indirect r381

** Fatal: pointer and stored value do not have same type
[0x7fc856a1008c] ../src/util.c:690 fatal_trace
[0x7fc856a8ea4b] ../src/vcode.c:3309 emit_store_indirect
[0x7fc856a822f7] ../src/lower.c:2452 lower_record_aggregate.lto_priv.140
[0x7fc856a829a7] ../src/lower.c:2778 lower_aggregate.lto_priv.253
[0x7fc856a60f67] ../src/lower.c:4735 lower_decls
[0x7fc856a5fa80] ../src/lower.c:7006 lower_decls
[0x7fc856a62482] ../src/lower.c:7487 lower_concurrent_block
[0x7fc856a62e33] ../src/lower.c:7519 lower_elab
[0x7fc856a62e33] ../src/lower.c:7584 lower_unit
[0x7fc856a1a7c7] ../src/nvc.c:318 elaborate
[0x7fc856a1741a] ../src/nvc.c:927 process_command
[0x7fc856a0d726] ../src/nvc.c:1026 main
```

I'm sorry for being long.

nickg, 2022-02-07:
Are you using the latest master branch? I think this is the same as a bug I discovered a few days ago and fixed in a6723e922.

ikwzm, 2022-02-07:
Thank you for your quick response. 
As you said, the version of nvc may be out of date.
However, the new version of nvc couldn't be built in my environment.

```console
shell$ cd ~/work/nvc
shell$ git log -n 1
commit 3a39bc5f77cdf99492dabbb40d180cd2c7303820 (HEAD -> master, origin/test, origin/master, origin/HEAD)
Author: Nick Gasson <nick@nickg.me.uk>
Date:   Mon Feb 7 16:11:08 2022 +0800

    Avoid sem crash after cascading error

shell$ mkdir build
shell$ cd build/
shell$ ../configure
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for flex... flex
checking lex output file root... lex.yy
checking lex library... -lfl
checking whether yytext is a pointer... yes
checking for inline... inline
checking for C/C++ restrict keyword... __restrict
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking sys/ptrace.h usability... yes
checking sys/ptrace.h presence... yes
checking for sys/ptrace.h... yes
checking sys/prctl.h usability... yes
checking sys/prctl.h presence... yes
checking for sys/prctl.h... yes
checking for struct stat.st_mtimespec.tv_nsec... no
checking for struct stat.st_mtim.tv_nsec... yes
[... 172 lines cut ...]
  CC       src/rt/rtkern.o
  CC       src/rt/alloc.o
  CC       src/rt/heap.o
  CC       src/rt/pprint.o
  CC       src/rt/cover.o
  CC       src/rt/wave.o
  CC       src/rt/jit.o
  CC       src/rt/ffi.o
  AR       lib/librt.a
  CC       src/nvc.o
  GEN      bin/nvc
/tmp/cc57PGLg.ltrans5.ltrans.o: In function `cgen_op_memset':
/mnt/d/ichiro/work/nvc/build/../src/cgen.c:2282: undefined reference to `LLVMBuildMemSet'
collect2: error: ld returned 1 exit status
Makefile:1710: recipe for target 'bin/nvc' failed
make[1]: *** [bin/nvc] Error 1
Makefile:1424: recipe for target 'all' failed
make: *** [all] Error 2
ichiro@Jabberwock:~/work/nvc/build$ 
```

```console
shell$ llvm-config --version
7.0.0
```

ikwzm, 2022-02-07:
What version of llvm are you using?

nickg, 2022-02-08:
I'm using LLVM 13 but the minimum supported version is supposed to be 6. I just realised the `LLVMBuildMemSet` function was only added in LLVM 8. The commit above should fix compatibility with earlier LLVM releases.

ikwzm, 2022-02-08:
Thank you.

The new version resolves this issue.

closed 2022-02-08 by ikwzm