nickg/nvc #447: "*** Caught signal 11 (SIGSEGV) [address=(nil), ip=0x7f837e96c58d] ***" occurs when

ikwzm, 2022-02-09:
I'm currently trying to simulate a VHDL project with nvc.

The project is included at https://github.com/ikwzm/PipeWorkTest.
You can download it as follows:

```console
shell$ git clone --branch nvc-issue-447 --depth=1 --recursive https://github.com/ikwzm/PipeWorkTest.git
Cloning into 'PipeWorkTest'...
remote: Enumerating objects: 418, done.        
remote: Counting objects: 100% (418/418), done.        
remote: Compressing objects: 100% (267/267), done.        
remote: Total 418 (delta 203), reused 258 (delta 101), pack-reused 0        
Receiving objects: 100% (418/418), 2.81 MiB | 13.39 MiB/s, done.
Resolving deltas: 100% (203/203), done.
Note: checking out '266a7475137fc0616d9005a684fa3bcbefb71b06'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Submodule 'Dummy_Plug' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug'
Submodule 'PipeWork' (https://github.com/ikwzm/PipeWork.git) registered for path 'PipeWork'
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/Dummy_Plug'...
remote: Enumerating objects: 5478, done.        
remote: Counting objects: 100% (118/118), done.        
remote: Compressing objects: 100% (75/75), done.        
remote: Total 5478 (delta 51), reused 90 (delta 32), pack-reused 5360        
Receiving objects: 100% (5478/5478), 6.80 MiB | 14.95 MiB/s, done.
Resolving deltas: 100% (4391/4391), done.
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/PipeWork'...
remote: Enumerating objects: 2718, done.        
remote: Counting objects: 100% (154/154), done.        
remote: Compressing objects: 100% (103/103), done.        
remote: Total 2718 (delta 99), reused 94 (delta 48), pack-reused 2564        
Receiving objects: 100% (2718/2718), 2.14 MiB | 13.85 MiB/s, done.
Resolving deltas: 100% (1993/1993), done.
Submodule path 'Dummy_Plug': checked out '1192b10219033972e418ae270244f72cc4edffbe'
Submodule 'docs' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug/docs'
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/Dummy_Plug/docs'...
remote: Enumerating objects: 5478, done.        
remote: Counting objects: 100% (118/118), done.        
remote: Compressing objects: 100% (75/75), done.        
remote: Total 5478 (delta 51), reused 90 (delta 32), pack-reused 5360        
Receiving objects: 100% (5478/5478), 6.80 MiB | 14.67 MiB/s, done.
Resolving deltas: 100% (4391/4391), done.
Submodule path 'Dummy_Plug/docs': checked out 'a801570f3d1d46f52edd64d02ba9960fef2d5728'
Submodule path 'PipeWork': checked out '854351a8c182c88d35a26eb09f6a530893c48e78'
```

The project is sim/nvc/image_stream_models
You can recreate the error by running:

```console
shell$ cd PipeWorkTest/sim/nvc/image_stream_models/
shell$ make test_8x0x0x2x2
sh analyze_libs.sh
nvc -L ./ -M 32M --work=WORK -e  image_stream_player_test_8x0x0x2x2
ruby ../../../src/test/scenarios/image_stream_models/make_8x0x0x2x2.rb
/mnt/d/ichiro/work/tmp/PipeWorkTest/src/test/scenarios/image_stream_models/image_stream.rb:76: warning: constant ::Fixnum is deprecated
/mnt/d/ichiro/work/tmp/PipeWorkTest/src/test/scenarios/image_stream_models/image_stream.rb:51: warning: constant ::TRUE is deprecated
/mnt/d/ichiro/work/tmp/PipeWorkTest/src/test/scenarios/image_stream_models/image_stream.rb:67: warning: constant ::TRUE is deprecated
/mnt/d/ichiro/work/tmp/PipeWorkTest/src/test/scenarios/image_stream_models/image_stream.rb:68: warning: constant ::TRUE is deprecated
/mnt/d/ichiro/work/tmp/PipeWorkTest/src/test/scenarios/image_stream_models/image_stream.rb:69: warning: constant ::TRUE is deprecated
/mnt/d/ichiro/work/tmp/PipeWorkTest/src/test/scenarios/image_stream_models/image_stream.rb:51: warning: constant ::FALSE is deprecated
nvc -L ./ -M 32M --work=WORK -r  image_stream_player_test_8x0x0x2x2
 PARAM
   ELEM_BITS   = 8
   ATRB_BITS   = 3
   INFO_BITS   = 0
   SHAPE.C SIZE=1 LO=0 HI=0 DE=T DA=F
        .D SIZE=1 LO=0 HI=0 DE=F DA=F
        .X SIZE=2 LO=0 HI=1 DE=T DA=T
        .Y SIZE=2 LO=0 HI=1 DE=T DA=T
   DATA            SIZE=44 LO=0 HI=43
      ELEM_FIELD   SIZE=32 LO=0 HI=31
      ATRB_FIELD   SIZE=12 LO=32 HI=43
      ATRB_FIELD.C SIZE=0 LO=44 HI=44
      ATRB_FIELD.D SIZE=0 LO=44 HI=44
      ATRB_FIELD.X SIZE=6 LO=32 HI=37
      ATRB_FIELD.Y SIZE=6 LO=38 HI=43
      INFO_FIELD   SIZE=0 LO=44 HI=44
        25 ns| MARCHAL < IMAGE_STREAM_PLAYER_TEST ELEM_BITS=8  C=0   D=0   X=2   Y=2  
        35 ns| MARCHAL < IMAGE_STREAM_PLAYER_TEST.1.1   C_SIZE=1   D_SIZE=1   X_SIZE=1   Y_SIZE=1  

*** Caught signal 11 (SIGSEGV) [address=(nil), ip=0x7f837e96c58d] ***

[0x7f8385627d8e] ../src/util.c:912 signal_handler.lto_priv.463
[0x7f8384e3ef1f] (/lib/x86_64-linux-gnu/libc-2.27.so) 
[0x7f837e96c58d] ../../../Dummy_Plug/src/main/vhdl/core/util.vhd:387 DUMMY_PLUG.UTIL.STRING_TO_STD_LOGIC_VECTOR(SVII) [VHDL]
                    case STR(pos+1) is
-->                     when 'X'|'x'=> STRING_TO_HEX(STR(pos+2 to STR'high), VAL, len, VAL_LEN);
                        when 'H'|'h'=> STRING_TO_HEX(STR(pos+2 to STR'high), VAL, len, VAL_LEN);
[0x7f837e97c3d6] ../../../src/test/vhdl/image_stream_models/image_stream_player.vhd:548 WORK.IMAGE_STREAM_PLAYER_TEST_8X0X0X2X2.TB.S.T.line_459.READ_VALUE(SV) [VHDL]
                READ_EVENT(core, stream, EVENT_SCALAR);
-->             STRING_TO_STD_LOGIC_VECTOR(
                    STR     => core.str_buf(1 to core.str_len),
[0x7f837e972c08] ../../../src/test/vhdl/image_stream_models/image_stream_player.vhd:647 WORK.IMAGE_STREAM_PLAYER_TEST_8X0X0X2X2.TB.S.T.line_459.READ_IMAGE_STREAM_ELEM(S71WORK.IMAGE_STREAM_PLAYER_TEST_8X0X0X2X2.TB.S.T.IMAGE_STREAM_SIGNAL_TYPE) [VHDL]
                            else
-->                             read_value(proc_name, signals.ELEM(seq_pos(SEQ_PARAM.Y), seq_pos(SEQ_PARAM.X), seq_pos(SEQ_PARAM.D), seq_pos(SEQ_PARAM.C)));
                                SET_ELEMENT_TO_IMAGE_STREAM_DATA(
[0x7f837e9630c2] ../../../src/test/vhdl/image_stream_models/image_stream_player.vhd:939 WORK.IMAGE_STREAM_PLAYER_TEST_8X0X0X2X2.TB.S.T.line_459.MAP_READ_IMAGE_STREAM_SIGNALS(71WORK.IMAGE_STREAM_PLAYER_TEST_8X0X0X2X2.TB.S.T.IMAGE_STREAM_SIGNAL_TYPE28DUMMY_PLUG.READER.EVENT_TYPE) [VHDL]
                                when KEY_ELEM  =>
-->                                 read_image_stream_elem(proc_name, signals);
                                when KEY_ATRB =>
[0x7f837e94a4b5] ../../../src/test/vhdl/image_stream_models/image_stream_player.vhd:985 WORK.IMAGE_STREAM_PLAYER_TEST_8X0X0X2X2.TB.S.T.line_459.EXECUTE_CHECK [VHDL]
                            );
-->                         map_read_image_stream_signals(
                                signals    => chk_signals     ,  -- I/O:
[0x7f837e939b0a] ../../../src/test/vhdl/image_stream_models/image_stream_player.vhd:1288 WORK.IMAGE_STREAM_PLAYER_TEST_8X0X0X2X2.TB.S.T.line_459 [VHDL]
                            when KEY_WAIT   => execute_wait;
-->                         when KEY_CHECK  => execute_check;
                            when others     => EXECUTE_UNDEFINED_MAP_KEY(core, stream, keyword);
[0x7f838567ffd7] ../src/rt/rtkern.c:2907 rt_resume.lto_priv.507
[0x7f83856882b5] ../src/nvc.c:3062 run
[0x7f83856882b5] ../src/rt/rtkern.c:935 process_command
[0x7f838560d01e] ../src/nvc.c:1032 main
Makefile:62: recipe for target 'test_8x0x0x2x2' failed
make: *** [test_8x0x0x2x2] Error 2
```
I am always grateful for your help.

nickg, 2022-02-10:
Thanks, I can reproduce this.

nickg, 2022-02-10:
Should be fixed now.

closed 2022-02-10 by nickg