nickg/nvc #450: ”** Fatal: accessed 96 bytes beyond process temp stack region" occurs when running VHDL.

ikwzm, 2022-02-11:
I'm currently trying to simulate a VHDL project with nvc.

The project is included at https://github.com/ikwzm/PipeWorkTest.
You can download it as follows:

```console
shell$ git clone --branch nvc-issue-449 --depth=1 --recursive https://github.com/ikwzm/PipeWorkTest.git
Cloning into 'PipeWorkTest'...
remote: Enumerating objects: 426, done.        
remote: Counting objects: 100% (426/426), done.        
remote: Compressing objects: 100% (273/273), done.        
remote: Total 426 (delta 212), reused 260 (delta 103), pack-reused 0        
Receiving objects: 100% (426/426), 3.10 MiB | 15.72 MiB/s, done.
Resolving deltas: 100% (212/212), done.
Note: checking out '68937a262530045b6cbef997753f2779ee6cc3e7'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Submodule 'Dummy_Plug' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug'
Submodule 'PipeWork' (https://github.com/ikwzm/PipeWork.git) registered for path 'PipeWork'
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/Dummy_Plug'...
remote: Enumerating objects: 5478, done.        
remote: Counting objects: 100% (118/118), done.        
remote: Compressing objects: 100% (75/75), done.        
remote: Total 5478 (delta 51), reused 90 (delta 32), pack-reused 5360        
Receiving objects: 100% (5478/5478), 6.80 MiB | 8.34 MiB/s, done.
Resolving deltas: 100% (4391/4391), done.
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/PipeWork'...
remote: Enumerating objects: 2718, done.        
remote: Counting objects: 100% (154/154), done.        
remote: Compressing objects: 100% (103/103), done.        
remote: Total 2718 (delta 99), reused 94 (delta 48), pack-reused 2564        
Receiving objects: 100% (2718/2718), 2.14 MiB | 7.04 MiB/s, done.
Resolving deltas: 100% (1993/1993), done.
Submodule path 'Dummy_Plug': checked out '1192b10219033972e418ae270244f72cc4edffbe'
Submodule 'docs' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug/docs'
Cloning into '/mnt/d/ichiro/work/tmp/PipeWorkTest/Dummy_Plug/docs'...
remote: Enumerating objects: 5478, done.        
remote: Counting objects: 100% (118/118), done.        
remote: Compressing objects: 100% (75/75), done.        
remote: Total 5478 (delta 51), reused 90 (delta 32), pack-reused 5360        
Receiving objects: 100% (5478/5478), 6.80 MiB | 12.31 MiB/s, done.
Resolving deltas: 100% (4391/4391), done.
Submodule path 'Dummy_Plug/docs': checked out 'a801570f3d1d46f52edd64d02ba9960fef2d5728'
Submodule path 'PipeWork': checked out '854351a8c182c88d35a26eb09f6a530893c48e78'
```

The project is sim/nvc/image_stream_buffer
You can recreate the error by running:

```console
shell$ cd PipeWorkTest/sim/nvc/image_stream_buffer/
shell$ make test_0_2_32x1x1_32x4x3x3
sh analyze_libs.sh
nvc -L ./ -M 64M --work=WORK -e  image_stream_buffer_test_0_2_32x1x1_32x4x3x3
cp ../../../src/test/scenarios/image_stream_buffer/test_0_2_32x1x1_32x4x3x3.snr  test_0_2_32x1x1_32x4x3x3.snr
nvc -L ./ -M 64M --work=WORK -r  image_stream_buffer_test_0_2_32x1x1_32x4x3x3
        25 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST ELEM_BITS=2 CHANNEL_SIZE=0 I.C=32 I.X=1 I.Y=1 O.C=32 O.D=4 O.X=3 O.Y=3
        35 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.1 C.SIZE=32 X.SIZE=1 Y.SIZE=1 
       155 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.2 C.SIZE=64 X.SIZE=1 Y.SIZE=1 
       305 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.3 C.SIZE=64 X.SIZE=2 Y.SIZE=2 
** Fatal: accessed 96 bytes beyond process temp stack region
[0x7fa93fa1bdcc] ../src/util.c:673 fatal_trace
[0x7fa93fa1c359] ../src/util.c:787 check_guard_page
[0x7fa93fa1c359] ../src/util.c:895 signal_handler.lto_priv.465
[0x7fa93f23ef1f] (/lib/x86_64-linux-gnu/libc-2.27.so) 
[0x7fa93f2bb71a] (/lib/x86_64-linux-gnu/libc-2.27.so) 
[0x7fa937e72f16] ../../../PipeWork/src/image/image_types.vhd:1129 PIPEWORK.IMAGE_TYPES.NEW_IMAGE_STREAM_PARAM(II37PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_TYPE51PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_STRIDE_PARAM_TYPE45PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_BORDER_TYPE)44PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_PARAM_TYPE [VHDL]
        is
-->         variable  param             :  IMAGE_STREAM_PARAM_TYPE;
        begin
[0x7fa937e9be21] ../../../src/test/vhdl/image_stream_models/image_stream_player.vhd:467 WORK.IMAGE_STREAM_BUFFER_TEST_0_2_32X1X1_32X4X3X3.TB.O.T.MATCH_IMAGE_STREAM_SIGNALS(25DUMMY_PLUG.CORE.CORE_TYPE81WORK.IMAGE_STREAM_BUFFER_TEST_0_2_32X1X1_32X4X3X3.TB.O.T.IMAGE_STREAM_SIGNAL_TYPEB) [VHDL]
                          Y         : integer;
-->                       X         : integer;
                          D         : integer;
[0x7fa937e806be] ../../../src/test/vhdl/image_stream_models/image_stream_player.vhd:1007 WORK.IMAGE_STREAM_BUFFER_TEST_0_2_32X1X1_32X4X3X3.TB.O.T.line_459.EXECUTE_CHECK [VHDL]
                        end loop;
-->                     match_image_stream_signals(core, chk_signals, match);
                        MATCH_GPI(core, gpi_signals, GPI, match);
[0x7fa937e6849f] ../../../src/test/vhdl/image_stream_models/image_stream_player.vhd:1288 WORK.IMAGE_STREAM_BUFFER_TEST_0_2_32X1X1_32X4X3X3.TB.O.T.line_459 [VHDL]
                            when KEY_WAIT   => execute_wait;
-->                         when KEY_CHECK  => execute_check;
                            when others     => EXECUTE_UNDEFINED_MAP_KEY(core, stream, keyword);
[0x7fa93fa809a7] ../src/rt/rtkern.c:2906 rt_resume.lto_priv.510
[0x7fa93fa88d16] ../src/nvc.c:3049 run
[0x7fa93fa88d16] ../src/rt/rtkern.c:935 process_command
[0x7fa93fa0d23e] ../src/nvc.c:1032 main
Makefile:61: recipe for target 'test_0_2_32x1x1_32x4x3x3' failed
make: *** [test_0_2_32x1x1_32x4x3x3] Error 1
```

nickg, 2022-02-13:
The error above is fixed now but I get some failures running this test, not sure if it's expected or not:

```
  ***
  ***  ERROR REPORT test_0_2_32x1x1_32x4x3x3
  ***
  ***  [ INTAKE ]
  ***    Error    : 0
  ***    Mismatch : 0
  ***    Warning  : 0
  ***
  ***  [ OUTLET ]
  ***    Error    : 0
  ***    Mismatch : 462208
  ***    Warning  : 0
  ***
** Fatal: 50466ns+0: Assertion Failure: Simulation complete(mismatch).
        Process :image_stream_buffer_test_0_2_32x1x1_32x4x3x3:tb:line_246
            File ../../../src/test/vhdl/image_stream_buffer/image_stream_buffer_test_bench.vhd, Line 268
```

ikwzm, 2022-04-03:
I tried nvc 1.6.2 and got the expected results. Thank you very much.

```
nvc -L ./ -M 64M --work=WORK -r  image_stream_buffer_test_0_2_32x1x1_32x4x3x3
        25 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST ELEM_BITS=2 CHANNEL_SIZE=0 I.C=32 I.X=1 I.Y=1 O.C=32 O.D=4 O.X=3 O.Y=3
        35 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.1 C.SIZE=32 X.SIZE=1 Y.SIZE=1 
       155 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.2 C.SIZE=64 X.SIZE=1 Y.SIZE=1 
       305 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.3 C.SIZE=64 X.SIZE=2 Y.SIZE=2 
       525 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.4 C.SIZE=64 X.SIZE=3 Y.SIZE=3 
       855 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.5 C.SIZE=64 X.SIZE=4 Y.SIZE=4 
      1415 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.6 C.SIZE=64 X.SIZE=5 Y.SIZE=5 
      2285 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.7 C.SIZE=64 X.SIZE=10 Y.SIZE=7 
      4795 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.8 C.SIZE=128 X.SIZE=15 Y.SIZE=11 
     16465 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.2.1 C.SIZE=32 X.SIZE=4 Y.SIZE=8 
     17605 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.2.2 C.SIZE=64 X.SIZE=4 Y.SIZE=8 
     19485 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.2.3 C.SIZE=128 X.SIZE=15 Y.SIZE=11 
  ***  
  ***  ERROR REPORT test_0_2_32x1x1_32x4x3x3
  ***  
  ***  [ INTAKE ]
  ***    Error    : 0
  ***    Mismatch : 0
  ***    Warning  : 0
  ***  
  ***  [ OUTLET ]
  ***    Error    : 0
  ***    Mismatch : 0
  ***    Warning  : 0
  ***  
** Note: 50466ns+0: Assertion Note: Simulation complete(success).
	File ../../../src/test/vhdl/image_stream_buffer/image_stream_buffer_test_bench.vhd, Line 274
```

nickg, 2022-04-03:
> I tried nvc 1.6.2 and got the expected results. Thank you very much.
> 

Sorry I didn't find time to look at this directly, but I guess it had the same cause as some other bug I fixed recently. You might want to try the master branch too which has some performance improvements.

ikwzm, 2022-04-03:
Sorry. I tried the master branch instead of nvc release 1.6.2. In case of r1.6.2, I got the following error.

```console
shell$ nvc --version
nvc 1.6.2 (Using LLVM 7.0.0)
Copyright (C) 2011-2022  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
shell$ nvc -L ./ -M 64M --work=WORK -e  image_stream_buffer_test_4_8_1x1x1_1x1x1x1

Name       WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.INIT_IMAGE_STREAM_SIGNAL(9STD_LOGIC)79WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.IMAGE_STREAM_SIGNAL_TYPE
Kind       function
Context    WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T
Blocks     17
Registers  604
Types      55
  WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.IMAGE_STREAM_SIGNAL_TYPE {[*, *, *, *] : 0..8, WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.IMAGE_ATRB_SIGNAL_STREAM{}, [1] : 0..8, [17] : 0..8, 0..8, 0..8}
  WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.IMAGE_ATRB_SIGNAL_STREAM {[*] : 0..8, [*] : 0..8, [*] : 0..8, [*] : 0..8}
  PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_PARAM_TYPE {-2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1, PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_STRIDE_PARAM_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_DATA_FIELD_TYPE{}, 0..2}
  PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_TYPE {-2^31..2^31-1, PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_SIDE_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_SIDE_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_SIDE_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_SIDE_TYPE{}}
  PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_SIDE_TYPE {0..2, 0..1, 0..1, -2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1}
  PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_STRIDE_PARAM_TYPE {-2^31..2^31-1, -2^31..2^31-1}
  PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_DATA_FIELD_TYPE {-2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1, PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_DATA_ELEM_FIELD_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_VECTOR_RANGE_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_DATA_ATRB_FIELD_TYPE{}}
  PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_DATA_ELEM_FIELD_TYPE {-2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1}
  PIPEWORK.IMAGE_TYPES.IMAGE_VECTOR_RANGE_TYPE {-2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1}
  PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_DATA_ATRB_FIELD_TYPE {-2^31..2^31-1, -2^31..2^31-1, -2^31..2^31-1, PIPEWORK.IMAGE_TYPES.IMAGE_VECTOR_RANGE_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_VECTOR_RANGE_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_VECTOR_RANGE_TYPE{}, PIPEWORK.IMAGE_TYPES.IMAGE_VECTOR_RANGE_TYPE{}}
Variables  6
  WIN_SIGNAL                            // WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.IMAGE_STREAM_SIGNAL_TYPE{}
  i246                                  // #, temp
  i247                                  // #, temp
  i248                                  // #, temp
  i249                                  // #, temp
  record                                // WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.IMAGE_ATRB_SIGNAL_STREAM{}, temp
Result     @<WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.IMAGE_STREAM_SIGNAL_TYPE{}>
Parameters 2
  r0    context                         // P<WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T>
  r1    D                               // 0..8
Begin
   0: r2 := index WIN_SIGNAL            // @<WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1.TB.I.T.IMAGE_STREAM_SIGNAL_TYPE{}>
      r3 := const 0                     // 0..1 => 0
      r4 := link package PIPEWORK.IMAGE_TYPES // P<PIPEWORK.IMAGE_TYPES>
      r5 := const 8                     // -2^31..2^31-1 => 8
      r6 := const -2147483648           // -2^31..2^31-1 => -2^31
      r7 := const 2147483647            // -2^31..2^31-1 => 2^31-1
      r8 := debug locus WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1+411 // D<>
      r9 := debug locus PIPEWORK.IMAGE_TYPES+2853 // D<>
      r10 := const 1                    // -2^31..2^31-1 => 1
      r11 := debug locus WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1+425 // D<>
      r12 := debug locus PIPEWORK.IMAGE_TYPES+2860 // D<>
      r13 := debug locus WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1+439 // D<>
      r14 := debug locus PIPEWORK.IMAGE_TYPES+2867 // D<>
      r15 := debug locus WORK.IMAGE_STREAM_BUFFER_TEST_4_8_1X1X1_1X1X1X1+453 // D<>
      r16 := debug locus PIPEWORK.IMAGE_TYPES+2874 // D<>
      r17 := fcall PIPEWORK.IMAGE_TYPES.NEW_IMAGE_STREAM_PARAM(IIII)44PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_PARAM_TYPE r4, r5, r10, r10, r10 // @<PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_PARAM_TYPE{}> => PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_PARAM_TYPE{}
      r18 := record ref r17 field 3     // @<PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_TYPE{}> => PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_TYPE{}
      r19 := record ref r18 field 4     // @<PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_SIDE_TYPE{}> => PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_SIDE_TYPE{}
      r20 := record ref r19 field 3     // @<-2^31..2^31-1> => -2^31..2^31-1
      r21 := load indirect r20          // -2^31..2^31-1
      r22 := fcall PIPEWORK.IMAGE_TYPES.NEW_IMAGE_STREAM_PARAM(IIII)44PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_PARAM_TYPE r4, r5, r10, r10, r10 // @<PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_PARAM_TYPE{}> => PIPEWORK.IMAGE_TYPES.IMAGE_STREAM_PARAM_TYPE{}
      r23 := record ref r22 field 3     // @<PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_TYPE{}> => PIPEWORK.IMAGE_TYPES.IMAGE_SHAPE_TYPE{}
[... cut ...]
        25 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST ELEM_BITS=2 CHANNEL_SIZE=0 I.C=32 I.X=1 I.Y=1 O.C=32 O.D=4 O.X=3 O.Y=3
        35 ns| MARCHAL < IMAGE_STREAM_BUFFER_TEST.1.1 C.SIZE=32 X.SIZE=8 Y.SIZE=8 
  ***  
  ***  ERROR REPORT test_0_2_32x1x1_32x4x3x3_bug1
  ***  
  ***  [ INTAKE ]
  ***    Error    : 0
  ***    Mismatch : 0
  ***    Warning  : 0
  ***  
  ***  [ OUTLET ]
  ***    Error    : 0
  ***    Mismatch : 0
  ***    Warning  : 0
  ***  
** Note: 4196ns+0: Assertion Note: Simulation complete(success).
	File ../../../src/test/vhdl/image_stream_buffer/image_stream_buffer_test_bench.vhd, Line 274

```

nickg, 2022-05-09:
@ikwzm I recently fixed some bugs in the implementation of inout ports in bf01812fe6ca but unfortunately it caused a regression in your tests:

```
       415 ns|       N < AXI4_ADAPTER_TEST_4096_32_32.1.4     READ  ADDR=0x00000001, SIZE=1  , DATA_SIZE=1
##### Failure :     100485 ns (M.R) EXECUTE_WAIT Time Out!
   name       : M
   stream_name: src/test/scenarios/axi4_adapter/axi4_adapter_test_bench_4096_32_32.snr(278,35,35)
   curr_state : STATE_FLOW_MAP_END(15,MAPKEY_NULL)
   prev_state : STATE_BLOCK_MAP_IMPLICIT_END(7,MAPKEY_NULL)
   prev_state : STATE_BLOCK_SEQ_END(5,MAPKEY_NULL)
   prev_state : STATE_BLOCK_MAP_IMPLICIT_END(5,MAPKEY_NULL)
   prev_state : STATE_BLOCK_SEQ_END(3,MAPKEY_NULL)
   prev_state : STATE_BLOCK_SEQ_END(1,MAPKEY_NULL)
   prev_state : STATE_DOCUMENT(0,MAPKEY_NULL)
   text_line  :     - WAIT  : {AVALID : 1, ON : on}
               |                                  ^|
** Fatal: 100485ns+1: Assertion Failure: EXECUTE_WAIT Time Out!
   Procedure EXECUTE_ABORT [CORE_TYPE, STRING, STRING]
        File ../dummyplug/src/main/vhdl/core/core.vhd, Line 1702
   Procedure EXECUTE_WAIT [CORE_TYPE, TEXT, BOOLEAN]
        File ../dummyplug/src/main/vhdl/axi4/axi4_channel_player.vhd, Line 2140
   Procedure EXECUTE_CHANNEL_MAP_OPERATION []
        File ../dummyplug/src/main/vhdl/axi4/axi4_channel_player.vhd, Line 3286
   Procedure EXECUTE_CHANNEL_OPERATION []
        File ../dummyplug/src/main/vhdl/axi4/axi4_channel_player.vhd, Line 3319
   Process :axi4_adapter_test_bench_4096_32_32:tb:m:r:channel_d:process_d
        File ../dummyplug/src/main/vhdl/axi4/axi4_channel_player.vhd, Line 3368
```

I tried to debug it but I can't figure out what's wrong. Do you have any ideas? Thanks!

ikwzm, 2022-05-10:
I am always grateful for your help

This happens when waiting for the ARVALID input port of AXI4_CHANNEL_PLAYTER to be '1', but not within the specified time.

This signal is driven by the ARVALID output port of DUT: AXI4_ADAPTER and reaches the ARVALID input port of R: AXI4_CHANNEL_PLAYER via the ARVALID input port of M: AXI4_SLAVE_PLAYER.

Maybe you couldn't recognize that the ARVALID output port on the DUT originally did not output '1', or the ARVALID input port on R became '1'?

nickg, 2022-05-15:
Thanks, I've fixed it now!

commit e1d8fb66 (2022-02-13): Eagerly fold all constants in generic map. Fixes #450