nickg/nvc #451: Simulation error on two dimensional array with hugh size

taneroksuz, 2022-02-20:
I get following error, when I test my design with very large arrays (**e.g. 131072x64 bits**):

```
** Fatal: accessed 32 bytes beyond global temp stack region
[0x55efa5e5a965] ../src/util.c:671 fatal_trace
[0x55efa5e5bd4b] ../src/loc.c:785 check_guard_page
[0x55efa5e5bd4b] ../src/util.c:893 signal_handler
[0x7f702dadf20f] (/usr/lib/x86_64-linux-gnu/libc-2.31.so) 
[0x7f702dc27b10] (/usr/lib/x86_64-linux-gnu/libc-2.31.so) 
[0x7f7028c50b1f] ../lib/ieee.08/std_logic_1164-body.vhdl:1509 IEEE.STD_LOGIC_1164.HREAD(15STD.TEXTIO.LINEY) [VHDL]
[0x7f7028c4e506] /home/taner/Project/riscv-w7/vhdl/tb/bram.vhd:43 WORK.SOC.BRAM_COMP.INIT_MEMORY(S)30WORK.BRAM-BEHAVIOR.MEMORY_TYPE [VHDL]
                        readline(memory_file, memory_line);
-->                     hread(memory_line, memory_block(i));
                end loop;
[0x7f7028c22332] /home/taner/Project/riscv-w7/vhdl/tb/soc.vhd:48 WORK.SOC.BRAM_COMP_reset [VHDL]
                generic(
-->                     bram_depth   : integer := bram_depth;
                        bram_latency : integer := bram_latency
[0x55efa5e6058c] ../src/enode.c:2288 run
[0x55efa5e6058c] ../src/rt/rtkern.c:935 process_command
[0x55efa5dcb1ed] ../src/nvc.c:1032 main
```

However I don't understand why I am getting this error. I test all my design on **WSL2** (Windows Subsystem for Linux 2). 
I get similar errors with other open source VHDL and Verilog similator but not commercial simulators. So I don't know what is the main reason of this error. I also disabled the upper limits of stack size in other open source simulator, but I encounter still the same error.

nickg, 2022-02-22:
> However I don't understand why I am getting this error. I test all my design on **WSL2** (Windows Subsystem for Linux 2). I get similar errors with other open source VHDL and Verilog similator but [...]

NVC uses a "secondary stack" to return large or dynamically sized objects like arrays from functions. I suspect GHDL does something similar as it's a standard technique for implementing Ada-like languages. That stack is fixed sized and is insufficient here as `init_memory` allocates and returns an ~8MB array (each `std_logic` element occupies one byte).

I've made the stack size configurable and also improved the error message so now it prints:

```
** Fatal: global secondary stack exhausted
        Procedure HREAD [LINE, STD_ULOGIC_VECTOR]
            File ../../lib/ieee.08/std_logic_1164-body.vhdl, Line 1509
        Function INIT_MEMORY [STRING return MEMORY_TYPE]
            File /home/nick/src/riscv-w7/vhdl/tb/bram.vhd, Line 43
** Note: the current limit is 8388608 bytes which you can increase with the
          -G option, for example -G 16384k
```

`nvc -G 16m ...` works ok for me. BTW I saw elaboration is very slow with your project: this is a known issue I hope to resolve at some point.

closed 2022-02-22 by nickg

nickg, 2022-03-24:
@taneroksuz are you still using nvc? The latest master branch has some performance improvements that should help with your project.