nickg/nvc #453: LSP Server

pidgeon777, 2022-03-15:
First of all, congratulation on the project.

It would be great for `nvc` to also act as an LSP server.

Any plan for the future?

nickg, 2022-03-15:
I've thought about this before and I think it would be a useful feature. But right now I'm focussing on improving VHDL-2008 support and maybe basic Verilog after that.

pidgeon777, 2022-03-15:
Hopefully, it will be implemented one day.

It would be an extremely useful feature to have, especially in modern IDEs (for example, Visual Studio Code, Neovim...).

nickg, 2024-07-24:
Going to convert this to a discussion as I don't have plans to add this any time soon and a language server requires quite a different type of parser that can cope gracefully with partial input. In the meantime there is [rust_hdl](https://github.com/VHDL-LS/rust_hdl) which is a dedicated VHDL language server.