nickg/nvc #459: Vector length attribute not considered static for generate condition

avelure, 2022-05-10:
I have some generate statements that base the condition on the length attribute of a vector. I believe this should be considered a static value, at least it is in ghdl and mentor.

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
  port(
    sig1 : std_logic_vector
  );
end entity test;
architecture beh of test is
begin
  g_gen : if sig1'length = 10 generate
    process
    begin
      wait;
    end process;
  end generate;
end architecture beh;
```
```
nvc --std=08 -a test.vhd
```
```
** Error: condition of generate statement must be static
        File test_std_logic.vhd, Line 13
    |
 13 |   g_gen : if sig1'length = 10 generate
```

avelure, 2022-05-10:
The same should applies for the range attribute as I found in another file.

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
  port(
    sig1 : std_logic_vector
  );
end entity test;
architecture beh of test is
begin
  g_gen : for i in sig1'range generate
    process
    begin
      wait;
    end process;
  end generate;
end architecture beh;
```
```
** Error: range of generate statement must be static
        File test_std_logic.vhd, Line 13
    |
 13 |   g_gen : for i in sig1'range generate
    |                    ^^^^^^^^^^
```

nickg, 2022-05-10:
Yes this is a bug.

nickg, 2022-05-11:
I think this is actually invalid in -1993 (at least, I can't figure out why it isn't from the rules for globally static expressions). However Aldec and GHDL accept it which makes me a bit suspicious. 

Anyway it definitely is valid in -2008 where the LRM text was changed to:

> A prefix is appropriate for a globally static attribute if it denotes a signal, a constant, a type or subtype, a
globally static function call, a variable that is not of an access type, or a variable of an access type whose
designated subtype is fully constrained.

Should be fixed by the commit above.

closed 2022-05-11 by nickg