nickg/nvc #464: Ambiguous use of enumeration literal

avelure, 2022-06-04:
The following code

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
  type t_alert_level is (NO_ALERT, NOTE, TB_NOTE, WARNING, TB_WARNING, MANUAL_CHECK, ERROR, TB_ERROR, FAILURE, TB_FAILURE);
  function f_test
    return boolean
  is
  begin
    for i in NOTE to t_alert_level'right loop
      null;
    end loop;
    return false;
  end function;
begin
end architecture beh;
```
gives
```
** Error: ambiguous use of enumeration literal NOTE (T_ALERT_LEVEL, SEVERITY_LEVEL)
        File test_visible.vhd, Line 13
    |
  8 |   type t_alert_level is (NO_ALERT, NOTE, TB_NOTE, WARNING, TB_WARNING, MANUAL_CHECK, ERROR, TB_ERROR, FAILURE, TB_FAILURE);
    |                                    ^^^^ visible declaration of NOTE as T_ALERT_LEVEL
 ...
 13 |     for i in NOTE to t_alert_level'right loop
    |              ^^^^ use of name NOTE here
    |
    = Note: visible declaration of NOTE as SEVERITY_LEVEL from STD.STANDARD
```
However I get no error in ghdl or modelsim, even set to 93 standard.

nickg, 2022-06-04:
Yes it's a bug, it shouldn't be ambiguous.

nickg, 2022-06-05:
Should be fixed now.

closed 2022-06-05 by nickg