nickg/nvc #465: EXCEPTION_ACCESS_VIOLATION

avelure, 2022-06-04:
I caught this while compiling a library file for 2008 from latest master. I can try to make an MWE, but maybe the backtrace is enough.

```
*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007ff681b5d500] ***

[00007ff681b366ce]
[00007ff84f81ff27] UnhandledExceptionFilter+0x1e7
[00007ff84f25f0b8]
[00007ff84f25eea1]
[00007ff851ff51b0] memset+0x13b0
[00007ff851fdc766] _C_specific_handler+0x96
[00007ff851ff209f] _chkstk+0x11f
[00007ff851fa1454] RtlRaiseException+0x434
[00007ff851ff0bce] KiUserExceptionDispatcher+0x2e
[00007ff681b5d500]
[00007ff681b637d9]
[00007ff681b64128]
[00007ff681b6a633]
[00007ff681b4ba9e]
[00007ff681b51daf]
[00007ff681b5256b]
[00007ff681b53bb9]
[00007ff681b59709]
[00007ff681b59e12]
[00007ff681b3286c]
[00007ff681c0aaea] vhpi_put_data+0x1372a
[00007ff681b313b1]
```

```
Thread 1 received signal SIGSEGV, Segmentation fault.
tree_kind (t=t@entry=0x0) at ../src/tree.c:526
526        return t->object.kind;
(gdb) bt
#0  tree_kind (t=t@entry=0x0) at ../src/tree.c:526
#1  0x00007ff681b637d9 in sem_check_lvalue (t=t@entry=0x25742838f00)
    at ../src/sem.c:1844
#2  0x00007ff681b64128 in sem_check_variable_target (
    target=target@entry=0x25742838f00) at ../src/sem.c:1903
#3  0x00007ff681b6a633 in sem_check_var_assign (t=0x25742839000,
    tab=0x25741a01a50) at ../src/sem.c:1997
#4  0x00007ff681b4ba9e in p_variable_assignment_statement (
    label=label@entry=0x0, name=name@entry=0x25742838f00)
    at ../src/parse.c:8038
#5  0x00007ff681b51daf in p_sequential_statement () at ../src/parse.c:8815
#6  p_sequence_of_statements (parent=parent@entry=0x257428369b0)
    at ../src/parse.c:6726
#7  0x00007ff681b5256b in p_subprogram_body (spec=spec@entry=0x257428369b0)
    at ../src/parse.c:6770
#8  0x00007ff681b53bb9 in p_protected_type_body_declarative_item (
    body=0x25742830110) at ../src/parse.c:6393
#9  p_protected_type_body_declarative_part (body=0x25742830110)
    at ../src/parse.c:6430
#10 p_protected_type_body (id=0x257419fc980) at ../src/parse.c:6468
#11 p_type_declaration (container=container@entry=0x25742830000)
    at ../src/parse.c:5688
#12 0x00007ff681b59709 in p_package_body_declarative_item (
    parent=0x25742830000) at ../src/parse.c:9571
#13 p_package_body_declarative_part (unit=0x25742830000)
    at ../src/parse.c:9610
#14 p_package_body (unit=0x25742830000) at ../src/parse.c:9654
#15 p_secondary_unit (unit=0x25742830000) at ../src/parse.c:9683
#16 0x00007ff681b59e12 in p_library_unit (unit=0x25742830000)
    at ../src/parse.c:9712
#17 p_design_unit () at ../src/parse.c:9766
#18 parse () at ../src/parse.c:9783
#19 0x00007ff681b3286c in analyse (argv=0x2573c921730, argc=3)
    at ../src/nvc.c:182
#20 process_command (argc=argc@entry=3, argv=argv@entry=0x2573c921730)
    at ../src/nvc.c:961
#21 0x00007ff681c0aaea in main (argc=3, argv=0x2573c921730)
    at ../src/nvc.c:1072
```

nickg, 2022-06-05:
Can you try this in gdb?

```
frame 1
call fmt_loc(stdout, tree_loc(t@entry))
```

That should show where the error is. Presumably a variable assignment within a protected type body.

avelure, 2022-06-07:
It was in a procedure in a protected body in a package with an assignment to an alias to a shared variable in another package. It seems to be caused by a required package not having been compiled where the shared variable was.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.testtest_pkg.all;

package test2_pkg is
  type t_test is protected
    procedure test_proc;
  end protected;
end package;
package body test2_pkg is
  type t_test is protected body
    procedure test_proc is
      alias a_test_val is test_val.rec_val;
    begin
      a_test_val := 1;
    end procedure;
  end protected body;
end package body;
```
```
nvc --std=08 -a --relax=shared test_alias_shared_4.vhd
** Error: unit WORK.TESTTEST_PKG not found in library WORK
        File test_alias_shared_4.vhd, Line 6
   |
 6 | use work.testtest_pkg.all;
   | ^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: design unit depends on WORK.TEST2_PKG which was analysed with errors
        File test_alias_shared_4.vhd, Line 13
    |
 13 | package body test2_pkg is
    | ^^^^^^^^^^^^^^^^^^^^^^

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007ff66975d570] ***

[00007ff66973673e]
[00007ff84f81ff27] UnhandledExceptionFilter+0x1e7
[00007ff84f25f0b8]
[00007ff84f25eea1]
[00007ff851ff51b0] memset+0x13b0
[00007ff851fdc766] _C_specific_handler+0x96
[00007ff851ff209f] _chkstk+0x11f
[00007ff851fa1454] RtlRaiseException+0x434
[00007ff851ff0bce] KiUserExceptionDispatcher+0x2e
[00007ff66975d570]
[00007ff669763849]
[00007ff669764198]
[00007ff66976a6a3]
[00007ff66974bb0e]
[00007ff669751e1f]
[00007ff6697525db]
[00007ff669753c29]
[00007ff669759779]
[00007ff669759e82]
[00007ff6697328c4]
[00007ff66980b5f8] vhpi_put_data+0x136f8
[00007ff6697313b1]
```
While making the MWE, I also came across this one, but it might be the same cause.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
begin
  process
    alias a_test_val is test_pkg.test_val;
  begin
    a_test_val := True;
    wait;
  end process;

end architecture beh;
```
```
nvc --std=08 -a --relax=shared test_alias_shared_4.vhd
** Error: no visible declaration for TEST_PKG
        File test_alias_shared_4.vhd, Line 13
    |
 13 |     alias a_test_val is test_pkg.test_val;
    |                         ^^^^^^^^

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007ff66975d570] ***

[00007ff66973673e]
[00007ff84f81ff27] UnhandledExceptionFilter+0x1e7
[00007ff84f25f0b8]
[00007ff84f25eea1]
[00007ff851ff51b0] memset+0x13b0
[00007ff851fdc766] _C_specific_handler+0x96
[00007ff851ff209f] _chkstk+0x11f
[00007ff851fa1454] RtlRaiseException+0x434
[00007ff851ff0bce] KiUserExceptionDispatcher+0x2e
[00007ff66975d570]
[00007ff669763849]
[00007ff669764198]
[00007ff66976a6a3]
[00007ff66974bb0e]
[00007ff669751e1f]
[00007ff669755322]
[00007ff669758633]
[00007ff669758fee]
[00007ff669759e78]
[00007ff6697328c4]
[00007ff66980b5f8] vhpi_put_data+0x136f8
[00007ff6697313b1]
```

closed 2022-06-08 by nickg