nickg/nvc #467: EXCEPTION_ACCESS_VIOLATION, possibly on unconstrained vectors

avelure, 2022-06-07:
I get this crash for another library with no other output than the exception. Here all the required packages compile fine. How can I determine where it fails in this file? It is possible it is related to unconstrained arrays.
```
*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=00000216b7f1e078, ip=00007ff6697a177a] ***

[00007ff66973673e]
[00007ff84f81ff27] UnhandledExceptionFilter+0x1e7
[00007ff84f25f0b8]
[00007ff84f25eea1]
[00007ff851ff51b0] memset+0x13b0
[00007ff851fdc766] _C_specific_handler+0x96
[00007ff851ff209f] _chkstk+0x11f
[00007ff851fa1454] RtlRaiseException+0x434
[00007ff851ff0bce] KiUserExceptionDispatcher+0x2e
[00007ff6697a177a]
[00007ff669790458]
[00007ff6697909d2]
[00007ff6697908fd]
[00007ff66978db6d]
[00007ff669793779]
[00007ff66978e0d5]
[00007ff6697936e5]
[00007ff669793905]
[00007ff66978ca2d]
[00007ff66978dd59]
[00007ff66979a220]
[00007ff669798cca]
```
```
[New Thread 25660.0x98a0]
[New Thread 25660.0x1fbc]
[New Thread 25660.0x9df0]
warning: Invalid parameter passed to C runtime function.

Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007ff6697a177a in vcode_reg_type (reg=reg@entry=-1) at ../src/vcode.c:548
548        return vcode_reg_data(reg)->type;
(gdb) bt
#0  0x00007ff6697a177a in vcode_reg_type (reg=reg@entry=-1)
    at ../src/vcode.c:548
#1  0x00007ff669790458 in lower_have_uarray_ptr (reg=-1) at ../src/lower.c:368
#2  lower_array_len (type=0x28722153c20, dim=0, reg=-1) at ../src/lower.c:441
#3  0x00007ff6697909d2 in lower_array_total_len (
    type=type@entry=0x28722153c20, reg=reg@entry=-1) at ../src/lower.c:516
#4  0x00007ff6697908fd in lower_array_stride (type=type@entry=0x28722434410,
    reg=reg@entry=-1) at ../src/lower.c:504
#5  0x00007ff66978db6d in lower_array_ref (ctx=EXPR_RVALUE, ref=0x2872ab96ce0)
    at ../src/lower.c:2721
#6  lower_expr (expr=expr@entry=0x2872ab96ce0, ctx=ctx@entry=EXPR_RVALUE)
    at ../src/lower.c:3886
#7  0x00007ff669793779 in lower_param (value=value@entry=0x2872ab96ce0,
    port=port@entry=0x0, mode=mode@entry=PORT_IN) at ../src/lower.c:1049
#8  0x00007ff66978e0d5 in lower_attr_ref (ctx=EXPR_RVALUE, expr=0x2872ab96da0)
    at ../src/lower.c:3672
#9  lower_expr (expr=expr@entry=0x2872ab96da0, ctx=EXPR_RVALUE)
    at ../src/lower.c:3898
#10 0x00007ff6697936e5 in lower_param (value=value@entry=0x2872ab96da0,
    port=0x0, mode=mode@entry=PORT_IN) at ../src/lower.c:1049
#11 0x00007ff669793905 in lower_subprogram_arg (
    fcall=fcall@entry=0x2872ab96e10, nth=nth@entry=0) at ../src/lower.c:1094
#12 0x00007ff66978ca2d in lower_builtin (fcall=fcall@entry=0x2872ab96e10,
    builtin=builtin@entry=S_MUL, out_r0=out_r0@entry=0x0,
    out_r1=out_r1@entry=0x0) at ../src/lower.c:1723
#13 0x00007ff66978dd59 in lower_fcall (ctx=EXPR_RVALUE, fcall=0x2872ab96e10)
    at ../src/lower.c:2017
#14 lower_expr (expr=expr@entry=0x2872ab96e10, ctx=ctx@entry=EXPR_RVALUE)
    at ../src/lower.c:3876
#15 0x00007ff66979a220 in lower_var_assign (stmt=0x2872ab97000)
    at ../src/lower.c:4601
#16 lower_stmt (stmt=0x2872ab97000, loops=loops@entry=0x0)
    at ../src/lower.c:5815
#17 0x00007ff669798cca in lower_if (loops=0x0, stmt=0x2872ab96840)
    at ../src/lower.c:4984
#18 lower_stmt (stmt=0x2872ab96840, loops=loops@entry=0x0)
    at ../src/lower.c:5827
#19 0x00007ff669798cca in lower_if (loops=0x0, stmt=0x2872ab95a30)
    at ../src/lower.c:4984
#20 lower_stmt (stmt=0x2872ab95a30, loops=loops@entry=0x0)
    at ../src/lower.c:5827
#21 0x00007ff66979cd7f in lower_func_body (context=0x2872a217940,
    body=0x2872ab95490) at ../src/lower.c:8321
#22 lower_decls (scope=<optimized out>, scope@entry=0x2872ab950b0,
    context=<optimized out>, context@entry=0x2872a217940)
    at ../src/lower.c:7084
#23 0x00007ff66979cd50 in lower_func_body (context=0x2872a0704d0,
    body=0x2872ab950b0) at ../src/lower.c:8317
#24 lower_decls (scope=<optimized out>, scope@entry=0x2872ab30000,
    context=<optimized out>, context@entry=0x2872a0704d0)
    at ../src/lower.c:7084
#25 0x00007ff66979f895 in lower_package (unit=<optimized out>)
    at ../src/lower.c:9170
#26 lower_unit (unit=unit@entry=0x2872ab30000, cover=cover@entry=0x0)
    at ../src/lower.c:9203
#27 0x00007ff669732949 in analyse (argv=0xaadfbff89c, argc=4)
    at ../src/nvc.c:196
#28 process_command (argc=argc@entry=4, argv=argv@entry=0x28723c71740)
    at ../src/nvc.c:965
#29 0x00007ff66980b5f8 in main (argc=4, argv=0x28723c71740)
    at ../src/nvc.c:1077
```

nickg, 2022-06-08:
I think it's probably caused by passing an array with unconstrained element type as a function argument? I saw a similar crash with the latest OSVVM release but haven't fixed it yet.

avelure, 2022-06-08:
That sounds plausible, there are some functions like that in it, though this is from the UVVM library. [Here ](https://github.com/UVVM/UVVM/blob/master/uvvm_util/src/string_methods_pkg.vhd) is the concerned file.

nickg, 2022-06-12:
I think the crash above should be fixed now: I haven't tried UVVM yet but OSVVM 2022.04 is working with above commit. Note that the implementation of arrays with unconstrained elements is not complete yet so there's likely more bugs.

closed 2022-06-12 by nickg

avelure, 2022-06-13:
It has been fixed !