nickg/nvc #469: Shared variable as default value in subprogram

avelure, 2022-06-13:
I have another one from the UVVM library. In the following case I'm not sure which simulator is correct.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  shared variable shared_var : boolean  := false;
  procedure log(
    va1    : boolean    := shared_var
  );
end package;
```
```
nvc --std=93 -a test_default_shared.vhd
** Error: default value must be a static expression
        File test_default_shared.vhd, Line 8
   |
 8 |     va1    : boolean    := shared_var
   |                            ^^^^^^^^^^
```
GHDL gives no error, Modelsim gives a warning
```
** Warning: test_default_shared.vhd(9): (vcom-1346) Default expression of interface object is not globally static.
```

nickg, 2022-06-15:
The grammar rules in the 93 and 2008 LRM use static_expression:

```
interface_constant_declaration ::=
[ constant ] identifier_list : [ in ] subtype_indication [ := static_expression ]

interface_variable_declaration ::=
[ variable ] identifier_list : [ mode ] subtype_indication [ := static_expression ]
```

"Static expressions" are defined in section 9.4 of the 2008 LRM where NOTE 1 says:

> An expression that is required to be a static expression shall either be a locally static expression or a globally static expression.

And the rules for locally/globally static expressions don't include references to shared variables. So I think this code is invalid. I'm not sure why GHDL doesn't report an error - I guess it's a bug or oversight?

That said I don't have any objections to adding another `--relax=` option to skip this check or downgrade it to a warning like Modelsim as I don't think it causes any implementation issues.

avelure, 2022-06-15:
Thanks for looking into it. 
Another relax option would be good as there is a lot of these currently in UVVM.
I'll create an issue in GHDL to see what their analysis is.

closed 2022-06-15 by nickg

avelure, 2022-06-15:
Thanks, downgrading to warning is what GHDL does as well. As long as -frelaxed only downgrades errors to warnings and doesn't alter behavior I agree that a common flag should be enough.

nickg, 2022-06-17:
I've implemented this now. There's a new `--relaxed` option which takes no additional arguments. The old `--relax=` is still there but it's the same as passing `--relaxed`. Details in the manual: https://www.nickg.me.uk/nvc/manual.html#RELAXED_RULES