nickg/nvc #470: Parallel files analysis

m-kru, 2022-06-15:
Is it safe to analyze files belonging to the same library in parallel? Is it the same in case of analyzing files belonging to different libraries?

nickg, 2022-06-15:
Yes it's completely safe: accesses to a library are protected by a reader/writer lock in the filesystem (using `flock` on POSIX or `LockFileEx` on Windows) which allow multiple concurrent readers or a single writer. `make bootstrap -j` works fine for example and if you calculate the SHAs of the library files you'll always get the same result (so builds are reproducible when run in parallel).

I can add a note to the man page about this.

m-kru, 2022-06-15:
But within the library user still needs to keep the valid order of files analysis?

nickg, 2022-06-15:
You need to analyse the files in the proper dependency order: e.g. an entity before its architecture. But if two files in a library are independent you can analyse them in parallel (i.e. there won't be any data corruption or other issues).

closed 2022-06-15 by nickg