nickg/nvc #471: Node position within file.

m-kru, 2022-06-15:
Is there any struct providing information on tree node position within file. Start and end byte, or line and column numbers? Something that allows fetching code for particular node from the source code? I can't find it skimming through the sources.

nickg, 2022-06-15:
Something like:

```
tree_t t = ...;
const loc_t *loc = tree_loc(t);
int lineno = loc->first_line;
int colno = loc->first_column;
const char *file = loc_file_str(loc);
```

There's a helper function `fmt_loc` that prints the file/line/source:

```
fmt_loc(stdout, loc);
```

nickg, 2022-06-15:
I'll close this now, hope the above is sufficient. I'm not sure what the right forum is for asking questions: perhaps I could resurrect the Gitter channel?

closed 2022-06-15 by nickg