nickg/nvc #475: Dumping to FST does not preserve case

Blebowski, 2022-07-03:
Hello,

when I am dumping waveforms to FST file, signal names are converted to upper case. Is there a reason for this?

nickg, 2022-07-03:
VHDL isn't case sensitive and NVC doesn't preserve the case of any names. Everything gets normalised to upper-case when the identifier is read in (`parse_id` in `lexer.c`). The advantage of this is that you can compare names with pointer comparisons of interned strings, rather than having to do a case-insensitve string comparison each time.

FWIW I didn't invent this, it's a standard thing to do in compilers for "old" languages like LISP and Fortan that are case-insensitive.

Blebowski, 2022-07-04:
Yup, I know about the case insensitivity, so handling possible differences as soon in the processing chain as possible makes sense. However, for debugging purposes, seeing signal names in different case might not be as intuitive as in the original case. WBU keeping one extra entry in the symbol table with "original name" and it instead of normalized name? Is that too big overhead?

nickg, 2022-07-04:
We could store an extra field with each declaration in the AST that holds the original name, but there doesn't seem to be enough benefit to justify it here. 

I just checked GHDL and it doesn't preserve case in waveform dumps either, but it outputs all signal names in lowercase instead of uppercase. That might be less surprising to users at matches some of the 2008 changes in LRM which specify identifiers are printed in lowercase. That would be relatively simple to do by downcasing the strings before passing them to the FST API.

Blebowski, 2022-07-05:
True, it does not justify it enough. I checked behavior of VCS, and it is also putting all VHDL names into capital letters. Maybe a configuration option for compilation would be good ? IMHO it is about designer habits. I am used to write everything with small letters and underscores. I have seen design rules in companies where:
- Every name was capital letter
- Only port names of entities were capital letter. This was a way to distinguish internal signals from intra-module signals.
- Only VHDL keywords were supposed to be written in capital, all identifiers were supposed to be small.

That will have no overhead and user can configured it so that the names in waves are more-less matching
what he is used to writing in RTL.

nickg, 2022-07-05:
Yes there could be an option like `--wave=out.fst --fst-style=lowercase`. But maybe the simplest thing to do is just always output lower case names in the FST file, like GHDL does. I think that would satisfy most people.

nickg, 2022-07-24:
I changed it to dump identifiers in lower case like GHDL.

closed 2022-07-24 by nickg