nickg/nvc #478: VHDL-2008 null array of partially constrained subtype of unconstrained type causes ** Fatal: constant array must have constrained array type

avelure, 2022-07-19:
This might be related to #476.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  type    t_slv_array         is array (natural range <>) of std_logic_vector;
  subtype t_word              is std_logic_vector(15 downto 0);
  subtype t_word_array        is t_slv_array(open)(t_word'range);
  constant C_NULL_DATA : t_word_array(0 to -1) := (others => (others => '0'));
end package;
```

```
nvc --std=08 -a test_null_array.vhd

Name       WORK.TEST_PKG
Kind       package
Blocks     1
Registers  15
Types      15
Variables  1
  C_NULL_DATA                           // [*, *] : 0..8 => 0..8, constant
Begin
   0: r0 := package init STD.STANDARD   // P<STD.STANDARD>
      r1 := package init IEEE.STD_LOGIC_1164 // P<IEEE.STD_LOGIC_1164>
      r2 := const 0                     // # => 0
      r3 := const 1                     // # => 1
      r4 := const 0                     // -2^31..2^31-1 => 0
      r5 := const -1                    // -2^31..2^31-1 => -1
      r6 := const 0                     // 0..1 => 0
      r7 := const 15                    // -2^31..2^31-1 => 15
      r8 := const 1                     // 0..1 => 1
      r9 := null                        // @<#>
      r10 := wrap r9 [r4 r5 r6, r7 r4 r8] // [*, *] : # => @<#>
      r11 := const 16                   // # => 16
      r12 := alloc r2                   // @<0..8> => 0..8
      r13 := wrap r12 [r4 r5 r6, r7 r4 r8] // [*, *] : 0..8
      C_NULL_DATA := store r13
      r14 := const []                   // [*, *] : 0..8

** Fatal: constant array must have constrained array type
[00007FF6816365F0]
[00007FF6816A995C]
[00007FF681698CDE]
[00007FF68169925C]
[00007FF68169D5BA]
[00007FF6816A0273]
[00007FF681632A55]
[00007FF68170FA1C] vhpi_put_data+0x1388c
[00007FF6816313AE]
[00007FF6816314E6]
[00007FFDCA5C7034] BaseThreadInitThunk+0x14
[00007FFDCBB62651] RtlUserThreadStart+0x21
```

nickg, 2022-07-19:
Closed the wrong bug...

nickg, 2022-07-19:
It's actually a different cause to #476, but both should be fixed now.

commit 4ea8aa9a (2022-07-19): Crash with partially constrained field subtype. Fixes #478
closed 2022-07-19 by nickg