nickg/nvc #483: Cannot reference signal GLOB_SIG during static elaboration

avelure, 2022-07-20:
It seems you have studied this subject well, so I'm just wondering, before I update my code, if this is also valid for a signal in a package, as I assumed it would be elaborated before a process.
In the following MWE I would expect to get the initial value of the signal in the package in the constant.
I get a warning in Mentor that the constant depends on the signal value, but GHDL and Aldec reports nothing.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  signal glob_sig : std_logic := '1';
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.test_pkg.all;

entity test1 is
end entity;

architecture rtl of test1 is
begin
  p_proc : process
    constant C_CONST : std_logic := glob_sig;
  begin
    wait;
  end process;
end architecture;
```
```
nvc --std=08 -a test_glob_sig.vhd
** Error: cannot reference signal GLOB_SIG during static elaboration
    > test_glob_sig.vhd:20
    |
 20 |     constant C_CONST : std_logic := glob_sig;
    |                                     ^^^^^^^^
    |
    = Note: the value of a signal is not defined until after the design hierarchy is elaborated
    = Help: IEEE Std 1076-2008 section 14.4 "Elaboration of a declarative part"
```

nickg, 2022-07-20:
I believe this is correct. There's a note about it on page 206 of the 2008 LRM (my emphasis):

> NOTE 1— It is a consequence of this rule that the name of a signal declared within a block cannot be referenced in expressions appearing in declarative items within that block, an inner block, **or process statement**; nor can it be passed as a parameter to a function called during the elaboration of the block. These restrictions exist because the value of a signal is not defined until after the design hierarchy is elaborated. However, a signal parameter name may be used within expressions in declarative items within a subprogram declarative part, provided that the subprogram is only called after simulation begins, because the value of every signal will be defined by that time.

nickg, 2022-07-20:
Actually I completely ignored the part about it being in a package ;-). I need to check that...

nickg, 2022-07-24:
I think this error is correct. When you read the value of a signal like `glob_sig`, what you read is the "effective value" of the signal. The initialisation steps for a VHDL design are:
1. elaborate the entire design hierarchy including any packages.
2. calculate the initial driving and effective value for every signal.

I can't see anything in the LRM that says during step 1 reading a signal should return the default value instead of the effective value. The note above is explanatory and doesn't have to exhaustively cover every case, so I think it applies to signals in packages too.

I've downgraded this to a warning when you pass `--relaxed`, in which case it reads the default value of the signal, the same as the other simulators.

closed 2022-07-24 by nickg

avelure, 2022-07-24:
That makes sense, I'll update my code.