nickg/nvc #485: Install for UVVM

avelure, 2022-07-20:
I tested the `--install uvvm` option as you suggested on Mingw.

Firstly there is an issue for the paths on mingw
```
nvc --install uvvm
** Fatal: C:\msys64\ucrt64\libexec\nvc/install-uvvm.sh failed with status -1
```
The path should be `C:\\msys64\\ucrt64\\libexec\\nvc\\install-uvvm.sh` 

Executing it manually the build fails when it gets to the context file as it refers to uvvm_util for the libraries instead of work and it is not able to find the library.
```
$ C:\\msys64\\ucrt64\\libexec\\nvc\\install-uvvm.sh
mkdir -p /home/user/.cache/nvc/UVVM-v2022.05.25
git clone --recursive https://github.com/UVVM/UVVM -b v2022.05.25 /home/user/.cache/nvc/UVVM-v2022.05.25 --depth=1
Cloning into '/home/user/.cache/nvc/UVVM-v2022.05.25'...
remote: Enumerating objects: 693, done.
remote: Counting objects: 100% (693/693), done.
remote: Compressing objects: 100% (515/515), done.
remote: Total 693 (delta 269), reused 411 (delta 145), pack-reused 0
Receiving objects: 100% (693/693), 27.23 MiB | 3.50 MiB/s, done.
Resolving deltas: 100% (269/269), done.
Note: switching to 'ba91436b3489da74b475a5e418d20fb728014063'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Updating files: 100% (708/708), done.
cd /home/user/.cache/nvc/UVVM-v2022.05.25
mkdir -p /home/user/.nvc/lib
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/types_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/adaptations_pkg.vhd
** Warning: shared variable SHARED_DEFAULT_LOG_DESTINATION must have protected type
    > uvvm_util/src/adaptations_pkg.vhd:72
    |
 72 |   shared variable shared_default_log_destination : t_log_destination := CONSOLE_AND_LOG;
    |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/string_methods_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/protected_types_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/global_signals_and_shared_variables_pkg.vhd
** Warning: shared variable SHARED_INITIALISED_UTIL must have protected type
    > uvvm_util/src/global_signals_and_shared_variables_pkg.vhd:27
    |
 27 |   shared variable shared_initialised_util        : boolean  := false;
    |                   ^^^^^^^^^^^^^^^^^^^^^^^
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/hierarchy_linked_list_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/alert_hierarchy_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/license_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/methods_pkg.vhd
** Warning: default value must be a static expression
    > uvvm_util/src/methods_pkg.vhd:84
    |
 84 |     msg_id_panel    : t_msg_id_panel    := shared_msg_id_panel;
    |                                            ^^^^^^^^^^^^^^^^^^^
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/bfm_common_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/generic_queue_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/data_queue_pkg.vhd
** Warning: shared variable SHARED_DATA_BUFFER must have protected type
    > uvvm_util/src/data_queue_pkg.vhd:30
    |
 30 |   shared variable shared_data_buffer : t_data_buffer;
    |                   ^^^^^^^^^^^^^^^^^^
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/data_fifo_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/data_stack_pkg.vhd
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/rand_pkg.vhd
** Warning: default value must be a static expression
     > uvvm_util/src/rand_pkg.vhd:162
     |
 162 |       constant msg_id_panel : in t_msg_id_panel := shared_msg_id_panel);
     |                                                    ^^^^^^^^^^^^^^^^^^^
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/func_cov_pkg.vhd
** Warning: default value must be a static expression
     > uvvm_util/src/func_cov_pkg.vhd:151
     |
 151 |     constant msg_id_panel : in t_msg_id_panel := shared_msg_id_panel);
     |                                                  ^^^^^^^^^^^^^^^^^^^
nvc --std=2008 --work=/home/user/.nvc/lib/uvvm_util -L/home/user/.nvc/lib -a --relaxed uvvm_util/src/uvvm_util_context.vhd
** Error: unit UVVM_UTIL.TYPES_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:19
    |
 19 |   use uvvm_util.types_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.GLOBAL_SIGNALS_AND_SHARED_VARIABLES_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:20
    |
 20 |   use uvvm_util.global_signals_and_shared_variables_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.HIERARCHY_LINKED_LIST_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:21
    |
 21 |   use uvvm_util.hierarchy_linked_list_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.STRING_METHODS_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:22
    |
 22 |   use uvvm_util.string_methods_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.ADAPTATIONS_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:23
    |
 23 |   use uvvm_util.adaptations_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.METHODS_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:24
    |
 24 |   use uvvm_util.methods_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.BFM_COMMON_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:25
    |
 25 |   use uvvm_util.bfm_common_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.ALERT_HIERARCHY_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:26
    |
 26 |   use uvvm_util.alert_hierarchy_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.LICENSE_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:27
    |
 27 |   use uvvm_util.license_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.PROTECTED_TYPES_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:28
    |
 28 |   use uvvm_util.protected_types_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.RAND_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:29
    |
 29 |   use uvvm_util.rand_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: unit UVVM_UTIL.FUNC_COV_PKG not found in library UVVM_UTIL
    > uvvm_util/src/uvvm_util_context.vhd:30
    |
 30 |   use uvvm_util.func_cov_pkg.all;
    |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

I thought that the `--work` parameter was supposed to be `--work=mylib:somedir` or `--work=mylib`, but can it also be `--work=somedir`? I tried adding the mylib to the install file, but then I just get `** Fatal: invalid character ';' in library name` which I don't understand.

nickg, 2022-07-21:
> I thought that the --work parameter was supposed to be --work=mylib:somedir or --work=mylib, but can it also be --work=somedir?

`--work=some/dir` is equivalent to `--work=dir:some/dir`.

nickg, 2022-07-21:
There were quite a few small issues with this on Windows but it's working OK for me now, and I've added it to the GitHub Actions tests. Could you try again?

closed 2022-07-21 by nickg

avelure, 2022-07-22:
Install works now in both mingw shell and native windows!

The UVVM patch can probably be reported to their github repo to get fixed so that you don't need to maintain it.