nickg/nvc #486: Elab: uvvm_util : report_alert_counters throws exception

avelure, 2022-07-20:
The following throws an exception.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

entity test is
end entity;

architecture tb of test is
begin
   p_main: process
   begin
     report_alert_counters(FINAL);
    wait;
  end process p_main;
end architecture;
```
```
nvc --std=08 -LC:\proj\library -a test_uvvm.vhd -e test-tb -r

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007FFD1012931E] ***

[00007FF790C27BAF]
[00007FFDC968FF27] UnhandledExceptionFilter+0x1e7
[00007FFDC8E2F0B8]
[00007FFDC8E2EEA1]
[00007FFDCBBB51B0] memset+0x13b0
[00007FFDCBB9C766] _C_specific_handler+0x96
[00007FFDCBBB209F] _chkstk+0x11f
[00007FFDCBB61454] RtlRaiseException+0x434
[00007FFDCBBB0BCE] KiUserExceptionDispatcher+0x2e
[00007FFD1012931E] ZN4llvm11PointerType3getEPNS_4TypeEj+0xe
[00007FFD10364D4D] ZN4llvm13IRBuilderBase26CreateConstInBoundsGEP2_32EPNS_4TypeEPNS_5ValueEjjRKNS_5TwineE+0x16d
[00007FFD0F1B9BDE] LLVMBuildStructGEP+0x5e
[00007FF790CEB698] _nvc_flush+0x22d74
[00007FF790CF1074] _nvc_flush+0x28750
[00007FF790CF15BE] _nvc_flush+0x28c9a
[00007FF790CF173F] _nvc_flush+0x28e1b
[00007FF790CF1BA2] _nvc_flush+0x2927e
[00007FF790CF6612] _nvc_flush+0x2dcee
[00007FF790CF69BE] _nvc_flush+0x2e09a
[00007FFD866A4B4B] pthread_create_wrapper+0x9b
[00007FFDC9251BB2] configthreadlocale+0x92
[00007FFDCA5C7034] BaseThreadInitThunk+0x14
```

nickg, 2022-07-21:
I think this is probably the same crash as #487

avelure, 2022-07-21:
You are probably right, the backtrace hits the same line in cgen
```
Thread 6 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 23576.0x4470]
0x00007ffd110b0b25 in ?? () from /ucrt64/bin/libLLVM-14.dll
(gdb) bt
#0  0x00007ffd110b0b25 in ?? () from /ucrt64/bin/libLLVM-14.dll
#1  0x00007ffd110b1f1d in ?? () from /ucrt64/bin/libLLVM-14.dll
#2  0x00007ffd0f1b4a7c in ?? () from /ucrt64/bin/libLLVM-14.dll
#3  0x00007ff70c396bf5 in cgen_uarray_dim (dim=0, meta=0x19eef6827f0) at ../src/cgen.c:898
#4  cgen_op_uarray_len (op=op@entry=56, ctx=ctx@entry=0x84b2dff9e0) at ../src/cgen.c:1965
#5  0x00007ff70c3a05db in cgen_op (i=i@entry=56, ctx=ctx@entry=0x84b2dff9e0) at ../src/cgen.c:3671
#6  0x00007ff70c3a1edd in cgen_block (ctx=0x84b2dff9e0, block=17) at ../src/cgen.c:3738
#7  cgen_code (ctx=ctx@entry=0x84b2dff9e0) at ../src/cgen.c:3776
#8  0x00007ff70c3a2520 in cgen_function () at ../src/cgen.c:3860
#9  cgen_units (primary=<optimized out>, obj_path=<optimized out>, tm_ref=<optimized out>, module_name=<optimized out>, cover=<optimized out>, top=<optimized out>, units=<synthetic pointer>) at ../src/cgen.c:5237
#10 cgen_worker_thread (__arg=<optimized out>) at ../src/cgen.c:5330
#11 0x00007ffd9adb4b4b in ?? () from /ucrt64/bin/libwinpthread-1.dll
#12 0x00007ffdc9251bb2 in ucrtbase!_configthreadlocale () from /c/Windows/System32/ucrtbase.dll
#13 0x00007ffdca5c7034 in KERNEL32!BaseThreadInitThunk () from /c/Windows/System32/KERNEL32.DLL
#14 0x00007ffdcbb62651 in ntdll!RtlUserThreadStart () from /c/Windows/SYSTEM32/ntdll.dll
#15 0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

commit 0a4e98ad (2022-07-21): Symbols truncated to 240 characters on Windows. Fixes #486 #487
closed 2022-07-21 by nickg