nickg/nvc #487: Elab: uvvm_util : check_value on vector throws exception

avelure, 2022-07-20:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

entity test is
end entity;

architecture tb of test is
  signal result : unsigned(3 downto 0);
begin
  result <= x"0";
  p_main: process
  begin
    check_value(result, x"0", error, "Checking result", C_SCOPE);
    wait;
  end process p_main;
end architecture;
```
```
nvc --std=08 -LC:\proj\library -a test_uvvm_2.vhd -e test-tb -r

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007FFD110B0B25] ***

[00007FF790C27BAF]
[00007FFDC968FF27] UnhandledExceptionFilter+0x1e7
[00007FFDC8E2F0B8]
[00007FFDC8E2EEA1]
[00007FFDCBBB51B0] memset+0x13b0
[00007FFDCBB9C766] _C_specific_handler+0x96
[00007FFDCBBB209F] _chkstk+0x11f
[00007FFDCBB61454] RtlRaiseException+0x434
[00007FFDCBBB0BCE] KiUserExceptionDispatcher+0x2e
[00007FFD110B0B25] ZN4llvm5Value11setNameImplERKNS_5TwineE+0x15
[00007FFD110B1F1D] ZN4llvm5Value7setNameERKNS_5TwineE+0xd
[00007FFD0F1B4A7C] LLVMBuildExtractValue+0x1dc
[00007FF790CE6E2E] _nvc_flush+0x1e50a
[00007FF790CEB4BC] _nvc_flush+0x22b98
[00007FF790CF13F1] _nvc_flush+0x28acd
[00007FF790CF15BE] _nvc_flush+0x28c9a
[00007FF790CF173F] _nvc_flush+0x28e1b
[00007FF790CF1BA2] _nvc_flush+0x2927e
[00007FF790CF6612] _nvc_flush+0x2dcee
[00007FF790CF69BE] _nvc_flush+0x2e09a
[00007FFD8F2F4B4B] pthread_create_wrapper+0x9b
[00007FFDC9251BB2] configthreadlocale+0x92
```

nickg, 2022-07-21:
This seems to be Windows specific. I get:

```
UVVM: ***  ERROR #1  ***
UVVM:             0 ns   TB seq.(uvvm)
UVVM:                    check_value() => Failed. unsigned  Was x"X" (b"UUUU"). Expected x"0".
UVVM:                    Checking result
```

Could you try catching the error in gdb and print a backtrace?

avelure, 2022-07-21:
```
Thread 7 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 28100.0x18c8]
0x00007ffd110b0b25 in ?? () from /ucrt64/bin/libLLVM-14.dll
(gdb) bt
#0  0x00007ffd110b0b25 in ?? () from /ucrt64/bin/libLLVM-14.dll
#1  0x00007ffd110b1f1d in ?? () from /ucrt64/bin/libLLVM-14.dll
#2  0x00007ffd0f1b4a7c in ?? () from /ucrt64/bin/libLLVM-14.dll
#3  0x00007ff70c396bf5 in cgen_uarray_dim (dim=0, meta=0x11a75719880) at ../src/cgen.c:898
#4  cgen_op_uarray_len (op=op@entry=56, ctx=ctx@entry=0xd331ffa80) at ../src/cgen.c:1965
#5  0x00007ff70c3a05db in cgen_op (i=i@entry=56, ctx=ctx@entry=0xd331ffa80) at ../src/cgen.c:3671
#6  0x00007ff70c3a1edd in cgen_block (ctx=0xd331ffa80, block=17) at ../src/cgen.c:3738
#7  cgen_code (ctx=ctx@entry=0xd331ffa80) at ../src/cgen.c:3776
#8  0x00007ff70c3a2520 in cgen_function () at ../src/cgen.c:3860
#9  cgen_units (primary=<optimized out>, obj_path=<optimized out>, tm_ref=<optimized out>, module_name=<optimized out>, cover=<optimized out>, top=<optimized out>, units=<synthetic pointer>) at ../src/cgen.c:5237
#10 cgen_worker_thread (__arg=<optimized out>) at ../src/cgen.c:5330
#11 0x00007ffd9b054b4b in ?? () from /ucrt64/bin/libwinpthread-1.dll
#12 0x00007ffdc9251bb2 in ucrtbase!_configthreadlocale () from /c/Windows/System32/ucrtbase.dll
#13 0x00007ffdca5c7034 in KERNEL32!BaseThreadInitThunk () from /c/Windows/System32/KERNEL32.DLL
#14 0x00007ffdcbb62651 in ntdll!RtlUserThreadStart () from /c/Windows/SYSTEM32/ntdll.dll
#15 0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

nickg, 2022-07-21:
I managed to reproduce this in a Windows 10 VM. It's caused by truncating all symbols on Windows to 240 characters. I can't remember why I did this - maybe a previous limitation of the linker? Anyway it doesn't seem to be necessary any more and removing that fixes it.

closed 2022-07-21 by nickg

avelure, 2022-07-22:
Great, now I can successfully run some of my module testbenches with UVVM!