nickg/nvc #492: Build breaks on armv6, armv7 architectures

yurivict, 2022-07-22:
```
src/debug.c:689:14: error: use of undeclared identifier '_URC_NO_REASON'
      return _URC_NO_REASON;
             ^
src/debug.c:693:19: warning: implicit declaration of function '_Unwind_GetIPInfo' is invalid in C99 [-Wimplicit-function-declaration]
   uintptr_t ip = _Unwind_GetIPInfo(ctx, &ip_before_instruction);
                  ^
src/debug.c:696:14: error: use of undeclared identifier '_URC_NO_REASON'
      return _URC_NO_REASON;
             ^
src/debug.c:707:14: error: use of undeclared identifier '_URC_NORMAL_STOP'
      return _URC_NORMAL_STOP;
             ^
src/debug.c:709:14: error: use of undeclared identifier '_URC_NO_REASON'
      return _URC_NO_REASON;
             ^
1 warning and 4 errors generated.
```

clang-{11,12,13,14}
FreeBSD 13.1

nickg, 2022-07-24:
Those symbols normally come from `unwind.h`. I think this is the file from FreeBSD base:

https://github.com/freebsd/freebsd-src/blob/14c99b43ef69beafa01a1574641c4cf19ecf2b9d/contrib/llvm-project/clang/lib/Headers/unwind.h

It looks like those should be defined on Arm (unless it's picking up libunwind from somewhere else?).

Is it possible to get shell access to one of the systems?

yurivict, 2022-07-24:
Hi @nickg ,

Thanks for your help.

This appears to be a bug in the FreeBSD headers. I reported it: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=265420

Cheers,
Yuri

closed 2022-07-24 by yurivict