nickg/nvc #494: fatal: : The operation completed successfully

avelure, 2022-07-26:
I get the mentioned error when running one of the testbenches. It just exits with exitcode 1 so there is no backtrace. It happens at time 0 before after printing some messages, probably before any delta-cycles. Any idea how I can debug it?
```
Assertion Warning: NUMERIC_STD."<": metavalue detected, returning FALSE
fatal: : The operation completed successfully.
```

nickg, 2022-07-26:
Try putting a breakpoint in `fatal_exit` (`b fatal_exit` just after you start gdb). Then it should stop just before exiting and you can get a stack trace.

avelure, 2022-07-26:
I had to break on `exit` as breaking on fatal_exit gave me some gdb issues
```
Program terminated with signal SIGHUP, Hangup.
The program no longer exists.
```

The issue is in the std.textio file_open
```
fatal: : The operation completed successfully.

Thread 1 hit Breakpoint 2, 0x00007ff84c0e03e0 in ucrtbase!exit () from /c/Windows/System32/ucrtbase.dll
(gdb) bt
#0  0x00007ff84c0e03e0 in ucrtbase!exit () from /c/Windows/System32/ucrtbase.dll
#1  0x00007ff60e89e2ed in rt_abort_sim (code=1) at ../src/rt/rtkern.c:450
#2  rt_msg (where=where@entry=0x0, level=DIAG_FATAL, fmt=fmt@entry=0x7ff60e90d687 <__emutls_t.my_thread+4087> "%s: %s", level=DIAG_FATAL) at ../src/rt/rtkern.c:547
#3  0x00007ff60e8a4a00 in _file_open (status=0x22024551948 "", _fp=0x22024551940, name_bytes=0x7fffbb75c609 <__nvc_abi_version+271881> "", name_len=<optimized out>, mode=0 '\000') at ../src/rt/rtkern.c:2342
#4  0x00007fffbb6a9589 in <WORK.EEPROM_TOP_TB.I_EEPROM_TOP.I_EEPROM_CACHE.I_EDAC_RAM.I_EDAC_RAM_MAP.G_EDAC_64X16.I_EDAC_64X16.G_RTAX_EDAC.I_EDAC_64X16.EDAC_64X16_0.RTAX_RAM.EDAC_64X16_EDAC_64X16_0_ACTRAM_0.AX_RAM_NO_FABRIC_PIPE.EDAC_64X16_EDAC_64X16_0_RAM_0.EDAC_64X16_EDAC_64X16_0_ACTRAM_R0C0.VITALBEHAVIOR> () at C:\proj\libraries\axcelerator/axcelerator.vhd:100339
#5  0x00007ff60e89ed28 in rt_run (proc=proc@entry=0x22020230a50) at ../src/rt/rtkern.c:2879
#6  0x00007ff60e89fc38 in rt_resume (list=list@entry=0x7ff60e928ef0 <resume>) at ../src/rt/rtkern.c:3623
#7  0x00007ff60e8a5d9e in rt_cycle (stop_delta=<optimized out>) at ../src/rt/rtkern.c:3758
#8  rt_run_sim (top=top@entry=0x22014a10000, stop_time=stop_time@entry=9223372036854775807) at ../src/rt/rtkern.c:4111
#9  0x00007ff60e812986 in run (argv=0x220108307a0, argc=2) at ../src/nvc.c:499
#10 process_command (argc=argc@entry=2, argv=argv@entry=0x220108307a0) at ../src/nvc.c:1049
#11 0x00007ff60e812ce6 in elaborate (argv=0x220108307a0, argc=2) at ../src/nvc.c:316
#12 process_command (argc=argc@entry=4, argv=argv@entry=0x22010830790) at ../src/nvc.c:1047
#13 0x00007ff60e8f169b in main (argc=4, argv=0x22010830790) at ../src/nvc.c:1164
```

I didn't manage to recreate successfully in an MWE, but the code is something like this
```
library ieee;
library STD;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

entity test is
  generic(
           MEMORYFILE      : String  := "");
end entity;

architecture rtl of test is
begin
  p_proc : process
   file     memfile        : text;
   variable status         : file_open_status;
  begin
    file_open(status, memfile, MEMORYFILE, read_mode);
    if ( status=open_ok ) then
      report "Open ok" severity note;
    else
      report "Open not ok" severity note;
    end if;
    wait;
  end process;
end architecture;
```
This returns another error though. If I set the generic to a non-empty string it either hits the "not ok" branch if the file is not there or the "ok" if I specify a correct one.
```
nvc --std=08 -a test_file_open.vhd -e test-rtl -r
** Fatal: : The specified procedure could not be found.
```

nickg, 2022-07-26:
There seem to be three problems with this error message:
1. If the file name is empty, it's difficult to see it comes from the a `FILE_OPEN` (because the message normally includes the file name) => I've added a special-case error message for empty file names.
2. On Windows the error message is unrelated to the file open failure => this is because we're using `GetLastError()` but according to the MS docs we should call `fopen_s` and then `strerror()` as on Linux.
3. On Windows (and probably macOS), the error message doesn't show the source location => should be fixed now.

Can you try with the latest code and see if that helps?

avelure, 2022-07-26:
Yes, this works now. Thanks!!

closed 2022-07-26 by nickg