nickg/nvc #495: cannot lower attribute STABLE (8)

avelure, 2022-07-27:
I get this error when trying to simulate something using the stable attribute.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture rtl of test is
  constant C_T_RCVON_TIED : time := 50 ns;
  signal rx_en     : std_logic;
  signal rx_en_del : std_logic;
begin
  p_proc : process
  begin
    if rx_en'stable(C_T_RCVON_TIED) then
      rx_en_del <= rx_en;
    end if;
    wait;
  end process;
end architecture;
```
```
nvc --std=08 -a test_stable.vhd -e test-rtl -r
** Fatal: cannot lower attribute STABLE (8)
    > test_stable.vhd:15
    |
 15 |     if rx_en'stable(C_T_RCVON_TIED) then
    |        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

nickg, 2022-07-28:
I've just implemented this in a simple way that should work for common cases like yours above but isn't fully correct according to the LRM (`X'STABLE` can't be used like a signal) so I'll need to revisit this at some point. Can you can see if it works with your code?

avelure, 2022-08-01:
This works ok now.

closed 2022-08-02 by nickg