nickg/nvc #496: cannot handle type 4 in irgen_align_of

avelure, 2022-07-27:
I'm not able to fully recreate this in an MWE, but the code is something like this:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  function get_length(
    constant msg_id : in natural) return natural;
end package;

package body test_pkg is
function get_length(
    constant msg_id : in natural)
  return natural is
  begin
    return 1;
  end function;
end package body;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.test_pkg.all;

entity test is
end entity;

architecture rtl of test is
  type t_slv_array is array (natural range <>) of std_logic_vector;
  subtype t_data_array is t_slv_array(open)(15 downto 0);
begin
  p_proc : process
  procedure proc(
    constant data      : in    t_data_array
  ) is
  begin
  end procedure;

  procedure proc_test is
    variable v_payload         : t_data_array(0 to 31);
  begin
    proc(v_payload(0 to get_length(1) -1));
  end procedure;

  begin
    proc_test;
    wait;
  end process;
end architecture;
```
`test_pkg` is in another library, while the entity is in `work`. If I move the `get_length` function into the procedure `proc_test` the error disappears. If I replace the `get_length` call with `1` it works.

This is the gdb backtrace from the original code
```
fatal: cannot handle type 4 in irgen_align_of
[00007FF74AA06580]
[00007FF74AA9BF36] _nvc_flush+0x6ff6
[00007FF74AA9C08E] _nvc_flush+0x714e
[00007FF74AAA299D] _nvc_flush+0xda5d
[00007FF74AAA4B7F] _nvc_flush+0xfc3f
[00007FF74AA9B706] _nvc_flush+0x67c6
[00007FF74AAA0355] _nvc_flush+0xb415
[00007FF74AAA2AE5] _nvc_flush+0xdba5
[00007FF74AAA4B7F] _nvc_flush+0xfc3f
[00007FF74AA9B706] _nvc_flush+0x67c6
[00007FF74AAA0355] _nvc_flush+0xb415
[00007FF74AAA2AE5] _nvc_flush+0xdba5
[00007FF74AAA4B7F] _nvc_flush+0xfc3f
[00007FF74AA9B706] _nvc_flush+0x67c6
[00007FF74AAA0355] _nvc_flush+0xb415
[00007FF74AAA2AE5] _nvc_flush+0xdba5
[00007FF74AAA4B7F] _nvc_flush+0xfc3f
[00007FF74AA9B706] _nvc_flush+0x67c6
[00007FF74AA9FF45] _nvc_flush+0xb005
[00007FF74AAA2AE5] _nvc_flush+0xdba5
[00007FF74AA9B8EF] _nvc_flush+0x69af
[00007FF74AA8AB71]

Thread 1 hit Breakpoint 1, 0x00007ff84c0e03e0 in ucrtbase!exit () from /c/Windows/System32/ucrtbase.dll
(gdb) bt
#0  0x00007ff84c0e03e0 in ucrtbase!exit () from /c/Windows/System32/ucrtbase.dll
#1  0x00007ff74aa05198 in fatal_exit (status=1) at ../src/util.c:449
#2  0x00007ff74aa0658a in fatal_trace (fmt=fmt@entry=0x7ff74aafef40 <__emutls_t.stack_limit+680> "cannot handle type %d in irgen_align_of") at ../src/util.c:526
#3  0x00007ff74aa9bf36 in irgen_align_of (vtype=vtype@entry=37) at ../src/jit/jit-irgen.c:601
#4  0x00007ff74aa9c08e in irgen_size_bytes (vtype=<optimized out>) at ../src/jit/jit-irgen.c:647
#5  0x00007ff74aaa299d in irgen_locals (g=0x1e7921ba8a0) at ../src/jit/jit-irgen.c:2430
#6  jit_irgen (f=0x1e791b13a30) at ../src/jit/jit-irgen.c:2471
#7  0x00007ff74aaa4b7f in jit_interp (f=f@entry=0x1e791b13a30, args=args@entry=0x1321fd810, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#8  0x00007ff74aa9b706 in jit_link (j=0x1e784e8fe20, handle=<optimized out>) at ../src/jit/jit-core.c:200
#9  0x00007ff74aaa0355 in irgen_op_package_init (op=5, g=0x1e791ac1690) at ../src/jit/jit-irgen.c:1948
#10 irgen_block (g=g@entry=0x1e791ac1690, block=block@entry=0) at ../src/jit/jit-irgen.c:2314
#11 0x00007ff74aaa2ae5 in jit_irgen (f=0x1e791ac1600) at ../src/jit/jit-irgen.c:2485
#12 0x00007ff74aaa4b7f in jit_interp (f=f@entry=0x1e791ac1600, args=args@entry=0x1321fddc0, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#13 0x00007ff74aa9b706 in jit_link (j=0x1e784e8fe20, handle=<optimized out>) at ../src/jit/jit-core.c:200
#14 0x00007ff74aaa0355 in irgen_op_package_init (op=8, g=0x1e7918a9aa0) at ../src/jit/jit-irgen.c:1948
#15 irgen_block (g=g@entry=0x1e7918a9aa0, block=block@entry=0) at ../src/jit/jit-irgen.c:2314
#16 0x00007ff74aaa2ae5 in jit_irgen (f=0x1e7918a96b0) at ../src/jit/jit-irgen.c:2485
#17 0x00007ff74aaa4b7f in jit_interp (f=f@entry=0x1e7918a96b0, args=args@entry=0x1321fe370, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#18 0x00007ff74aa9b706 in jit_link (j=0x1e784e8fe20, handle=<optimized out>) at ../src/jit/jit-core.c:200
#19 0x00007ff74aaa0355 in irgen_op_package_init (op=6, g=0x1e78994ce40) at ../src/jit/jit-irgen.c:1948
#20 irgen_block (g=g@entry=0x1e78994ce40, block=block@entry=0) at ../src/jit/jit-irgen.c:2314
#21 0x00007ff74aaa2ae5 in jit_irgen (f=0x1e78fdba480) at ../src/jit/jit-irgen.c:2485
#22 0x00007ff74aaa4b7f in jit_interp (f=f@entry=0x1e78fdba480, args=args@entry=0x1321fe920, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#23 0x00007ff74aa9b706 in jit_link (j=0x1e784e8fe20, handle=<optimized out>) at ../src/jit/jit-core.c:200
#24 0x00007ff74aa9ff45 in irgen_op_link_package (op=0, g=0x1e7899d2450) at ../src/jit/jit-irgen.c:1964
#25 irgen_block (g=g@entry=0x1e7899d2450, block=block@entry=0) at ../src/jit/jit-irgen.c:2317
#26 0x00007ff74aaa2ae5 in jit_irgen (f=0x1e78fa875f0) at ../src/jit/jit-irgen.c:2485
#27 0x00007ff74aa9b8ef in jit_call_thunk (j=0x1e784e8fe20, unit=unit@entry=0x1e78994cb70, result=result@entry=0x1321feff8) at ../src/jit/jit-core.c:301
#28 0x00007ff74aa8ab71 in eval_do_fold (ex=0x1e784e8fde0, expr=0x1e78a180dd0) at ../src/eval.c:130
#29 0x00007ff74aa59747 in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:660
#30 0x00007ff74aa59892 in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:631
#31 0x00007ff74aa5994c in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:637
#32 0x00007ff74aa59892 in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:631
#33 0x00007ff74aa5994c in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:637
#34 0x00007ff74aa5994c in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:637
#35 0x00007ff74aa59892 in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:631
#36 0x00007ff74aa59892 in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:631
#37 0x00007ff74aa5994c in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:637
#38 0x00007ff74aa5994c in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:637
#39 0x00007ff74aa5994c in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x1321ff650) at ../src/object.c:637
#40 0x00007ff74aa5994c in object_rewrite (object=<optimized out>, object@entry=0x1e78a1814e0, ctx=ctx@entry=0x1321ff650) at ../src/object.c:637
#41 0x00007ff74aa317e7 in tree_rewrite (t=t@entry=0x1e78a1814e0, pre_fn=pre_fn@entry=0x7ff74aa46020 <simp_pre_cb>, tree_post_fn=tree_post_fn@entry=0x7ff74aa46140 <simp_tree>,
    type_post_fn=0x7ff74aa45200 <simp_type>, context=context@entry=0x1321ff6f0) at ../src/tree.c:1191
#42 0x00007ff74aa48004 in simplify_global (top=top@entry=0x1e78a1814e0, generics=generics@entry=0x1e7897a8cb0, ex=<optimized out>) at ../src/simp.c:1714
#43 0x00007ff74aa44be3 in elab_top_level (arch=arch@entry=0x1e784fd0000, ctx=<optimized out>, ctx@entry=0x1321ff880) at ../src/elab.c:1662
#44 0x00007ff74aa44f2d in elab (top=top@entry=0x1e784fd0000) at ../src/elab.c:1730
#45 0x00007ff74aa02be6 in elaborate (argv=0x1e784ddafc0, argc=7) at ../src/nvc.c:290
#46 process_command (argc=argc@entry=7, argv=argv@entry=0x1e784ddafc0) at ../src/nvc.c:1047
#47 0x00007ff74aae1a7b in main (argc=7, argv=0x1e784ddafc0) at ../src/nvc.c:1164
```

nickg, 2022-07-27:
Is there a record signal in your package? If I add one I can get the same error. Could you try with the fix I just pushed?

avelure, 2022-07-27:
There are records, constants, types, functions and procedures.
I get an exception now instead:
```
*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000010, ip=00007FF7F0018CE5] ***

[00007FF7EFFA6A24]
[00007FF84C460057] UnhandledExceptionFilter+0x1e7
[00007FF84BB0F0B8]
[00007FF84BB0EEA1]
[00007FF84E8953B0] memset+0x13b0
[00007FF84E87C766] _C_specific_handler+0x96
[00007FF84E89229F] _chkstk+0x11f
[00007FF84E841454] RtlRaiseException+0x434
[00007FF84E890DCE] KiUserExceptionDispatcher+0x2e
[00007FF7F0018CE5]
[00007FF7F0040141] _nvc_flush+0xb481
[00007FF7F0042885] _nvc_flush+0xdbc5
[00007FF7F004493F] _nvc_flush+0xfc7f
[00007FF7F003B486] _nvc_flush+0x67c6
[00007FF7F00400D5] _nvc_flush+0xb415
[00007FF7F0042885] _nvc_flush+0xdbc5
[00007FF7F004493F] _nvc_flush+0xfc7f
[00007FF7F003B486] _nvc_flush+0x67c6
[00007FF7F00400D5] _nvc_flush+0xb415
[00007FF7F0042885] _nvc_flush+0xdbc5
[00007FF7F004493F] _nvc_flush+0xfc7f
[00007FF7F003B486] _nvc_flush+0x67c6
```
```

Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007ff7f0018ce5 in vcode_param_reg (param=param@entry=0) at ../src/vcode.c:2764
2764       return active_unit->params.items[param].reg;
(gdb) bt
#0  0x00007ff7f0018ce5 in vcode_param_reg (param=param@entry=0) at ../src/vcode.c:2764
#1  0x00007ff7f0040141 in irgen_op_var_upref (op=13, g=0x1b36dadbc60) at ../src/jit/jit-irgen.c:1549
#2  irgen_block (g=g@entry=0x1b36dadbc60, block=block@entry=0) at ../src/jit/jit-irgen.c:2318
#3  0x00007ff7f0042885 in jit_irgen (f=0x1b36dadbb10) at ../src/jit/jit-irgen.c:2510
#4  0x00007ff7f004493f in jit_interp (f=f@entry=0x1b36dadbb10, args=args@entry=0x2fddfcec0, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#5  0x00007ff7f003b486 in jit_link (j=0x1b364fceca0, handle=<optimized out>) at ../src/jit/jit-core.c:200
#6  0x00007ff7f00400d5 in irgen_op_package_init (op=9, g=0x1b36de40480) at ../src/jit/jit-irgen.c:1951
#7  irgen_block (g=g@entry=0x1b36de40480, block=block@entry=0) at ../src/jit/jit-irgen.c:2327
#8  0x00007ff7f0042885 in jit_irgen (f=0x1b36de403f0) at ../src/jit/jit-irgen.c:2510
#9  0x00007ff7f004493f in jit_interp (f=f@entry=0x1b36de403f0, args=args@entry=0x2fddfd470, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#10 0x00007ff7f003b486 in jit_link (j=0x1b364fceca0, handle=<optimized out>) at ../src/jit/jit-core.c:200
#11 0x00007ff7f00400d5 in irgen_op_package_init (op=11, g=0x1b36d893a40) at ../src/jit/jit-irgen.c:1951
#12 irgen_block (g=g@entry=0x1b36d893a40, block=block@entry=16) at ../src/jit/jit-irgen.c:2327
#13 0x00007ff7f0042885 in jit_irgen (f=0x1b36cea7bc0) at ../src/jit/jit-irgen.c:2510
#14 0x00007ff7f004493f in jit_interp (f=f@entry=0x1b36cea7bc0, args=args@entry=0x2fddfda20, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#15 0x00007ff7f003b486 in jit_link (j=0x1b364fceca0, handle=<optimized out>) at ../src/jit/jit-core.c:200
#16 0x00007ff7f00400d5 in irgen_op_package_init (op=5, g=0x1b36b3e8a40) at ../src/jit/jit-irgen.c:1951
#17 irgen_block (g=g@entry=0x1b36b3e8a40, block=block@entry=0) at ../src/jit/jit-irgen.c:2327
#18 0x00007ff7f0042885 in jit_irgen (f=0x1b36d224310) at ../src/jit/jit-irgen.c:2510
#19 0x00007ff7f004493f in jit_interp (f=f@entry=0x1b36d224310, args=args@entry=0x2fddfdfd0, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#20 0x00007ff7f003b486 in jit_link (j=0x1b364fceca0, handle=<optimized out>) at ../src/jit/jit-core.c:200
#21 0x00007ff7f00400d5 in irgen_op_package_init (op=8, g=0x1b36cb58180) at ../src/jit/jit-irgen.c:1951
#22 irgen_block (g=g@entry=0x1b36cb58180, block=block@entry=0) at ../src/jit/jit-irgen.c:2327
#23 0x00007ff7f0042885 in jit_irgen (f=0x1b36cb580f0) at ../src/jit/jit-irgen.c:2510
#24 0x00007ff7f004493f in jit_interp (f=f@entry=0x1b36cb580f0, args=args@entry=0x2fddfe580, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#25 0x00007ff7f003b486 in jit_link (j=0x1b364fceca0, handle=<optimized out>) at ../src/jit/jit-core.c:200
#26 0x00007ff7f00400d5 in irgen_op_package_init (op=3, g=0x1b365106460) at ../src/jit/jit-irgen.c:1951
#27 irgen_block (g=g@entry=0x1b365106460, block=block@entry=0) at ../src/jit/jit-irgen.c:2327
#28 0x00007ff7f0042885 in jit_irgen (f=0x1b365105a70) at ../src/jit/jit-irgen.c:2510
#29 0x00007ff7f004493f in jit_interp (f=f@entry=0x1b365105a70, args=args@entry=0x2fddfeb30, nargs=nargs@entry=1, backedge=backedge@entry=0, caller=caller@entry=0x0) at ../src/jit/jit-interp.c:1195
#30 0x00007ff7f003b486 in jit_link (j=0x1b364fceca0, handle=<optimized out>) at ../src/jit/jit-core.c:200
#31 0x00007ff7f003fcc5 in irgen_op_link_package (op=0, g=0x1b36b3303e0) at ../src/jit/jit-irgen.c:1967
#32 irgen_block (g=g@entry=0x1b36b3303e0, block=block@entry=0) at ../src/jit/jit-irgen.c:2330
#33 0x00007ff7f0042885 in jit_irgen (f=0x1b36aaa2350) at ../src/jit/jit-irgen.c:2510
#34 0x00007ff7f003b66f in jit_call_thunk (j=0x1b364fceca0, unit=unit@entry=0x1b3650e03a0, result=result@entry=0x2fddff208) at ../src/jit/jit-core.c:301
#35 0x00007ff7f002a8d1 in eval_do_fold (ex=0x1b364fce910, expr=0x1b365890c90) at ../src/eval.c:130
#36 0x00007ff7efff94a7 in object_rewrite (object=<optimized out>, ctx=ctx@entry=0x2fddff860) at ../src/object.c:660
[... cut ...]
#49 0x00007ff7effe7d64 in simplify_global (top=top@entry=0x1b365891d00, generics=generics@entry=0x1b364f2b750, ex=<optimized out>) at ../src/simp.c:1714
#50 0x00007ff7effe4943 in elab_top_level (arch=arch@entry=0x1b3606b0000, ctx=<optimized out>, ctx@entry=0x2fddffa90) at ../src/elab.c:1662
#51 0x00007ff7effe4c8d in elab (top=top@entry=0x1b3606b0000) at ../src/elab.c:1730
#52 0x00007ff7effa2be6 in elaborate (argv=0x1b3603dcc40, argc=7) at ../src/nvc.c:290
#53 process_command (argc=argc@entry=7, argv=argv@entry=0x1b3603dcc40) at ../src/nvc.c:1047
#54 0x00007ff7f008183b in main (argc=7, argv=0x1b3603dcc40) at ../src/nvc.c:1164
```

nickg, 2022-07-29:
I'm not sure if it's exactly the same issue but I managed to reproduce a similar crash with an instantiated generic package. Could you test again?

avelure, 2022-08-01:
The package was not generic, but it solved the issue.

nickg, 2022-08-02:
OK great. Did you actually manage to get your other tests running?

closed 2022-08-02 by nickg

avelure, 2022-08-03:
Yes, out of 30 tests only 3 have issues, I'll make some more MWEs soon. This is better stats than for GHDL.