nickg/nvc #499: fstdump is not being built in 1.6.2

daym, 2022-08-02:
`wave1` test is trying to use bin/fstdump, but it's not being built.

Manually invoking `make bin/fstdump` [1] before does not work either. Neither does `make fstdump` [2]. Not sure what's up with the automake files here.

Fiddling around with it more, doing `make CC=gcc -L../lib -lfst -I../thirdparty fstdump` manually also doesn't work[3].

[1] no target bin/fstdump
[2] cc     fstdump.c   -o fstdump
make: cc: No such file or directory

[3] gcc -I../thirdparty  -L../lib  fstdump.c -lfst -lz  -o fstdump
/tmp/guix-build-nvc-1.6.2.drv-0/source/test/../thirdparty/fstapi.c:5538: undefined reference to `fastlz_decompress'

grep -r didn't find any library including this symbol.

Hence this bug report.

nickg, 2022-08-02:
It should get built when you run `make check` as it's defined in `check_PROGRAMS` in `test/Makemodule.am`. How are you running the tests? Could you share the full output of `configure` as well as `make V=1 check`?

daym, 2022-08-02:
I build it as a Guix package in a nix build container, just like the previous version.

Full log output is this:

`
starting phase `set-SOURCE-DATE-EPOCH'
phase `set-SOURCE-DATE-EPOCH' succeeded after 0.0 seconds
starting phase `set-paths'
environment variable `PATH' set to `/gnu/store/d8l3aaq7vs2kzkwg0afj1lv1hb2s5fxm-automake-1.16.3/bin:/gnu/store/km9l89sd4wm9jp358481v2z6qacwl2h9-autoconf-2.69/bin:/gnu/store/b4mskl4py1zqmxdy1v260r3h6x5p92fm-flex-2.6.4/bin:/gnu/store/ny7ifgnkk7fbfhvpnhj32yg2mk6cdk8c-gettext-0.21/bin:/gnu/store/81n58x4293zwhvydzvvnypcz7y59jqmx-libtool-2.4.6/bin:/gnu/store/2b3blhwbag1ial0dhxw7wh4zjxl0cqpk-pkg-config-0.29.2/bin:/gnu/store/af9svhgvcy0xzphpv4fq17g8zm40x675-which-2.21/bin:/gnu/store/2hywg4rlq56v33xjw4ygxlcg6d18370n-check-0.15.2/bin:/gnu/store/dpkh77jsjcy8g87nj9y2xb0c1xi1mhki-llvm-9.0.1/bin:/gnu/store/g2ajyl8xk9aarxrgjbng2hkj3qm2v0z2-tar-1.34/bin:/gnu/store/iixwcv3k49ks1rf34pjgfzmzyhhgwng3-gzip-1.10/bin:/gnu/store/s3hl12jxz9ybs7nsy7kq7ybzz7qnzmsg-bzip2-1.0.8/bin:/gnu/store/c8isj4jq6knv0icfgr43di6q3nvdzkx7-xz-5.2.5/bin:/gnu/store/4ic6244i3ca4b4rxc2wnrgllsidyishv-file-5.39/bin:/gnu/store/ahmmvw21p11ik80lg1f953y7fd8bqkjm-diffutils-3.8/bin:/gnu/store/z39hnrwds1dgcbpfgj8dnv2cngjb2xbl-patch-2.7.6/bin:/gnu/store/39rsx3nl4c31952jybbjb8d6idr5hx7r-findutils-4.8.0/bin:/gnu/store/690qz3fg334dpwn3pn6k59n4wc943p2b-gawk-5.1.0/bin:/gnu/store/wxgv6i8g0p24q5gcyzd0yr07s8kn9680-sed-4.8/bin:/gnu/store/xjwp2hsd9256icjjybfrmznppjicywf6-grep-3.6/bin:/gnu/store/d251rfgc9nm2clzffzhgiipdvfvzkvwi-coreutils-8.32/bin:/gnu/store/55cbpsi18mahg131nmiya6km5b4mscfa-make-4.3/bin:/gnu/store/4y5m9lb8k3qkb1y9m02sw9w9a6hacd16-bash-minimal-5.1.8/bin:/gnu/store/s2pg5k98fl2g2szg9dykxyd9zl3xihv9-ld-wrapper-0/bin:/gnu/store/rc781v4k0drhaqn90xfwwpspki5x0bvf-binutils-2.37/bin:/gnu/store/069aq2v993kpc41yabp5b6vm4wb9jkhg-gcc-10.3.0/bin:/gnu/store/5h2w4qi9hk1qzzgi1w83220ydslinr4s-glibc-2.33/bin:/gnu/store/5h2w4qi9hk1qzzgi1w83220ydslinr4s-glibc-2.33/sbin:/gnu/store/gwrii9zfm1vl70cx3z16i0s5wbvng997-m4-1.4.18/bin'
environment variable `ACLOCAL_PATH' set to `/gnu/store/d8l3aaq7vs2kzkwg0afj1lv1hb2s5fxm-automake-1.16.3/share/aclocal:/gnu/store/ny7ifgnkk7fbfhvpnhj32yg2mk6cdk8c-gettext-0.21/share/aclocal:/gnu/store/81n58x4293zwhvydzvvnypcz7y59jqmx-libtool-2.4.6/share/aclocal:/gnu/store/2b3blhwbag1ial0dhxw7wh4zjxl0cqpk-pkg-config-0.29.2/share/aclocal:/gnu/store/2hywg4rlq56v33xjw4ygxlcg6d18370n-check-0.15.2/share/aclocal'
environment variable `PKG_CONFIG_PATH' set to `/gnu/store/2hywg4rlq56v33xjw4ygxlcg6d18370n-check-0.15.2/lib/pkgconfig:/gnu/store/jp6027624wl2f4xx5yz1vjzd2b9yvwl9-elfutils-0.183/lib/pkgconfig:/gnu/store/c8isj4jq6knv0icfgr43di6q3nvdzkx7-xz-5.2.5/lib/pkgconfig:/gnu/store/4ic6244i3ca4b4rxc2wnrgllsidyishv-file-5.39/lib/pkgconfig:/gnu/store/8qv5kb2fgm4c3bf70zcg9l6hkf3qzpw9-zlib-1.2.11/lib/pkgconfig'
environment variable `BASH_LOADABLES_PATH' unset
environment variable `C_INCLUDE_PATH' set to `/gnu/store/b4mskl4py1zqmxdy1v260r3h6x5p92fm-flex-2.6.4/include:/gnu/store/ny7ifgnkk7fbfhvpnhj32yg2mk6cdk8c-gettext-0.21/include:/gnu/store/2hywg4rlq56v33xjw4ygxlcg6d18370n-check-0.15.2/include:/gnu/store/jp6027624wl2f4xx5yz1vjzd2b9yvwl9-elfutils-0.183/include:/gnu/store/dpkh77jsjcy8g87nj9y2xb0c1xi1mhki-llvm-9.0.1/include:/gnu/store/s3hl12jxz9ybs7nsy7kq7ybzz7qnzmsg-bzip2-1.0.8/include:/gnu/store/c8isj4jq6knv0icfgr43di6q3nvdzkx7-xz-5.2.5/include:/gnu/store/4ic6244i3ca4b4rxc2wnrgllsidyishv-file-5.39/include:/gnu/store/690qz3fg334dpwn3pn6k59n4wc943p2b-gawk-5.1.0/include:/gnu/store/55cbpsi18mahg131nmiya6km5b4mscfa-make-4.3/include:/gnu/store/rc781v4k0drhaqn90xfwwpspki5x0bvf-binutils-2.37/include:/gnu/store/069aq2v993kpc41yabp5b6vm4wb9jkhg-gcc-10.3.0/include:/gnu/store/5h2w4qi9hk1qzzgi1w83220ydslinr4s-glibc-2.33/include:/gnu/store/8qv5kb2fgm4c3bf70zcg9l6hkf3qzpw9-zlib-1.2.11/include:/gnu/store/6mjww4iz4xdan74d5bbjfh7il8rngfkk-linux-libre-headers-5.10.35/include'
environment variable `CPLUS_INCLUDE_PATH' set to `/gnu/store/b4mskl4py1zqmxdy1v260r3h6x5p92fm-flex-2.6.4/include:/gnu/store/ny7ifgnkk7fbfhvpnhj32yg2mk6cdk8c-gettext-0.21/include:/gnu/store/2hywg4rlq56v33xjw4ygxlcg6d18370n-check-0.15.2/include:/gnu/store/jp6027624wl2f4xx5yz1vjzd2b9yvwl9-elfutils-0.183/include:/gnu/store/dpkh77jsjcy8g87nj9y2xb0c1xi1mhki-llvm-9.0.1/include:/gnu/store/s3hl12jxz9ybs7nsy7kq7ybzz7qnzmsg-bzip2-1.0.8/include:/gnu/store/c8isj4jq6knv0icfgr43di6q3nvdzkx7-xz-5.2.5/include:/gnu/store/4ic6244i3ca4b4rxc2wnrgllsidyishv-file-5.39/include:/gnu/store/690qz3fg334dpwn3pn6k59n4wc943p2b-gawk-5.1.0/include:/gnu/store/55cbpsi18mahg131nmiya6km5b4mscfa-make-4.3/include:/gnu/store/rc781v4k0drhaqn90xfwwpspki5x0bvf-binutils-2.37/include:/gnu/store/069aq2v993kpc41yabp5b6vm4wb9jkhg-gcc-10.3.0/include/c++:/gnu/store/069aq2v993kpc41yabp5b6vm4wb9jkhg-gcc-10.3.0/include:/gnu/store/5h2w4qi9hk1qzzgi1w83220ydslinr4s-glibc-2.33/include:/gnu/store/8qv5kb2fgm4c3bf70zcg9l6hkf3qzpw9-zlib-1.2.11/include:/gnu/store/6mjww4iz4xdan74d5bbjfh7il8rngfkk-linux-libre-headers-5.10.35/include'
environment variable `LIBRARY_PATH' set to `/gnu/store/b4mskl4py1zqmxdy1v260r3h6x5p92fm-flex-2.6.4/lib:/gnu/store/ny7ifgnkk7fbfhvpnhj32yg2mk6cdk8c-gettext-0.21/lib:/gnu/store/2hywg4rlq56v33xjw4ygxlcg6d18370n-check-0.15.2/lib:/gnu/store/jp6027624wl2f4xx5yz1vjzd2b9yvwl9-elfutils-0.183/lib:/gnu/store/dpkh77jsjcy8g87nj9y2xb0c1xi1mhki-llvm-9.0.1/lib:/gnu/store/s3hl12jxz9ybs7nsy7kq7ybzz7qnzmsg-bzip2-1.0.8/lib:/gnu/store/c8isj4jq6knv0icfgr43di6q3nvdzkx7-xz-5.2.5/lib:/gnu/store/4ic6244i3ca4b4rxc2wnrgllsidyishv-file-5.3 [... cut ...]
make: Leaving directory '/tmp/guix-build-nvc-1.6.2.drv-0/source/test'
error: in phase 'prepare-check': uncaught exception:
%exception #<&invoke-error program: "make" arguments: ("-C" "test" "CC=gcc" "CFLAGS=-I../thirdparty" "LDFLAGS=-L../lib" "LOADLIBES=-lfst -lz" "fstdump") exit-status: 2 term-signal: #f stop-signal: #f> 
phase `prepare-check' failed after 1.5 seconds
command "make" "-C" "test" "CC=gcc" "CFLAGS=-I../thirdparty" "LDFLAGS=-L../lib" "LOADLIBES=-lfst -lz" "fstdump" failed with status 2
`

nickg, 2022-08-02:
Looks like fstdump gets built ok:

```
gcc -DHAVE_CONFIG_H -I.  -I./thirdparty -I./src  -Wall  -flto -pthread -pthread  -g -O2 -MT test/fstdump.o -MD -MP -MF $depbase.Tpo -c -o test/fstdump.o test/fstdump.c &&\
mv -f $depbase.Tpo $depbase.Po
```

All the tests that are run through the shell are failing, not just the ones that have fstdump. E.g.

```
libdir1 : failed
Exec failed: No such file or directory
         vests2 : ok
         vests3 : ok
```

`bin/run_regr` tries to exec `/bin/sh` here. Is that available inside the guix environment?

daym, 2022-08-02:
>bin/run_regr tries to exec /bin/sh here. Is that available inside the guix environment?

No. I mean there is a sh in the PATH, but it always has a hash in the name. It's something like /gnu/store/y3y532hr3252332-bash/bin/sh

daym, 2022-08-02:
Huh, patching the /bin/sh out of run_regr.c makes it build just fine (all of nvc).

nickg, 2022-08-02:
OK, maybe run_regr should use `execvp` instead of hard coding the full path.

nickg, 2022-08-02:
> Yeah, but that's just the object file. The wave1 test tried to run an executable--and that one is missing.

Ah right, but the link step is a bit further down:

```
gcc -Wall  -flto -pthread -pthread  -g -O2  -L/gnu/store/dpkh77jsjcy8g87nj9y2xb0c1xi1mhki-llvm-9.0.1/lib  -lpthread  -o bin/fstdump test/fstdump.o lib/libfst.a lib/libfastlz.a -lz -lm -ldl 
```

daym, 2022-08-02:
>OK, maybe run_regr should use execvp instead of hard coding the full path.

Yeah, since it's only a test runner, taking sh from path is totally fine even here (since there's only the correct sh in PATH in the container anyway).

nickg, 2022-08-02:
I tried to fix this in a slightly different way in 062c763fd8. Now the configure script tries to detect the path to sh with a fall-back to `/bin/sh`. Can you see if that works in Guix? It should apply on the 1.6 branch too.

daym, 2022-08-03:
I tried the patch and it works just fine without any other workarounds with it.

nickg, 2022-08-03:
OK, good news. I'll do a 1.7 release sometime in the near future with that included.

closed 2022-08-03 by nickg