nickg/nvc #502: "** Fatal: missing vcode unit " occurs when elabolating VHDL.

ikwzm, 2022-08-10:
I'm currently trying to simulate a VHDL project with nvc.

The project is included at https://github.com/ikwzm/Merge_Sorter
You can download it as follows:

```console
shell$ git clone --branch nvc-issue-502 --depth=1 --recursive https://github.com/ikwzm/Merge_Sorter.git
Cloning into 'Merge_Sorter'...
remote: Enumerating objects: 395, done.        
remote: Counting objects: 100% (395/395), done.        
remote: Compressing objects: 100% (273/273), done.        
remote: Total 395 (delta 225), reused 240 (delta 114), pack-reused 0        
Receiving objects: 100% (395/395), 4.45 MiB | 13.35 MiB/s, done.
Resolving deltas: 100% (225/225), done.
Submodule 'Dummy_Plug' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug'
Submodule 'PipeWork' (https://github.com/ikwzm/PipeWork.git) registered for path 'PipeWork'
Cloning into '/mnt/d/ichiro/work/tmp/Merge_Sorter/Dummy_Plug'...
remote: Enumerating objects: 5478, done.        
remote: Counting objects: 100% (118/118), done.        
remote: Compressing objects: 100% (75/75), done.        
remote: Total 5478 (delta 51), reused 90 (delta 32), pack-reused 5360        
Receiving objects: 100% (5478/5478), 6.80 MiB | 9.27 MiB/s, done.
Resolving deltas: 100% (4391/4391), done.
Cloning into '/mnt/d/ichiro/work/tmp/Merge_Sorter/PipeWork'...
remote: Enumerating objects: 2730, done.        
remote: Counting objects: 100% (166/166), done.        
remote: Compressing objects: 100% (113/113), done.        
remote: Total 2730 (delta 101), reused 103 (delta 49), pack-reused 2564        
Receiving objects: 100% (2730/2730), 2.17 MiB | 15.11 MiB/s, done.
Resolving deltas: 100% (1995/1995), done.
Submodule path 'Dummy_Plug': checked out '1192b10219033972e418ae270244f72cc4edffbe'
Submodule 'docs' (https://github.com/ikwzm/Dummy_Plug.git) registered for path 'Dummy_Plug/docs'
Cloning into '/mnt/d/ichiro/work/tmp/Merge_Sorter/Dummy_Plug/docs'...
remote: Enumerating objects: 5478, done.        
remote: Counting objects: 100% (118/118), done.        
remote: Compressing objects: 100% (75/75), done.        
remote: Total 5478 (delta 51), reused 90 (delta 32), pack-reused 5360        
Receiving objects: 100% (5478/5478), 6.80 MiB | 6.71 MiB/s, done.
Resolving deltas: 100% (4391/4391), done.
Submodule path 'Dummy_Plug/docs': checked out 'a801570f3d1d46f52edd64d02ba9960fef2d5728'
Submodule path 'PipeWork': checked out 'fe8a74d327be76f020c67254f74369e444892319'
```

The project is sim/nvc/argsort_axi
You can recreate the error by running:

```console
shell$ cd Merge_Sorter/sim/nvc/argsort_axi/
shell$ make
sh analyze_libs.sh
nvc -L ./ -M 32M --work=WORK -e  ArgSort_AXI_Test_Bench_x04_w1_f0
** Fatal: missing vcode unit 
          WORK.ARGSORT_AXI_TEST_BENCH_X04_W1_F0.TB.DUT.ARGSORT_IF.MRG_RD.MRG(0).READER.CTRL.I2O.SYNC.SET_VEC_RANGE()105WORK.ARGSORT_AXI_TEST_BENCH_X04_W1_F0.TB.DUT.ARGSORT_IF.MRG_RD
Makefile:57: recipe for target 'WORK/_WORK.ARGSORT_AXI_TEST_BENCH_X04_W1_F0.elab.so' failed
make: *** [WORK/_WORK.ARGSORT_AXI_TEST_BENCH_X04_W1_F0.elab.so] Error 1
```

The version of nvc used is:

```console
shell$ nvc --version
nvc 1.8-devel (Using LLVM 7.0.0)
Copyright (C) 2011-2022  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```

nickg, 2022-08-11:
Thanks for the bug report. This particular error is fixed now but I found another problem when I ran the simulation, I'll look at that soon.

nickg, 2022-08-13:
This is fixed now!

closed 2022-08-13 by nickg