nickg/nvc #506: SIGSEGV in simulation

avelure, 2022-08-11:
I caught this while running some CI simulations on ubuntu20.04, but rerunning the test did not reproduce the fault.

```
*** Caught signal 11 (SIGSEGV) [address=(nil), ip=0x7f20e7736c04] ***
[0x5600f1a8de0f] ../src/util.c:781 signal_handler.lto_priv.0
[0x7f20ec1e508f] (/usr/lib/x86_64-linux-gnu/libc-2.31.so)
[0x7f20e7736c04] (/usr/lib/x86_64-linux-gnu/libLLVM-12.so.1) _ZN4llvm15ValueHandleBase14ValueIsDeletedEPNS_5ValueE
[0x7f20e77370e1] (/usr/lib/x86_64-linux-gnu/libLLVM-12.so.1) _ZN4llvm5ValueD1Ev
[0x7f20e76171cb] (/usr/lib/x86_64-linux-gnu/libLLVM-12.so.1)
[0x7f20e76b5123] (/usr/lib/x86_64-linux-gnu/libLLVM-12.so.1) _ZN4llvm8Function17dropAllReferencesEv
[0x7f20e770c7af] (/usr/lib/x86_64-linux-gnu/libLLVM-12.so.1) _ZN4llvm6Module17dropAllReferencesEv
[0x7f20e770c4b3] (/usr/lib/x86_64-linux-gnu/libLLVM-12.so.1) _ZN4llvm6ModuleD1Ev
[0x7f20e7649fcd] (/usr/lib/x86_64-linux-gnu/libLLVM-12.so.1) LLVMDisposeModule
[0x5600f19ffb9b] ../src/cgen.c:5290 cgen_worker_thread.lto_priv.0
[0x7f20e67e4608] (/usr/lib/x86_64-linux-gnu/libpthread-2.31.so) start_thread
[0x7f20ec2c1132] (/usr/lib/x86_64-linux-gnu/libc-2.31.so) __clone
Error: Program ended with exit code 2
```

nickg, 2022-08-11:
I've seen this once or twice before but I'm not sure what triggers it.

nickg, 2022-08-23:
I managed to reproduce this somewhat reliably using some code from @amb5l. It's going wrong somewhere in the LLVM optimiser so it's a bit difficult for me to debug. As a temporary workaround I've lowered the LLVM optimisation level slightly which seems to fix it for me, and I'm not sure it makes much practical difference to the execution speed anyway.

Please comment if you still see the crash with the commit above.

nickg, 2022-08-24:
> 
> Please comment if you still see the crash with the commit above.

Ignore that, it doesn't fix anything. The real fix is https://github.com/nickg/nvc/commit/71d5882e2f49fa17685855f4ef445d72d7ff9ad4

commit 71d5882e (2022-08-24): LLVMBasicBlockRef shared by multiple contexts. Fixes #506
commit 1becc0bf (2022-08-31): LLVMBasicBlockRef shared by multiple contexts. Fixes #506
closed 2022-08-24 by nickg