nickg/nvc #511: Split records in parameter map

avelure, 2022-08-11:
This is from the UVVM bitvis_vip_axistream axistream_bfm_pkg.vhd
It is splitting a procedure parameter that is a record in the parameter mapping.
The original error is `axistream_bfm_pkg.vhd:986:7: error: no possible overload of AXISTREAM_TRANSMIT_BYTES has formal AXISTREAM_IF`
I didn't manage to recreate the same error in an MWE, but this one also fails with a non-implemented error.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
  generic(val : natural);
end entity test;
architecture beh of test is
  type t_rec is record
    sig1 : natural;
    sig2 : natural;
  end record;
  procedure proc(sig_if : t_rec) is
  begin
  end procedure;
  
begin

  process
  begin
    proc(sig_if.sig1 => 1, sig_if.sig2 => 2);
    wait;
  end process;

end architecture beh;
```
```
nvc --std=08 -a test_proc_split_args.vhd -e test-beh -r
** Error: sorry, this form of parameter name is not yet supported
    > test_proc_split_args.vhd:21
    |
 21 |     proc(sig_if.sig1 => 1, sig_if.sig2 => 2);
    |          ^^^^^^^^^^^
```

nickg, 2022-09-19:
I think this should be implemented now, could you test again?

avelure, 2022-09-28:
Yes, this seems to have solved it, thanks!

closed 2022-09-29 by nickg