nickg/nvc #512: missing class_of for T_FCALL

avelure, 2022-08-11:
This might be a follow error due to the file in #511 not compiling, but I guess it is not supposed to fatal on incomplete designs anyway.
This is for the UVVM bitvis_vip_axistream and the axistream_vvc.vhd file.

```
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:27:1: error: unit WORK.AXISTREAM_BFM_PKG not found in library WORK
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:28:1: error: unit WORK.VVC_METHODS_PKG not found in library WORK
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:29:1: error: unit WORK.VVC_CMD_PKG not found in library WORK
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:30:1: error: unit WORK.TD_TARGET_SUPPORT_PKG not found in library WORK
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:31:1: error: unit WORK.TD_VVC_ENTITY_SUPPORT_PKG not found in library WORK
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:32:1: error: unit WORK.TD_CMD_QUEUE_PKG not found in library WORK
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:33:1: error: unit WORK.TD_RESULT_QUEUE_PKG not found in library WORK
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:34:1: error: unit WORK.TRANSACTION_PKG not found in library WORK
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:68:1: error: design unit depends on BITVIS_VIP_AXISTREAM.AXISTREAM_VVC which was analysed with errors
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:79:35: error: cannot find unit BITVIS_VIP_AXISTREAM.TD_CMD_QUEUE_PKG
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:80:35: error: cannot find unit BITVIS_VIP_AXISTREAM.TD_RESULT_QUEUE_PKG
c:/proj/uvvm/bitvis_vip_axistream/src/axistream_vvc.vhd:82:50: error: no visible subprogram declaration for SHARED_AXISTREAM_VVC_CONFIG
fatal: missing class_of for T_FCALL
[00007FF76A6165B0]
[00007FF76A65CB7A]
[00007FF76A6927EE]
[00007FF76A693F00]
[00007FF76A6960F7]
[00007FF76A62EBF8]
[00007FF76A636D3B]
[00007FF76A6381C9]
[00007FF76A63AED8]
[00007FF76A612A3C]
[00007FF76A6F2CFB] vhpi_put_data+0x138eb
[00007FF76A6113AE]
[00007FF76A6114E6]
[00007FFA7BCA7034] BaseThreadInitThunk+0x14
[00007FFA7D102651] RtlUserThreadStart+0x21

Thread 1 hit Breakpoint 1, 0x00007ffa7adb03e0 in ucrtbase!exit () from /c/Windows/System32/ucrtbase.dll
(gdb) bt
#0  0x00007ffa7adb03e0 in ucrtbase!exit () from /c/Windows/System32/ucrtbase.dll
#1  0x00007ff76a615198 in fatal_exit (status=1) at ../src/util.c:450
#2  0x00007ff76a6165ba in fatal_trace (fmt=fmt@entry=0x7ff76a7055a5 <has_map+25285> "missing class_of for %s") at ../src/util.c:527
#3  0x00007ff76a65cb7a in class_of (t=<optimized out>) at ../src/common.c:522
#4  0x00007ff76a6927ee in name_mask_for (t=t@entry=0x14ecf051570) at ../src/names.c:639
#5  0x00007ff76a693f00 in make_visible (s=0x14ece385540, name=0x14ece2f7a50, decl=0x14ecf051570, kind=DIRECT, origin=0x14ece385540) at ../src/names.c:719
#6  0x00007ff76a6960f7 in make_visible_slow (decl=0x14ecf051570, name=0x14ec5140000, s=0x1) at ../src/names.c:915
#7  insert_name (tab=0x14ecd2f1d80, decl=decl@entry=0x14ecf051570, alias=alias@entry=0x0) at ../src/names.c:1116
#8  0x00007ff76a62ebf8 in p_alias_declaration () at ../src/parse.c:6490
#9  0x00007ff76a636d3b in p_block_declarative_item (parent=parent@entry=0x14ecf050000) at ../src/parse.c:8215
#10 0x00007ff76a6381c9 in p_architecture_declarative_part (arch=0x14ecf050000) at ../src/parse.c:9777
#11 p_architecture_body (unit=0x14ecf050000) at ../src/parse.c:9842
#12 p_secondary_unit (unit=unit@entry=0x14ecf050000) at ../src/parse.c:10026
#13 0x00007ff76a63aed8 in p_library_unit (unit=0x14ecf050000) at ../src/parse.c:10052
#14 p_design_unit () at ../src/parse.c:10108
#15 parse () at ../src/parse.c:10125
#16 0x00007ff76a612a3c in analyse (argv=0x14ec514a6c0, argc=4) at ../src/nvc.c:157
#17 process_command (argc=argc@entry=4, argv=argv@entry=0x14ec514a6c0) at ../src/nvc.c:1045
#18 0x00007ff76a6f2cfb in main (argc=4, argv=0x14ec514a6c0) at ../src/nvc.c:1164
```

nickg, 2022-08-12:
Is this on the very latest master? I think I fixed this recently in 5a5caa1.

avelure, 2022-08-12:
Ah, you are right. Sorry, I didn't keep up.

closed 2022-08-12 by avelure