nickg/nvc #525: MSYS2 MinGW compilation warning

amb5l, 2022-08-25:
When compiling with **gcc** on MSYS2, the following warning appears:

`../src/rt/rtkern.c:2241:13: warning: unknown conversion type character 'z' in format`

This [Stack Overflow post](https://stackoverflow.com/questions/59147875/how-do-i-work-around-the-unknown-conversion-type-character-z-in-format-compi) may be helpful.

nickg, 2022-08-25:
Looks like Microsoft's C library does support `%zu`: https://docs.microsoft.com/en-us/cpp/c-runtime-library/format-specification-syntax-printf-and-wprintf-functions?view=msvc-170#size-prefixes-for-printf-and-wprintf-format-type-specifiers

amb5l, 2022-08-25:
From one of the comments on Stack Overflow: "MinGW uses microsoft's c library msvcrt-- this library only conforms to c89 and doesn't understand z". Another says "You can instead link MinGW-provided alternative stdio functions that support C11 with #define __USE_MINGW_ANSI_STDIO 1. I prefer not to link extra code if I can get around the peculiarities of msvcrt, but that's of course a matter of taste." I've had a look at the Makefile but am out of my depth!

nickg, 2022-08-25:
Perhaps it's a difference between the original msvcrt and the new UCRT, and GCC hasn't been updated for UCRT.

avelure, 2022-08-25:
The warning is also present for UCRT version of MINGW.

Here's a full build log
```
  CC       src/nvc.o
  CC       src/lib.o
  CC       src/util.o
../src/util.c: In function 'xmalloc_flex':
../src/util.c:165:52: warning: unknown conversion type character 'z' in format [-Wformat=]
  165 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                                                    ^
../src/util.c:165:58: warning: unknown conversion type character 'z' in format [-Wformat=]
  165 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                                                          ^
../src/util.c:165:19: warning: too many arguments for format [-Wformat-extra-args]
  165 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/util.c: In function 'xcalloc_flex':
../src/util.c:188:52: warning: unknown conversion type character 'z' in format [-Wformat=]
  188 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                                                    ^
../src/util.c:188:58: warning: unknown conversion type character 'z' in format [-Wformat=]
  188 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                                                          ^
../src/util.c:188:19: warning: too many arguments for format [-Wformat-extra-args]
  188 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/util.c: In function 'xrealloc_array':
../src/util.c:211:52: warning: unknown conversion type character 'z' in format [-Wformat=]
  211 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                                                    ^
../src/util.c:211:58: warning: unknown conversion type character 'z' in format [-Wformat=]
  211 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                                                          ^
../src/util.c:211:19: warning: too many arguments for format [-Wformat-extra-args]
  211 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/util.c: In function 'xrealloc_flex':
../src/util.c:221:52: warning: unknown conversion type character 'z' in format [-Wformat=]
  221 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                                                    ^
../src/util.c:221:58: warning: unknown conversion type character 'z' in format [-Wformat=]
  221 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                                                          ^
../src/util.c:221:19: warning: too many arguments for format [-Wformat-extra-args]
  221 |       fatal_trace("array size overflow: requested %zd * %zd bytes",
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC       src/ident.o
  CC       src/parse.o
  LEX      src/lexer.c
  CC       src/lexer.o
  CC       src/tree.o
  CC       src/type.o
  CC       src/sem.o
  CC       src/elab.o
  CC       src/simp.o
  CC       src/dump.o
  CC       src/common.o
../src/common.c: In function 'mangle_one_type':
../src/common.c:1423:24: warning: unknown conversion type character 'z' in format [-Wformat=]
 1423 |       tb_printf(buf, "%zu", ident_len(ident));
      |                        ^
../src/common.c:1423:22: warning: too many arguments for format [-Wformat-extra-args]
 1423 |       tb_printf(buf, "%zu", ident_len(ident));
      |                      ^~~~~
  CC       src/fbuf.o
  CC       src/hash.o
  CC       src/bounds.o
  CC       src/make.o
  CC       src/object.o
[... 929 lines cut ...]
C:/msys64/ucrt64/include/llvm-c/Core.h:3908:1: note: declared here
 3908 | LLVM_ATTRIBUTE_C_DEPRECATED(
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/cgen.c:4240:4: warning: 'LLVMBuildStructGEP' is deprecated [-Wdeprecated-declarations]
 4240 |    LLVMValueRef state_ptr = LLVMBuildStructGEP(builder, priv_ptr, 1, "");
      |    ^~~~~~~~~~~~
C:/msys64/ucrt64/include/llvm-c/Core.h:3908:1: note: declared here
 3908 | LLVM_ATTRIBUTE_C_DEPRECATED(
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/cgen.c:4243:4: warning: 'LLVMBuildStructGEP' is deprecated [-Wdeprecated-declarations]
 4243 |    LLVMValueRef pcall_ptr = LLVMBuildStructGEP(builder, priv_ptr, 2, "");
      |    ^~~~~~~~~~~~
C:/msys64/ucrt64/include/llvm-c/Core.h:3908:1: note: declared here
 3908 | LLVM_ATTRIBUTE_C_DEPRECATED(
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/cgen.c: In function 'cgen_reset_function':
../src/cgen.c:4341:4: warning: 'LLVMBuildStructGEP' is deprecated [-Wdeprecated-declarations]
 4341 |    LLVMValueRef context_ptr = LLVMBuildStructGEP(builder, ctx.state, 0, "");
      |    ^~~~~~~~~~~~
C:/msys64/ucrt64/include/llvm-c/Core.h:3908:1: note: declared here
 3908 | LLVM_ATTRIBUTE_C_DEPRECATED(
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  AR       lib/libcgen.a
  CCLD     bin/nvc.exe
```