nickg/nvc #530: Several tests fail on FreeBSD

yurivict, 2022-09-01:
See the log: [nvc-tests.log](https://github.com/nickg/nvc/files/9466839/nvc-tests.log)

Version: 1.7.1

nickg, 2022-09-01:
All of the failing tests are missing some expected debug output. At the same time there is this error from libdwarf:

```
** Warning: dwarf_init: /disk-samsung/freebsd-ports/cad/nvc/work/nvc-r1.7.1/.build/bin/nvc: ELF error : Request error: invalid ELF_C_* argument [dwarf_init(112)]
** Warning: dwarf_init: /tmp/nvco6Fa26/work/_WORK.DEBUG3.elab.so: ELF error : Request error: invalid ELF_C_* argument [dwarf_init(112)]
```

I think FreeBSD has its own implementation of libdwarf (i.e. not the SGI-derived one).

Which version of FreeBSD is this? I tried the 1.7 branch on FreeBSD 13.1 and the tests passed: https://builds.sr.ht/~nickg/job/836667

yurivict, 2022-09-01:
I use FreeBSD 13.1-STABLE.

nickg, 2022-10-13:
How can I reproduce this? I tried `sudo make` and `sudo make test` in `/usr/ports/cad/nvc` on a FreeBSD 13.1 system but all the tests passed:

```
PASS: bin/run_regr
==================
All 2 tests passed
==================
[build@build /usr/ports/cad/nvc]$ uname -a
FreeBSD build 13.1-RELEASE-p2 FreeBSD 13.1-RELEASE-p2 GENERIC amd64
```

yurivict, 2023-04-09:
All tests pass in 1.9.0.

Thank you.

closed 2023-04-09 by yurivict