nickg/nvc #531: Minor: UVVM install script compiles examples as well as libs and VIPs

amb5l, 2022-09-05:
The UVVM install script builds the bitvis_irqc and bitvis_uart examples as precompiled libraries, unlike GHDL.

I believe that the intention of the UVVM developers is that the DUT is compiled at simulation time into a DUT library, the test harness / test bench are compiled into work, and they then reference the DUT library. If NVC gives local libraries precedence over precompiled libs then the NVC script behaviour is harmless.

Feature request: alter UVVM install script to read UVVM "compile_order.txt" files like GHDL's compile-uvvm.sh script to reduce future maintenance of NVC to keep up with UVVM changes. I will have a go at this.

nickg, 2022-09-05:
OK, sounds good.

amb5l, 2022-09-19:
This script is a potential alternative to NVC's `contrib/install-uvvm.sh` that pulls the list of files to be compiled from UVVM's `component_list.txt` and `compile_order.txt` files. Note that it can take a tag as a argument to check out a specific commit of UVVM.
```
#!/usr/bin/env bash
#
# Called by "nvc --install uvvm".
#
# Arguments:
#  $1 = UVVM repository tag (optional) e.g. v2022.05.25

. $(dirname $BASH_SOURCE)/functions.sh

if [ -z "$1" ]; then
    branch="master"
fi

git_wrapper https://github.com/UVVM/UVVM $branch

STD=2008
A_OPTS="--relaxed"

component_list=$(tr -d '\r' <script/component_list.txt | tr '\n' ' ')
for component_name in $component_list; do
    echo "################################################################################"
    echo "compiling: $component_name"
    echo "################################################################################"
    cd $component_name/script
        readarray -t compile_order < compile_order.txt
        first_line=(${compile_order[0]})
        library_name=${first_line[2]}
        unset compile_order[0]
        lines=$( IFS=$'\n'; echo "${compile_order[*]}" )
        echo "$lines"
        analyse_list $component_name <<<$lines
    cd ../..
done
```

amb5l, 2022-09-19:
Note you can drop the `echo "$lines"` from the above.

closed 2022-09-19 by nickg

amb5l, 2022-09-19:
Great work. I tested https://github.com/nickg/nvc/commit/813728d00c468838b97db35f1d1e6d55226dbe39 against https://github.com/UVVM/UVVM/commit/fd7f5056a4d2dd1547414dcc2bb89baa2a451c43 and it works perfectly.

Totally agree with the default branch.

commit 813728d0 (2022-09-19): Use UVVM component_list.txt in install-uvvm.sh. Fixes #531