nickg/nvc #532: Error when overloading procedures with record parameters

amb5l, 2022-09-07:
The code below produces an error when analysed:
** Error: no possible overload of TEST_PROC has formal T

This is a minimal example of a problem I encountered analysing some of the sources in UVVM.

```
library ieee;
use ieee.std_logic_1164.all;

package test_pkg is

    type t_test_rec is record
        a  : std_logic_vector;
        b  : std_logic_vector;
    end record t_test_rec;

    -- first declaration
    procedure test_proc(
        t : in t_test_rec
    );

    -- overload
    procedure test_proc(
        a : in std_logic_vector;
        b : in std_logic_vector
    );

end package test_pkg;

package body test_pkg is

    -- first declaration
    procedure test_proc(
        t : in t_test_rec
    ) is
    begin
        -- do something
    end procedure test_proc;

    -- overload
    procedure test_proc(
        a : in std_logic_vector;
        b : in std_logic_vector
    ) is
    begin
        test_proc(
            t.a => a,
            t.b => b
        );
    end procedure test_proc;

end package body test_pkg;
```

nickg, 2022-09-08:
I think this is basically the same as #511 - individually associating record elements in subprogram calls doesn't work quite right at the moment.

amb5l, 2022-09-08:
Agreed, apologies for not spotting #511. Please feel free to close this issue.

nickg, 2022-09-19:
This should be fixed now. Would you mind testing UVVM again?

amb5l, 2022-09-19:
I checked out https://github.com/nickg/nvc/commit/846d0aa23e39bf8064341e83ad800151444febb5 and tested it against UVVM.

The reported error seems to have been resolved, however the compilation threw a new error that seems to be associated with strings:
```
nvc --std=2008 --work=/c/work/.nvc/lib/bitvis_vip_clock_generator -L/c/work/.nvc/lib -a --relaxed ../src/clock_generator_vvc.vhd
** Error: index range of array aggregate with others choice cannot be determined from the context
     > ../src/clock_generator_vvc.vhd:110
     |
 110 |     clock_name      := (others => NUL);
     |                         ^^^^^^^^^^^^^

```

nickg, 2022-09-19:
That error has been there for a while (see #509) but the old install-uvvm.sh used to patch it out. GHDL also raises an error for this (I believe it's invalid VHDL) but allows it with `-frelaxed`. I probably ought to be pragmatic and do the same.

closed 2022-09-19 by nickg