nickg/nvc #534: Broken on armv7: no member named '_gregs' in 'mcontext_t'; did you mean '__gregs'?

yurivict, 2022-09-09:
```
/wrkdirs/usr/ports/cad/nvc/work/nvc-r1.7.1/thirdparty/cpustate.c:117:30: error: no member named '_gregs' in 'mcontext_t'; did you mean '__gregs'?
   cpu->pc = uc->uc_mcontext._gregs[_REG_PC];
                             ^~~~~~
                             __gregs
/usr/include/machine/ucontext.h:77:14: note: '__gregs' declared here
        __gregset_t     __gregs;
                        ^
/wrkdirs/usr/ports/cad/nvc/work/nvc-r1.7.1/thirdparty/cpustate.c:118:30: error: no member named '_gregs' in 'mcontext_t'; did you mean '__gregs'?
   cpu->sp = uc->uc_mcontext._gregs[_REG_SP];
                             ^~~~~~
                             __gregs
/usr/include/machine/ucontext.h:77:14: note: '__gregs' declared here
        __gregset_t     __gregs;
                        ^
/wrkdirs/usr/ports/cad/nvc/work/nvc-r1.7.1/thirdparty/cpustate.c:121:38: error: no member named '_gregs' in 'mcontext_t'; did you mean '__gregs'?
      cpu->regs[i] = uc->uc_mcontext._gregs[i];
                                     ^~~~~~
                                     __gregs
/usr/include/machine/ucontext.h:77:14: note: '__gregs' declared here
        __gregset_t     __gregs;
                        ^
3 errors generated.
```

Version: 1.7.1
FreeBSD 13.1

nickg, 2022-09-19:
Yes it's a simple typo, unfortunately I didn't have access to a FreeBSD/arm machine to test. I'll fix it for the 1.7.2 release.

nickg, 2022-10-16:
This is fixed in the 1.7.2 release.

closed 2022-10-16 by nickg