nickg/nvc #536: Crash when outputting VCD

amb5l, 2022-09-26:
I encounted a crash when outputting a VCD from a simulation that works otherwise:
```
C:\work\tyto2\build\tb_saa5050>make nvc gtkwave
/mingw64/bin/nvc  --std=2008 --work=work -a  --relaxed C:/work/tyto2/src/common/tyto_types_pkg.vhd C:/work/tyto2/src/common/retro/saa5050/saa5050_rom_data.vhd C:/work/tyto2/src/common/retro/saa5050/saa5050.vhd C:/work/tyto2/src/common/retro/hd6845/hd6845.vhd C:/work/tyto2/src/common/tyto_sim_pkg.vhd C:/work/tyto2/src/common/retro/saa5050/test/tb_saa5050.vhd
/mingw64/bin/nvc  --std=2008 --work=work -e tb_saa5050  -ginfile=C:/work/tyto2/src/common/retro/saa5050/test/engtest.bin -goutfile=C:/work/tyto2/build/tb_saa5050//engtest
/mingw64/bin/nvc  --std=2008 --work=work -r tb_saa5050  --ieee-warnings=off --format=vcd --wave=wave1.vcd

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=000000bde7a00000, ip=00007ffb328e7e44] ***

[00007ff62eac69c4]
[00007ffb30995e31] UnhandledExceptionFilter+0x1f1
[00007ffb32deabac] memset+0x2ec
[00007ffb32dd3b36] _C_specific_handler+0x96
[00007ffb32de8b1f] _chkstk+0x12f
[00007ffb32d75b9a] RtlRestoreContext+0x8aa
[00007ffb32de7b2e] KiUserExceptionDispatcher+0x2e
[00007ffb328e7e44] memcpy+0x104
[00007ff62eb6dade] _nvc_mspace_alloc+0x3b8e
[00007ff62eb4f224]
[00007ff62eb4fdf4]
[00007ff62eb502ed]
[00007ff62eb5047d]
[00007ff62eac29c6]
[00007ff62eba51b2] vhpi_put_data+0x239f2
[00007ff62eac13ae]
[00007ff62eac14e6]
[00007ffb30e654e0] BaseThreadInitThunk+0x10
[00007ffb32d4485b] RtlUserThreadStart+0x2b
make: *** [C:/work/tyto2/submodules/multisim-mk/multisim.mk:239: sim] Segmentation fault
```
To reproduce, clone this commit: https://github.com/amb5l/tyto2/commit/a20687a9543b5c3e4efee2e4917a3fe2fc591fa2, navigate to the `build/tb_saa5050` directory and try both `make nvc` and `make nvc gtkwave`.

nickg, 2022-09-26:
It's caused by the array of integers 

```vhdl
    signal ttx_data  : uint8_array_t(0 to 1023);
```

closed 2022-09-26 by nickg

amb5l, 2022-09-27:
Confirmed - sorry I missed this commit (must have been by minutes!).

commit 09220028 (2022-09-26): FST dumping for arrays of integer. Fixes #536