nickg/nvc #537: Minor: install-vivado.sh on Windows

amb5l, 2022-09-27:
On Windows, Vivado's `settings.sh` and `settings.bat` both set **XILINX_VIVADO** to a DOS/Windows path, which causes NVC's `install-vivado.sh` to fail. I work around this by setting **XILINX_VIVADO** by hand.

Changing line 29 of `install-vivado.sh` from...
`src=$XILINX_VIVADO/data/vhdl/src`
...to...
`src=$(echo "/$XILINX_VIVADO" | sed -e 's/\\/\//g' -e 's/://' -e 's,//,/,g')/data/vhdl/src`
...handles both path formats on Windows and *should* work on Linux.

nickg, 2022-09-27:
Can we do the same as in the Quartus script?

```
if [ -n "$MSYSTEM" ]; then
  XILINX_VIVADO=$(cygpath -u $XILINX_VIVADO)
fi
```

amb5l, 2022-09-27:
D'oh. Yes - that works fine.

nickg, 2022-09-27:
I've just added this, could you test?

amb5l, 2022-09-27:
Tested and working. Thanks!

closed 2022-09-27 by nickg

amb5l, 2022-09-29:
The sed recipe for this might actually make sense. MSYSTEM is defined in the MSYS2 shell, but is usually not in a Windows Command Prompt. Restricting `nvc --install` operations to an MSYS2 shell is no hardship at all, but it would be nice to loosen that restriction. If you're open to this, I'll do a bit more testing to make sure, then maybe you could tweak both the Vivado and Quartus scripts?