nickg/nvc #539: "tree kind T_CONFIGURATION does not have item I_RANGES"

alphanumericnonsense, 2022-09-27:
Context:
The AUR package (1.6.2) didn't support some `--std=08` things like
```
type slv_1_t is array (natural range <>) of std_logic_vector;
type slv_2_t is array (natural range <>,natural range <>) of std_logic_vector;
```
so I installed 1.7.1 (which seemed to go OK).

Trying to analyze my files now gave an error:
> ** Fatal: tree kind T_CONFIGURATION does not have item I_RANGES
[0x55c11f28c27a] ../src/util.c:536 fatal_trace
[0x55c11f2def9b] ../src/object.c:192 object_lookup_failed
[0x55c11f2b847b] ../src/tree.c:1008 tree_ranges
[0x55c11f3496ec] ../src/lower.c:889 lower_constraints.constprop.0
[0x55c11f2e4cfd] ../src/lower.c:492 lower_array_stride.lto_priv.0
[0x55c11f2e4bca] ../src/lower.c:526 lower_array_total_len.lto_priv.0
[0x55c11f2fbfe2] ../src/lower.c:6260 lower_decls
[0x55c11f2fb03b] ../src/lower.c:8658 lower_decls
[0x55c11f2fe27b] ../src/lower.c:9564 lower_package
[0x55c11f2fe27b] ../src/lower.c:9594 lower_unit
[0x55c11f291ab1] ../src/nvc.c:166 analyse
[0x55c11f29260e] ../src/nvc.c:1043 process_command
[0x55c11f28b0a0] ../src/nvc.c:1162 main

I have no idea what this means, so I thought I'd put it here.

nickg, 2022-09-28:
Is is possible to share the file that causes the crash? If you can't share the whole file it would be helpful if you could try to narrow it down to something minimal the reproduces the bug (just delete bits of the file until it doesn't crash any more). Did you try with the latest master branch?

alphanumericnonsense, 2022-09-28:
I can't share the file(s).  I could try to narrow it down, but the error output means nothing to me (without going into the src listed).  I've pointed to what an earlier version 1.6.2 choked on (unconstrained array elements), although that may be unrelated.  The files simulate just fine in GHDL (2.0) and ModelSim.

Sorry I can't be more helpful at the moment.

nickg, 2022-10-03:
I improved the message so it shows the location of the error. Please re-test with the latest master branch if you are able, and see if that gives some clue.

alphanumericnonsense, 2022-10-03:
> I improved the message so it shows the location of the error. Please re-test with the latest master branch if you are able, and see if that gives some clue.

I added some new error messages to the original post and code for which the error occurs.  I think the error could be associated with the `'subtype` attribute.

nickg, 2022-10-04:
Thanks for the small test case. It's a problem with the implementation of `'subtype`.

nickg, 2022-10-16:
I think this should be fixed now, could you test again?

alphanumericnonsense, 2022-10-17:
It gets further than it did before but chokes on something else, perhaps an unrelated separate issue:
```
** Fatal: attempt to add to already finished block 12
[0x5606945a04be] ../src/util.c:540 fatal_trace
[0x56069461e86d] ../src/vcode.c:267 vcode_add_op
[0x56069461e86d] ../src/vcode.c:262 vcode_add_op.lto_priv.0
[0x5606946259e1] ../src/vcode.c:4950 emit_case
[0x5606946259e1] ../src/vcode.c:4940 emit_case
[0x560694613c3a] ../src/lower.c:6038 lower_case_array
[0x560694611f52] ../src/lower.c:6048 lower_case
[0x560694611f52] ../src/lower.c:6245 lower_stmt.lto_priv.0
[0x560694611609] ../src/lower.c:5623 lower_for
[0x560694611609] ../src/lower.c:6235 lower_stmt.lto_priv.0
[0x560694610f97] ../src/lower.c:5703 lower_sequence
[0x560694610f97] ../src/lower.c:6238 lower_stmt.lto_priv.0
[0x5606946133ad] ../src/lower.c:5814 lower_case_scalar
[0x560694610f65] ../src/lower.c:6046 lower_case
[0x560694610f65] ../src/lower.c:6245 lower_stmt.lto_priv.0
[0x560694611609] ../src/lower.c:5623 lower_for
[0x560694611609] ../src/lower.c:6235 lower_stmt.lto_priv.0
[0x560694611609] ../src/lower.c:5623 lower_for
[0x560694611609] ../src/lower.c:6235 lower_stmt.lto_priv.0
[0x5606946194a8] ../src/lower.c:8716 lower_decls
[0x56069461d43d] ../src/lower.c:9614 lower_package
[0x56069461d43d] ../src/lower.c:9644 lower_unit
[0x5606945a6aa1] ../src/nvc.c:169 analyse
[0x5606945a6aa1] ../src/nvc.c:1100 process_command
[0x56069459ef59] ../src/nvc.c:1219 main
```
Sorry I can't share the source.

nickg, 2022-10-20:
This one should be fixed too now. Thanks again for the test case.

alphanumericnonsense, 2022-10-20:
OK, everything I had analyzed, but when I went to elaborate the testbench, I encountered an error (perhaps unrelated):
```
** Fatal: null type must be file, access, context, or pointer
[0x563c75d884ee] ../src/util.c:540 fatal_trace
[0x563c75e0df31] ../src/vcode.c:5051 emit_null
[0x563c75dfdcf8] ../src/lower.c:6577 lower_sub_signals.lto_priv.0
[0x563c75e00fc1] ../src/lower.c:6722 lower_decls
[0x563c75e04afe] ../src/lower.c:9577 lower_concurrent_block
[0x563c75e04d82] ../src/lower.c:9587 lower_concurrent_block
[0x563c75e04d82] ../src/lower.c:9587 lower_concurrent_block
[0x563c75e054c7] ../src/lower.c:9609 lower_elab
[0x563c75e054c7] ../src/lower.c:9661 lower_unit
[0x563c75d8ee08] ../src/nvc.c:297 elaborate
[0x563c75d8ee08] ../src/nvc.c:1102 process_command
[0x563c75d86f89] ../src/nvc.c:1219 main
```
The last thing before the error is pointing to a record type, something like:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

package my_package is
	type my_type_t is record
		state : std_logic_vector;
		aux : std_logic_vector;
	end record my_type_t;
end package trlraes_v1_package;
```

nickg, 2022-10-23:
I've just pushed a fix for a similar crash that had the same stack trace. Could you test again and see if it resolves your issue too?

alphanumericnonsense, 2022-10-24:
Analysis OK, elaborating testbench gives new error:
```
** Fatal: cannot use uarray left with non-uarray type
[0x56068fe3544e] ../src/util.c:540 fatal_trace
[0x56068feb8ad1] ../src/vcode.c:4342 emit_uarray_op.lto_priv.0
[0x56068fea0531] ../src/vcode.c:4356 lower_array_slice
[0x56068fe9e7d9] ../src/lower.c:4174 lower_expr.lto_priv.0
[0x56068feaafcd] ../src/lower.c:4197 lower_driver_target
[0x56068feaafcd] ../src/lower.c:8767 lower_driver_target
[0x56068fe9508c] ../src/object.c:599 object_visit.part.0
[0x56068fe9518d] ../src/object.c:553 object_visit.part.0
[0x56068fe9518d] ../src/object.c:553 object_visit.part.0
[0x56068fe9518d] ../src/object.c:553 object_visit.part.0
[0x56068feb18af] ../src/object.c:553 lower_concurrent_block
[0x56068feb1942] ../src/lower.c:9588 lower_concurrent_block
[0x56068feb1942] ../src/lower.c:9588 lower_concurrent_block
[0x56068feb2087] ../src/lower.c:9610 lower_elab
[0x56068feb2087] ../src/lower.c:9662 lower_unit
[0x56068fe3bef8] ../src/nvc.c:297 elaborate
[0x56068fe3bef8] ../src/nvc.c:1104 process_command
[0x56068fe33f8b] ../src/nvc.c:1221 main
```

nickg, 2023-08-13:
I haven't been able to reproduce this and the code has changed quite a lot recently so I don't think there's much point leaving this open. If you're still interested please re-test with the latest master branch and open a new bug if it crashes.

closed 2023-08-13 by nickg