nickg/nvc #544: fatal: invalid type kind T_ARRAY in range_of

avelure, 2022-09-29:
This issue is probably partly related to #519.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  type t_axi_wr_slave_in_if is record
    awid      : std_logic_vector;
    awaddr    : std_logic_vector;
    awlen     : std_logic_vector(7 downto 0);
    awsize    : std_logic_vector(2 downto 0);
    awburst   : std_logic_vector(1 downto 0);
    awlock    : std_logic;
    awcache   : std_logic_vector(3 downto 0);
    awprot    : std_logic_vector(2 downto 0);
    awqos     : std_logic_vector(3 downto 0);
    awregion  : std_logic_vector(3 downto 0);
    awuser    : std_logic_vector;
    awvalid   : std_logic;
    wdata   : std_logic_vector;
    wstrb   : std_logic_vector;
    wlast   : std_logic;
    wuser   : std_logic_vector;
    wvalid  : std_logic;
    bready  : std_logic;
  end record;
  type t_axi_wr_slave_out_if is record
    awready : std_logic;
    wready  : std_logic;
    bid     : std_logic_vector;
    bresp   : std_logic_vector(1 downto 0);
    buser   : std_logic_vector;
    bvalid  : std_logic;
  end record;
  type t_axi_rd_slave_in_if is record
    arid      : std_logic_vector;
    araddr    : std_logic_vector;
    arlen     : std_logic_vector(7 downto 0);
    arsize    : std_logic_vector(2 downto 0);
    arburst   : std_logic_vector(1 downto 0);
    arlock    : std_logic;
    arcache   : std_logic_vector(3 downto 0);
    arprot    : std_logic_vector(2 downto 0);
    arqos     : std_logic_vector(3 downto 0);
    arregion  : std_logic_vector(3 downto 0);
    aruser    : std_logic_vector;
    arvalid   : std_logic;
    rready  : std_logic;
  end record;
  type t_axi_rd_slave_out_if is record
    arready : std_logic;
    rid     : std_logic_vector;
    rdata   : std_logic_vector;
    rresp   : std_logic_vector(1 downto 0);
    rlast   : std_logic;
    ruser   : std_logic_vector;
    rvalid  : std_logic;
  end record;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.test_pkg.all;

entity test2 is
  port (
    wr_port_in     : in  t_axi_wr_slave_in_if;
    wr_port_out    : out t_axi_wr_slave_out_if;
[... 154 lines cut ...]
  end process;

end architecture beh;
```
```
nvc --std=08 -a test_port_map_interface_2.vhd -e test -r
** Fatal: invalid type kind T_ARRAY in range_of
[00007FF674FA655A]
[00007FF674FED526]
[00007FF674FFE56C]
[00007FF675000F1F]
[00007FF67500531B]
[00007FF675005484]
[00007FF675005883]
[00007FF67500EAC3]
[00007FF67500ED0B]
[00007FF67500F30B]
[00007FF674FA2BA8]
[00007FF674FA2B1C]
[00007FF675085EE2] vhpi_put_data+0x231f2
[00007FF674FA13AE]
[00007FF674FA14E6]
[00007FFCC4567034] BaseThreadInitThunk+0x14
[00007FFCC5E626A1] RtlUserThreadStart+0x21
```

amb5l, 2022-10-11:
I am seeing something similar when running OSVVM's Axi4Full VC test suite, the TbAxi4_Shared1 test case produces this error:
```
nvc --std=08 -H 128m --work=osvvm_TbAxi4_MultipleMemory:C:/work/osvvm_tests/nvc/VHDL_LIBS/nvc-1.8-devel/OSVVM_TBAXI4_MULTIPLEMEMORY -L C:/work/osvvm_tests/nvc/VHDL_LIBS/nvc-1.8-devel -e TbAxi4_Shared1
** Fatal: invalid type kind T_ARRAY in range_of
[00007ff70561646a]
[00007ff70565c546]
[00007ff70566d54c]
[00007ff70566feff]
[00007ff7056742fb]
[00007ff705674464]
[00007ff70567460a]
[00007ff705674863]
[00007ff70567daa3]
[00007ff70567dceb]
[00007ff70567e2eb]
[00007ff705612c19]
[00007ff7056f7422] vhpi_put_data+0x24532
[00007ff7056113ae]
[00007ff7056114e6]
[00007fffadba54e0] BaseThreadInitThunk+0x10
[00007fffaea2485b] RtlUserThreadStart+0x2b
```
To reproduce, use the same OSVVM and NVC recipe as required for issue #551 and run the following in a TCL shell:
```
cd ~/work/tmp
source /work/OsvvmLibraries/Scripts/StartNVC.tcl
build /work/OsvvmLibraries/OsvvmLibraries
build V:/work/OsvvmLibraries/AXI4/Axi4/RunAllTests
```
You'll need to scroll back a little bit, the failed test is 2nd to last IIRC.

nickg, 2022-10-11:
@avelure's case is working now. The OSVVM test gets a bit further and then fails for a different reason. The code to constrain these ports is quite tricky so I'm still not convinced it's completely correct. Please test again, thanks!

amb5l, 2022-10-12:
Great progress. I agree with your OSVVM result - I see...
```
nvc --std=08 -H 128m --work=osvvm_TbAxi4_MultipleMemory:C:/work/osvvm_tests/nvc/VHDL_LIBS/nvc-1.8-devel/OSVVM_TBAXI4_MULTIPLEMEMORY -L C:/work/osvvm_tests/nvc/VHDL_LIBS/nvc-1.8-devel -e TbAxi4_Shared1

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=0000000000000000, ip=00007ff6283e2690] ***
```

avelure, 2022-10-12:
The fix solved this issue, though another one popped up, I'll add another issue.

nickg, 2022-10-13:
@amb5l the OSVVM crash is fixed by 88b945e

closed 2022-10-13 by nickg