nickg/nvc #546: Installing OSVVM

amb5l, 2022-10-03:
I have started experimenting with OSVVM. The code seems to be evolving quite fast. Might it be an idea to support up to date source snapshots by running the osvvm.pro TCL script? Similar to the recent change to the UVVM install script? The following `install-osvvm.tcl` script seems to work:

```
set osvvm_path [lindex $argv 0]

proc library {name} {
    if {$name != "osvvm"} {
        puts stderr "expected library osvvm"
        exit 1
    }
}

proc analyze {filename} {
    global osvvm_path
    if {[catch {exec nvc --std=2008 --work=osvvm -a $osvvm_path/$filename} result]} {
        puts stderr "ERROR\n$result"
        exit 1
    } else {
        puts stdout "$result"
    }
}

namespace eval osvvm {
    variable ToolVendor NVC
    variable ToolSupportsGenericPackages 0
}

cd ~/.nvc/lib
source "$osvvm_path/osvvm.pro"
```

There is one strange problem: the TCL script does not receive any output from the nvc command, although it is working. If `exec nvc --std=2008 --work=osvvm -a $osvvm_path/$filename` is changed to `exec nvc --version` there is plenty of output...

Obviously this approach depends on a TCL interpreter, but OSVVM users are likely to have one installed. The `install-osvvm.sh` script could then be simplified. You could still default to a known tested branch but allow the user to specify main if they want the latest, greatest?

amb5l, 2022-10-04:
OSVVM users are likely to build OSVVM along with their DUT. This recipe for precompiling the library might be nice but it's a low priority.

nickg, 2022-10-04:
So the shell script would just check out the git repository and then call the TCL script? That sounds reasonable, especially if OSVVM's own script supports NVC natively.

amb5l, 2022-10-04:
Yes that’s the idea. I’ll have a go at that script at some point soon and post my efforts here.

amb5l, 2022-10-11:
OK - here are candidate scripts.
1) install-osvvm.sh
```
#!/usr/bin/env bash
#
# Called by "nvc --install osvvm".
#
# Arguments:
#  $1 = UVVM repository tag (optional) e.g. 2022.09

SCRIPT_DIR=$( cd -- "$( dirname -- "${BASH_SOURCE[0]}" )" &> /dev/null && pwd )

. $(dirname $BASH_SOURCE)/functions.sh

if [ -z "$1" ]; then
    branch="2022.09"
fi

git_wrapper https://github.com/osvvm/OsvvmLibraries $branch

tclsh $SCRIPT_DIR/install-osvvm.tcl
```
2) install-osvvm.tcl
```
variable _dest [exec cygpath -m "${::env(HOME)}"]/.nvc/lib

proc library {name} {
    if {$name != "osvvm"} {
        puts stderr "expected library osvvm"
        exit 1
    }
}

proc analyze {filename} {
    variable _dest
    puts "nvc --std=2008 --work=osvvm:${_dest}/osvvm -a osvvm/${filename}"
    if {[catch {exec nvc --std=2008 --work=osvvm:$_dest/osvvm -a osvvm/$filename >@ stdout 2>@ stdout} result]} {
        puts stderr "ERROR\n$result"
        exit 1
    }
}

namespace eval osvvm {
    variable ToolVendor NVC
    variable ToolSupportsGenericPackages 0
}

source "osvvm/osvvm.pro"

```
These are tested on Windows/MSYS2.

nickg, 2022-10-13:
This seems fine, but can we extend it to also build the UART, AXI4, etc libraries? I'm using some of those as regression tests at the moment.

BTW in [88b945e](https://github.com/nickg/nvc/commit/88b945e08840381d0bce49745ec232ca2a362317) I just updated `functions.sh` so it ensures `NVC_INSTALL_DEST` is set to the install directory with correct `$HOME` expansion on Windows, so you should be able to use that environment variable directly.

amb5l, 2022-10-13:
That makes a lot of sense. I'll change it compile everything and post here when I've got it working.

amb5l, 2022-10-16:
This TCL script compiles all of OSVVM (the Bash script does not need changing):
```
variable _dest [exec cygpath -m "${::env(HOME)}"]/.nvc/lib
variable _lib ""

proc DirectoryExists {Directory} {
    return [file exists $Directory]
}

proc include {PathAndFile} {
    set calling_dir [pwd]
    cd [file dirname $PathAndFile]
    puts "Working in [pwd]:"
    source [file tail $PathAndFile]
    cd $calling_dir
}

proc library {LibName} {
    variable _lib
    set _lib $LibName
}

proc analyze {PathAndFile} {
    variable _dest
    variable _lib
    puts "nvc --std=2008 --work=${_lib}:${_dest}/${_lib} -a ${PathAndFile}"
    if {[catch {exec nvc --std=2008 --work=${_lib}:$_dest/$_lib -a $PathAndFile >@ stdout 2>@ stdout} result]} {
        puts stderr "ERROR\n$result"
        exit 1
    }
}

namespace eval osvvm {
    variable ToolVendor NVC
    variable ToolSupportsGenericPackages "true"
}

source "OsvvmLibraries.pro"
```

nickg, 2022-10-18:
Committed, thanks.

closed 2022-10-18 by nickg